espressif/arduino-esp32 #8181: Since platform-espressif32 version 4.0 ETHERNET not working on ESP32-EVB

CANguru-System, 2023-05-10:
### Board

Olimex ESP32-EVB

### Device Description

Windows PC via ETHERNET

### Hardware Configuration

USB-Cable to TERA TERM on the PC

### Version

latest master (checkout manually)

### IDE Name

PlatformIO with platform = espressif32 @ ~6.2.0

### Operating System

Windows 11 PRO

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

460000

### Description

If I compile and upload the following sketch (it is the well known LAN example) on the ESP32-EVB with the last espressif32 platform 6.0.2 I get the following output

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (131) lan87xx: lan87xx_init(499): wrong chip ID
E (131) esp_eth: esp_eth_driver_install(215): init phy failed
[   136][E][ETH.cpp:333] begin(): esp_eth_driver_install failed

So the ethernet communication with the PC does not work.
If I go back and use platform = espressif32 @ ~3.5.0, I get the connection and I can read the google.com answer. But for other reasons it is not possible for me to use 3.5.0 because I also use CAN communication and under 3.5.0 TWAIN is not present.

### Sketch

```cpp
#include <Arduino.h>

/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  }

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Message

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (131) lan87xx: lan87xx_init(499): wrong chip ID
E (131) esp_eth: esp_eth_driver_install(215): init phy failed
[   136][E][ETH.cpp:333] begin(): esp_eth_driver_install failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

CANguru-System, 2023-05-18:
Item can be closed.
If you change/add WiFiEvent in the following way everything is fine.
void WiFiEvent(WiFiEvent_t event)
{
  #define ETH_START 18
  #define ETH_CONNECTED 20
  #define ETH_GOT_IP 22
  #define ETH_DISCONNECTED 24
  #define ETH_STOP 26
  switch (event)
  {
  case ETH_START: //SYSTEM_EVENT_ETH_START:
    Serial.println("ETH Started");
    // set eth hostname here
    ETH.setHostname("esp32-ethernet");
    break;
  case ETH_CONNECTED: //SYSTEM_EVENT_ETH_CONNECTED:
    Serial.println("ETH Connected");
    break;
  case ETH_GOT_IP: //SYSTEM_EVENT_ETH_GOT_IP:
    Serial.print("ETH MAC: ");
    Serial.print(ETH.macAddress());
    Serial.print(", IPv4: ");
    Serial.print(ETH.localIP());
    if (ETH.fullDuplex())
    {
      Serial.print(", FULL_DUPLEX");
    }
    Serial.print(", ");
    Serial.print(ETH.linkSpeed());
    Serial.println("Mbps");
    eth_connected = true;
    break;
  case ETH_DISCONNECTED: //SYSTEM_EVENT_ETH_DISCONNECTED:
    Serial.println("ETH Disconnected");
    eth_connected = false;
    break;
  case ETH_STOP: //SYSTEM_EVENT_ETH_STOP:
    Serial.println("ETH Stopped");
    eth_connected = false;
    break;
  default:
    break;
  }
}
The bug is that all event numbers are wrong!

mrengineer7777, 2023-05-18:
@CANguru-System The signature for WiFi.onEvent() changed from arduino-esp32 1.x to 2.x.

Here's how I use it.  `iNetEvtCB` is my callback function, which is equivalent to your WiFiEvent().
```
inet_evt_hnd = WiFi.onEvent(iNetEvtCB);
//Network event callback
void iNetEvtCB(arduino_event_id_t event, arduino_event_info_t info) {
    IPAddress ip;

    switch(event) {
        //ETH
        case ARDUINO_EVENT_ETH_CONNECTED:
            Serial.printf("ETH connected");
            break;
        case ARDUINO_EVENT_ETH_DISCONNECTED:
            Serial.println("ETH disconnected"); 
            break;
        case ARDUINO_EVENT_ETH_GOT_IP:
....
}
```

CANguru-System, 2023-05-19:
@mrengineer7777 
Thank you for explanation.
So I guess I had 2 problems. First is that  there is a short delay to add as the first line of the script because there is a problem with the PHY on the Olimex ESP32-EVB. And second was the change of the signature of WiFi.onEvent(), was I was not aware of.
Both problems have a solution and everything works fine.
But in the end I think someone should modify the ETHERNET example to take into account the change of the signature.

closed 2023-05-19 by Jason2866