espressif/arduino-esp32 #8191: Failed to connect to ESP32: No serial data received.

afsheenfaiq, 2023-05-11:
### Board

ESP32 s

### Device Description

On Arduino UNO

### Hardware Configuration

no

### Version

v2.0.1

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Failed to connect to ESP32: No serial data received.

### Sketch

```cpp
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
//
//            You must select partition scheme from the board menu that has at least 3MB APP space.
//            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15 
//            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well

// ===================
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
//#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
// ** Espressif Internal Boards **
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD

#include "camera_pins.h"

// ===========================
// Enter your WiFi credentials
// ===========================
//const char* ssid = "**********";
//const char* password = "**********";
const char* ssid = "STAFF";
const char* password = "Staff2023$nas";
//const char* ssid = "faiq001";
//const char* password = "FA0508420374";
void startCameraServer();
void setupLedFlash(int pin);

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_UXGA;
  config.pixel_format = PIXFORMAT_JPEG; // for streaming
  //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){
      config.jpeg_quality = 10;
      config.fb_count = 2;
      config.grab_mode = CAMERA_GRAB_LATEST;
    } else {
      // Limit the frame size when PSRAM is not available
      config.frame_size = FRAMESIZE_SVGA;
      config.fb_location = CAMERA_FB_IN_DRAM;
    }
  } else {
    // Best option for face detection/recognition
    config.frame_size = FRAMESIZE_240X240;
#if CONFIG_IDF_TARGET_ESP32S3
    config.fb_count = 2;
#endif
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1); // flip it back
    s->set_brightness(s, 1); // up the brightness just a bit
    s->set_saturation(s, -2); // lower the saturation
  }
  // drop down frame size for higher initial frame rate
  if(config.pixel_format == PIXFORMAT_JPEG){
    s->set_framesize(s, FRAMESIZE_QVGA);
  }

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

#if defined(CAMERA_MODEL_ESP32S3_EYE)
  s->set_vflip(s, 1);
#endif

// Setup LED FLash if LED pin is defined in camera_pins.h
#if defined(LED_GPIO_NUM)
  setupLedFlash(LED_GPIO_NUM);
#endif

  WiFi.begin(ssid, password);
  WiFi.setSleep(false);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // Do nothing. Everything is done in another task by the web server
  delay(10000);
}
```

### Debug Message

```plain
Serial port COM9
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-05-11:
Hi @afsheenfaiq, at first glance, it looks as a wrong usage. Marking this as a question and hopefully someone from the community will help answering your issue.

lbernstone, 2023-05-11:
Most likely a bad cable.  Try a high quality cable, and a different USB port.

SuGlider, 2023-05-14:
@afsheenfaiq 

I have a couple ESP32 boards with a similar issue.... In order to solve it, I press and release a few times the `BOOT` button of the board, while it is try to upload (displaying the dots "..................").

Some 3rd party boards have such "hardware" issue.

mrengineer7777, 2023-05-21:
Does it work if you lower the upload speed?

SuGlider, 2023-05-25:
Closing the issue. Please feel free to reopen it, whenever necessary.

closed 2023-05-25 by SuGlider

Mosazghi, 2023-07-17:
Same issue 
i have tried holding and releasing the boot button during upload but doesnt fix the problem.

afsheenfaiq, 2023-07-18:
It happens because of incorrect WiFi name and password.Put the correct
credentials of your network and it will work.

tesarect, 2023-07-27:
Facing same issue. Here are my settings,
[image]

For most people, i could see the options under "programmer" shows various and they choose `AVR ISP`, strangely for me only `esptool` appears. And by default flash freq was pointing to 80MHz, but for others its in 40MHz. Also people choose `ESP Wrover Module` as board but under code they choose `AI Thinker Esp board`.

Tried pressing on-board reset button while "connecting . . . . ."  appears (didn't work),
also not pressed while same text appeared (didn't work),
expressive site suggested to use 3.3v instead of 5v (didn't work),
rechecked my wire connectivity including bread board slots (didn't work),
both flash freq 40 & 80MHz (didn't work),
both `AI thinker esp` & `ESP Wrover module` under board manager (didn't work).

subsubyee, 2023-08-06:
I am also facing the same problem.
only esptool comes out of the programmer too,
and it is not uploaded with the same error as the questioner.

pabasara404, 2023-08-06:
> 

You have to install USBAsp AVR Arduino Programmer Drivers. Zidag is easiest way to install it.

raghuvaran-nampelly, 2023-08-08:
I HAVE ALSO FACING THE SAME ISSUE

ElectricXion, 2023-08-08:
I was also having the same issue but after a bit of troubleshooting I was to resolve the problem. For my case I am using the AI Thinker board and I made no changes to the software settings.

ElectricXion, 2023-08-08:
If you are using the ESP32-CAM-MB to program the AI THINKER CAM board then I suggest you try using another board or another cable, however if you are using the FTDI module and getting the same issue then you do not need to change anything in the software. The following are the connections to be made:
 FTDI         ESP32 CAM
  5V                5V
 GND            GND (use the GND pin that is near to the 5v pin)
 DTR              IO0
  RX               UOT
  TX               UOR

Ensure that the FTDI module is set to 5v using the jumper if necessary.

Next press "Upload" and then hold down the RST button on the ESP32-CAM until the Arduino IDE or PlatformIO gives an indication that the code is beginning to upload. For the Arduino IDE, once you see "Connecting..................", release the RST button and the code should upload just fine.

If your ftdi module doesn't have a DTR pin then you will have to manually connect IO0 to ground using a jumper, press "upload" while holding down the RST button, then remove the jumper when the code begins to upload then you should release the RST button. A bit overkill for a single upload......

murrayreinhart, 2023-09-29:
@ElectricXion thanks so much you solved my issue. I've been struggling with this error for many hours and following 2 tutorials and checking and rechecking my wiring matched theirs with no joy. You recommended a different wire pattern that works!

atharAIM, 2023-10-27:
Connnect the ground (GND) close to the 5 Volt pin.  :')
Its stupid ik but it works. XD

zedee, 2023-10-29:
> (use the GND pin that is near to the 5v pin)

Wait a minute.... **this also worked for me** finally after an hour and a half going nuts checking cables, rechecking, changing them, trying another USB port, changing USB cable, trying another USB to TTL chip....

Does anyone have an explanation on how can this be even possible? Shouldn't GND be GND regardless of it's one side or another???? What I'm missing?

ubhere9, 2023-11-01:
hiii....i tried everything but still the same error remains ...pls help me out here

zedee, 2023-11-04:
I know it's not the best or more clear picture but........

[image]

```
USB to TTL  ------------------ ESP32

5V   -------------------------    5V [White Cable]
GND  -------------------------   GND (The one next to ESP's 5V, NOT the one on the 3V3 side) [Black Cable]
RX   -------------------------   UOT (On the 3V3 side) [Grey Cable]
TX   -------------------------   UOR (On the 3V3 side) [Purple Cable]

                                 IO0 -> GND (On the 3V3 side) [Brown Cable]  
                                 :: Yes, connect those pins on the ESP32 together.
```

Also, check: https://dronebotworkshop.com/esp32-cam-intro/

camp-easy, 2023-11-05:
> ### Board
> ESP32 s
> 
> ### Device Description
> On Arduino UNO
> 
> ### Hardware Configuration
> no
> 
> ### Version
> v2.0.1
> 
> ### IDE Name
> Arduino
> 
> ### Operating System
> Windows 10
> 
> ### Flash frequency
> 40
> 
> ### PSRAM enabled
> yes
> 
> ### Upload speed
> 115200
> 
> ### Description
> Failed to connect to ESP32: No serial data received.
> 
> ### Sketch
> ```c++
> #include "esp_camera.h"
> #include <WiFi.h>
> 
> //
> // WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
> //            Ensure ESP32 Wrover Module or other board with PSRAM is selected
> //            Partial images will be transmitted if image exceeds buffer size
> //
> //            You must select partition scheme from the board menu that has at least 3MB APP space.
> //            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15 
> //            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well
> 
> // ===================
> // Select camera model
> // ===================
> //#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
> //#define CAMERA_MODEL_ESP_EYE // Has PSRAM
> //#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
> //#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
> //#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
> //#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
> //#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
> //#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
> #define CAMERA_MODEL_AI_THINKER // Has PSRAM
> //#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
> //#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
> // ** Espressif Internal Boards **
> //#define CAMERA_MODEL_ESP32_CAM_BOARD
> //#define CAMERA_MODEL_ESP32S2_CAM_BOARD
> //#define CAMERA_MODEL_ESP32S3_CAM_LCD
> 
> #include "camera_pins.h"
> 
> // ===========================
> // Enter your WiFi credentials
> // ===========================
> //const char* ssid = "**********";
> //const char* password = "**********";
[... 113 lines cut ...]
> }
> 
> void loop() {
>   // Do nothing. Everything is done in another task by the web server
>   delay(10000);
> }
> ```
> 
> ### Debug Message
> ```
> Serial port COM9
> Connecting......................................
> 
> A fatal error occurred: Failed to connect to ESP32: No serial data received.
> For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
> Failed uploading: uploading error: exit status 2
> ```
> 
> ### Other Steps to Reproduce
> _No response_
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Bro just try to press boot button, keep press, connect with usb-c cable to the pc and then release the boot button, and now try to uploade a simple code with a Serial.println("...") in the loop with a delay(3000) .the code should be uploaded in the esp32 and in the serial monitor should compare the correct output.

Erick-LB, 2023-11-18:
> Connnect the ground (GND) close to the 5 Volt pin. :') Its stupid ik but it works. XD
How exactly did you do it? 
Did you connect it to the 5V with what type of resistor?

zedee, 2023-11-18:
> > Connnect the ground (GND) close to the 5 Volt pin. :') Its stupid ik but it works. XD
> 
> How exactly did you do it? Did you connect it to the 5V with what type of resistor?

No, no resistors on the 5V input!!!
5V from ESP to 5V of USB to UART device.
GND next to 5V to GND of USB to UART device.

You need a usb to UART converter to program the vanilla" esp32 cam (like cp2012, you'll find them for cheap online on any marketplace) 

Check my pic and description, each cable color says where is connected to the esp32

Gowthamipriya05, 2023-11-19:
Iam also facing same issue

Gowthamipriya05, 2023-11-19:
Fatal error ..cannot connect to esp32 .no serial data received

Gowthamipriya05, 2023-11-19:
Can any one resolve this

camp-easy, 2023-11-19:
> Can any one resolve this

Try to press boot button, connect the esp32 and now release the boot button and try to upload the code.

WolverineRising, 2023-12-05:
I tried every solution posted here but nothing worked, then I noticed the board doesn't reset when I press the "EN" push-button, so I shorted the EN pin (reset) to ground pin with a jumper wire, that's when the MCU was actually reset and displayed the boot message on the serial terminal, followed by erasing the flash using esptool and everything worked fine afterwards. I have no explanation why that happen or how the solution works.

teddy1908, 2023-12-13:
Hi all,

I am getting a similar issue. I hope any of you can help. Namely ...
I designed my own PCB with a ESP32 S3 WROOM - N32R8V which is to be programmed via USB-C and CP2102N.
Schematic is as follows:

[image]

I installed the CP2102 drivers and find "the Silicon Labcs CP210c US to UART Bridge" on COM13 ... I am guessing its connected here.

When I go into arduino IDE I can get board info
[image]
only when the board is connecte and ON ... so this also looks promising.

After I try to download the program to ESP 32 I get the dreaded "No serial data received." message.
[image]

These are my settings in arduino (one of several I tried)
[image]

Any idea what is the problem?

teddy1908, 2023-12-13:
> Pull-up resistors for EN and BOOT should be added. 4.7K-10K should be OK.

I just added them and still the same.

> when you just open the serial monitor to the board and hit EN, do you see output on the serial? One option is not to get output and the other is that the board is not properly resetting. Start with checking the Serial output, then if that is OK, disconnect, press and hold BOOT, then while holding, press and release EN and keep holding the BOOT. Start upload (do not release BOOT) and see if it goes through.

I opened the serial monitor in Arduino, connected the board (windows made the silly noise that it is connected), pressed the BOOT/EN in all combinations and nothing happened. Also tried programming and still the same message.

teddy1908, 2023-12-13:
Guys ... found the mistake.
I fell like an utter dum...s.
In the schematic i switched the connection TX->RX twice ... so in default mode it was TX->TX. Luckily I know how dumb I can be and left solder jumpers for both options ... works now.

Thanks for your comments.

harhar111, 2024-04-25:
In my case, I short circuit pin ΤΧ0 with EN pin and then it started work normally again.

jacksaxi, 2024-05-08:
aftef a short circuit the esp has stoped over heating but it still does not upload code

LiamPerson, 2024-09-14:
I am dumb and forgot to connect the `IO0` connector to `GRD` 
[image]

norio-hayashi, 2025-01-08:
The USB port may be in conflict with another service. Removing that service will resolve the issue.

daahl, 2025-02-28:
I found the following solution that I haven't read about before. For my board, **it was the reset button on the ESP32-CAM board that was shorted**. The solution was to simply desolder the button.

If you have a multimeter at hand, you can easily test this.
- Check voltage between GND and VIN. Should be 3.3 V.
- Check voltage between GND and EN. Should be 3.3 V. (EN is the chip enable pin. It has to be high for the ESP to power on.)

If VIN is 3.3 V but EN is 0 V, then the resest button is possibly shorted (reset button pulls EN low to reboot the ESP).
- Check continuity on both pads of the button. Be careful to not touch the metal housing of the button with the probes, as that will give a false positive.

If the button is shorted, then simply desoldering it might be the solution. It worked for me at least. Glad I didn't spend money on a new one before I fixed this one. :) 

Note that the reset button on the ESP-CAM-MB will still work after this.

[image]

[image]

DimitriosPap, 2025-04-28:
For me the solution was to connect GND next to 5V pin and not in any other GND pins of the board...

MeowCata, 2025-05-17:
> I found the following solution that I haven't read about before. For my board, **it was the reset button on the ESP32-CAM board that was shorted**. The solution was to simply desolder the button.我找到了以下之前从未见过的解决方案。我的主板**是 ESP32-CAM 板上的重置按钮短路了 **。解决办法很简单，就是把按钮拆焊一下。
> 
> If you have a multimeter at hand, you can easily test this.如果您手边有万用表，您可以轻松地测试这一点。
> 
> * Check voltage between GND and VIN. Should be 3.3 V.检查 GND 和 VIN 之间的电压。应为 3.3V。
> * Check voltage between GND and EN. Should be 3.3 V. (EN is the chip enable pin. It has to be high for the ESP to power on.)检查 GND 和 EN 之间的电压。应为 3.3 V。（EN 是芯片使能引脚。ESP 必须处于高电平才能启动。）
> 
> If VIN is 3.3 V but EN is 0 V, then the resest button is possibly shorted (reset button pulls EN low to reboot the ESP).如果 VIN 为 3.3 V 但 EN 为 0 V，则重置按钮可能短路（重置按钮将 EN 拉低以重新启动 ESP）。
> 
> * Check continuity on both pads of the button. Be careful to not touch the metal housing of the button with the probes, as that will give a false positive.检查按钮两侧焊盘的导通性。注意不要让探针接触按钮的金属外壳，否则会导致误报。
> 
> If the button is shorted, then simply desoldering it might be the solution. It worked for me at least. Glad I didn't spend money on a new one before I fixed this one. :)如果按钮短路了，那么直接拆焊或许就能解决问题。至少对我来说，这个方法有效。还好我没在修好这个之前花钱买个新的。:)
> 
> Note that the reset button on the ESP-CAM-MB will still work after this.请注意，ESP-CAM-MB 上的重置按钮此后仍然有效。
> 
> [image]
> 
> [image]

For my board, the voltage between GND and EN is 5V, and voltage between VIN and GND is 5V, the problem still exists