espressif/arduino-esp32 #8192: Wrong SPIs typology for ESP32C3

mgrenonville, 2023-05-11:
### Board

esp32-c3-devkitm-1

### Device Description

ESP32-C3 in devkitm-1

### Hardware Configuration

Master SPI pins (4 - `SCK`, 5 - `MISO`, 6 - `MOSI`) to a slave module

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Linux

### Flash frequency

160Mhz

### PSRAM enabled

yes

### Upload speed

N/A

### Description

According to documentation, ESP32C3 has [2 SPIs peripherals](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html#overview-of-esp32-c3-s-spi-peripherals), splitted in SPI0 & SPI1 used for attached flash memory, and SPI2 that is a general purpose SPI controller.

When instanciating a SPI on ESP32C3, the default spi number is `FSPI` that expands to`0`. But this is confusing with the documentation that says SPI0 is for flash memory (and is compatible with convention in FSPI). 

Other issue is that `SPI.begin` returns `void`, and when passing a value larger than `SPI_COUNT`, we don't get errors. Arduino library for esp8266 returns a boolean with success or failure.

### Sketch

```cpp
#include <SPI.h>
#include <Arduino.h>

#define HSPI_MISO MISO
#define HSPI_MOSI MOSI
#define HSPI_SCLK SCK
#define HSPI_SS 18

static const int spiClk = 100000; // 1 MHz

SPIClass *working = NULL;
SPIClass *notWorking = NULL;

void setup()
{

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  working = new SPIClass();
  //notWorking = new SPIClass(HSPI);
  
  working->begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI);

  working->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
  Serial.printf("worked: %i != 0", working->getClockDivider());
  working->endTransaction();

  // set up slave select pins as outputs as the Arduino API
  // doesn't handle automatically pulling SS low
  pinMode(HSPI_SS, OUTPUT); // HSPI SS
}

void loop()
{
  delay(100);
}
```

### Debug Message

```plain
Depending on SPIClass, sketch returns getClockDivider() == 0 or a correct value
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-05-11:
Thanks for the report @mgrenonville . We will look into it for 3.0.0

VojtechBartoska, 2024-02-20:
@P-R-O-C-H-Y Please take a look on this, postponing it to 3.1.0 milestone, there are 2 relevant tickets I am closing in favor of this one.

PR #11477 (2025-06-17): feat(spi): Add return values to SPI begin + comments on SPI buses
## Description of Change
This pull request is modifying the `SPIClass::begin` method to return a `bool` for better error handling. These updates enhance compatibility with newer ESP32 variants and provide clearer feedback on initialization success.

## Tests scenarios

## Related links
Closes #8192
closed 2025-06-17 by me-no-dev