espressif/arduino-esp32 #8193: Incompatibility issue with lwip napt on ESP32-S3 using Arduino framework

Gspohu, 2023-05-11:
### Board

ESP32-S3

### Device Description

DevKitC

### Hardware Configuration

Out of context

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Kubuntu Linux 22.04

### Flash frequency

1000

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello @espressif team,

I have been developing a connected object project that is running successfully on an ESP32-S3 using the Arduino framework. I am now trying to add a functionality that involves lwip napt for a WiFi extender. 

I tried prototyping this in Arduino but encountered an issue where I get the following error messages:
```
undefined reference to `ip_napt_enable_no'
undefined reference to `ip_napt_enable'
```

Interestingly, I had no such problems while implementing the same functionality with espidf. 
For me, it is not possible to recode the whole firmware on espidf.

I have made several attempts to troubleshoot this issue to no avail:
- I tried including lwip as a "lib" or "component" in my project directory.
- I attempted to use Arduino as a component of espidf, but this was quite challenging and resulted in a performance drop in my Arduino code. Plus, the lwip version remained as the Arduino one.
- I tried replacing the lwip directory in the Arduino framework with the espidf one. As expected, this didn't work because Arduino uses precompiled libraries. I would love to understand the rationale behind this choice if anyone can help.
- I manipulated my platformio.ini to add the necessary compilation flags, but this was unsuccessful because, again, Arduino uses precompiled libraries.
- I also tried this: platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.3/platform-espressif32-2.0.5.3.zip. However, it crashed my platformio, and I had to reset the software.

The issues I encountered seem to be similar to the ones described here, but I couldn't find a feasible solution:
- https://github.com/espressif/arduino-esp32/issues/6421
- https://github.com/espressif/arduino-esp32/issues/2826

I noticed there was supposed to be a fix in the Arduino lib builder (https://github.com/espressif/esp32-arduino-lib-builder/pull/69), but it seems to be forgotten at the moment.

Interestingly, I found an implementation for ESP8266 (https://github.com/martin-ger/lwip_nat_arduino) but couldn't find an equivalent for ESP32.

For your reference, I am using platformio on vscode. 

Is there any way to get lwip napt working with the Arduino framework on ESP32-S3? Any guidance or help would be greatly appreciated. 

Thank you in advance for your support.

### Sketch

```cpp
#include "lwip/lwip_napt.h"

#include "lwip/app/dhcpserver.h"
// Initialize the NAT feature
ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);

// Enable NAT on the AP interface
ip_napt_enable_no(1, 1);

// Set the DNS server for clients of the AP to the one we also use for the STA interface
dhcps_set_DNS(WiFi.dnsIP());
```

### Debug Message

```plain
undefined reference to `ip_napt_enable_no'
undefined reference to `ip_napt_enable`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-11:
It seems that https://github.com/espressif/esp32-arduino-lib-builder/pull/69 should be reviewed and merged.
@me-no-dev - maybe add that to the 2.x and 3.0.0 branches?

Gspohu, 2023-05-11:
@VojtechBartoska I just saw that you have assigned it to 3.0, which will be out in quite some time.

It's only one line in a configuration file, why wait so long?

me-no-dev, 2023-05-11:
The problem is that NAPT also enables other things in ESP-IDF that cause performance issues for everyone, even when NAPT is off. We check from time to time to see if they have "fixed" that in IDF and when they do, we can add support for it.

Jason2866, 2023-05-11:
@Gspohu I am the maintainer of the Arduino fork for Tasmota. We use NAPT and it does work. You can use latest Tasmota Core 2.0.9 with Platformio
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.05.00/platform-espressif32.zip
```
The version is trimed to use low flash and RAM. Do not expect logging messages from IDF.

Gspohu, 2023-05-11:
Thanks for your message @Jason2866 , but what you suggest was my first try, but the result was not glorious.

Ok I was able to compile, but once the firmware was uploaded, my ESP32-S3 was rebooting in a loop, and other projects on my platformio were affected, for some reason, everything I compiled and upoloaded with my platformio, even when there was never any mention of your solution, gave an ESP32 that constantly crashed. The only solution I found was to delete the .plateformio folder, uninstall the extension, uninstall vscode and reinstall everything. Really strange.

Gspohu, 2023-05-11:
@Jason2866 I just tried it and the problem is exactly the same. My ESP restarts in a loop

Jason2866, 2023-05-11:
@Gspohu  S3 needs 100% correct config. If not it bootloops.
What type of flash and PSRAM? QIO / OPI? It needs be correct for flash AND PSRAM.
And flash size too. You started with the zero error forgiving device ;-)
Just can say S3 and NAPT is working with our framework. Running Tasmota range extender on it.

Gspohu, 2023-05-12:
@Jason2866 Okay, but I'm not sure if I understand.
My card is this one: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

When I use the classic tools, the configuration must be perfect, because I never had any problems like this.

I'm going to put an issue on Tasmota's repository, so as not to pollute the discussion here. We can continue to discuss this there

Gspohu, 2023-05-12:
I decided to test the Tasmota web installer, thinking that maybe it was my platformio configuration that was the problem, or even my code (https://tasmota.github.io/install/), but the result is the same. There still seems to be some bugs with the ESP32-S3 support on Tasmota.

Finally, it's not possible to file an issue on the Tasmota repository. To finish on this topic here, could you point me to a place to discuss this topic @Jason2866 ?

Jason2866, 2023-05-12:
@Gspohu It is a config thing. There is not just one S3 Devkit. The have variants with Flash and PSRAM types! You found us in Discord already. Let's talk there.

RobertGnz, 2023-12-26:
> The problem is that NAPT also enables other things in ESP-IDF that cause performance issues for everyone, even when NAPT is off. We check from time to time to see if they have "fixed" that in IDF and when they do, we can add support for it.

I recently compiled espressif idf 5.1.2 with napt feature using eclipse ide. The result was that napt works as expected provided you use esp_netif interface to enable napt. I did not noticed any overhead when enabling CONFIG_LWIP_L2_TO_L3_COPY which makes a copy of every tcp frame before passing a reference to it to L3 layer.
Furthermore I tested another buid disabling CONFIG_LWIP_L2_TO_L3_COPY but keeping enabled all other parameters needed for napt enabling, and the result was that napt worked perfectly as expected.
It appears that CONFIG_LWIP_L2_TO_L3_COPY is unnecessary to make napt working. So making a copy of tcp frames is unnecessary.
I believe that the CONFIG_LWIP_L2_TO_L3_COPY was initially programmed for debuging purpose during the development phase but now it is unnecessary so it can be removed.

VojtechBartoska, 2024-07-01:
Hello, napt was implemented and this issue is supposed to be solved. I am closing it, if needed you can reopen.

closed 2024-07-01 by VojtechBartoska

Rich968, 2024-07-30:
@VojtechBartoska I have updated esp32 board libraries and get:  lwip/napt.h  no such file or directory after including it in a project. Has the arduino esp32 lib builder been updated? (If that's relevant)

Jason2866, 2024-07-30:
@Rich968 Yes, NAPT support has been added some time ago. The feature needs to be enabled in the sdkconfig. So yes there was an update in Arduino Lib Builder when NAPT support was added. You have to use actual Arduino Lib Builder and settings.

Rich968, 2024-07-30:
@VojtechBartoska  I have core 3.0.2 installed and updated today. However lwip/napt.h when included in a project is not found. Have the file names changed compared to NAPT with esp8266?

Rich968, 2024-07-30:
@VojtechBartoska I have now found new lwip directory for esp32 boards after updating today, and the files names etc are all completely different..........  Thanks for putting me in the right direction.

Rich968, 2024-07-31:
@Jason2866  Thanks for that. I have found the relevant NAPT stuff in the library now. Also found dhcpserver.h However if include that dhcp file in a project I get error: ospi.h file missing.   Any ideas on this?  I'm attempting to replicate the esp8266 range extender arduino example using an ESP32. Has anybody successfully done this with the Arduino IDE?  Are the updated board libraries capable of this now anyway?

Rich968, 2024-07-31:
@Jason2866 Thanks for that. I have got napt and dhcpserver files included successfully in the project now.

Rich968, 2024-07-31:
Seems like the lwipdhcpserver.h from esp8266 library is very different from dhcpserver.h file from ESP32 library. I'm attempting to replicate the range extender esp8266 example on the esp32. In view of these differences - has anybody successfully implemented this on an ESP32 with the Arduino IDE?