espressif/arduino-esp32 #8201: std::random_device always returns constant value

Zhu-jiatong, 2023-05-13:
### Board

Irrelevant

### Device Description

ESP32 NodeMCU

### Hardware Configuration

Bare ESP32 Board

### Version

v2.0.9

### IDE Name

Visual Micro (Arduino IDE)

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I intended to use a real random number generator for cryptographic purposes. So, I choose `std::random_device`, which should be a platform independent way that is able to return real random numbers.

However, the generator produces the same value `3499211612` every time the ESP32 is powered on, which is not random.

### Sketch

```cpp
By using this code:

void setup()
{
 Serial.begin(115200);
 ... // wait until Serial is ready;
 std::cout << std::random_device()();
}
```
```

### Debug Message

```plain
Output on serial monitor every time the board is powered on: `3499211612`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

45gfg9, 2023-05-13:
Reproducible for me. Just had a quick look at cppreference and found this:

> `std::random_device` may be implemented in terms of an implementation-defined pseudo-random number engine if a non-deterministic source (e.g. a hardware device) is not available to the implementation. In this case each [`std::random_device`](https://en.cppreference.com/w/cpp/numeric/random/random_device#top) object may generate the same number sequence.

But, yes, ESP32 does have hardware RNG so PRNG does not seem like an acceptable behavior. I think this is a bug.

mrengineer7777, 2023-05-13:
This behavior may be intentional. Possibly related: https://github.com/espressif/arduino-esp32/pull/7848. Does the issue occur for an older version, say 2.0.6?

Zhu-jiatong, 2023-05-14:
I haven't been able to test it on older versions yet, as the Visual Micro board manager only provides the option to install the newest version.

igrr, 2023-05-14:
This is not intentional, we have overlooked checking how this function is implemented in libstdc++. I would recommend opening a feature request in IDF for this, sounds like something we could support.

Zhu-jiatong, 2023-05-14:
Ok, I submitted a feature request in the IDF's repository.

SuGlider, 2023-05-14:
@Zhu-jiatong  - `std::random_device` is not part of Arduino Core implementation, as already explained above. 
Therefore, I'll close this issue and let you follow it under IDF's github.

As for Arduino related random functionality, #7848 has an example about how to generate pseudo and real random sequences using Arduino APIs.

closed 2023-05-14 by SuGlider