espressif/arduino-esp32 #8202: Bluetooth SSP fails and crash under a specific motorola model

italocjs, 2023-05-13:
### Board

ESP32 WROOM32 - Proprietary board

### Device Description

The device is a Bluetooth to RS232 adapter, very simple has plenty of power available (lm2596 3A at 3.3v).  The same issues persist using esp32 dev modules.

### Hardware Configuration

Serial2 connected to pins gpio 16,17 (default for arduino)
LED output on gpio2.

### Version

v2.0.9

### IDE Name

VSCODE + ESPIDF + Arduino as component

### Operating System

Windows 11

### Flash frequency

Not changed (40)

### PSRAM enabled

no

### Upload speed

default

### Description

There are two ongoing issues, which may or not be related,  if they are not i'll open another issue,  I have set verbose level for the bluetooth settings under menuconfig to help reading the logs, 

Issue 01 - With any android phone ive tested, under Arduino BluetoothSerial, the first connection (opening an serial terminal) always work, but if the device disconnects and try to connect again, the ESP32 refuses the connection until the esp32 reset button is pressed.  I've found an workaround by stoping and restarting the bluetooth when the number of clients is changed. (note task 

Issue 02 - Both under spp_acceptor example (ESP_IDF) and BluetoothSerial (arduino) example, when starting an serial connection to the esp32 with some specific android phones (listed below), the esp32 crashes and reboots.   I have traced the problem down to the SSP (secure simple pairing),  when its disabled the phone pair and connects just fine!.  (note: for disabling it on BluetoothSerial some extra changes were needed, the modified file is also attached).

Phones crashing:
    - Motorola E7 (sporadic)
    - Motorola E6 Play (sporadic)
    - Galaxy A03 core (rarely)
    - Motorola MOTO E20 (always)

There are commented (step by step) logs in the github for the project to help diagnose the issue.  Somehow it crashes duo a integer divide by zero, this is coming from somewhere within the bluetooth libraries, but the log isnt helpful finding out exactly why.

I do have an jtag (esp-prog) here, i've tried using it but ended up crashing everything when i tried add breakpoints,  not sure how to debug deeper in the bluetooth,

### Sketch

```cpp
https://github.com/italocjs/xBTSAT_V1.3/tree/ssp_disabled_test
```

### Debug Message

```plain
They are under the folders SERIAL LOG, in the github
```

### Other Steps to Reproduce

I have only found issues with these specific devices, tried a dozen others with no issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-17:
@italocjs - Thank you for such detailed reporting and logs. It sounds like you have done a deep testing process.

IDF 4.4 SPP is a separated repository. Here, at the Arduino workgroup, we use IDF 4.4 for the Arduino Core 2.0.9, but we don't fix or develop any IDF code.

I'd suggest you to open this issue at IDF repository issues.

https://github.com/espressif/esp-idf/issues

italocjs, 2024-07-03:
update, i tested this again with latest master and the issue still happens, unfortunately i did not get an solution on ESP-IDF issues either, so i gave up classic bluetooth long ago and chosen to use nimBLE + nordic uart service instead. no connection issues on it.

There is something going on behind the scenes when SSP is enabled (simple secure pairing - enabled by default on menuconfig), i tried my best to diagnose but i could not find exactly what . The only thing that helped me to connect to them was disabling SSP (esp-idf required to access menuconfig).

I'll close it as i wont investigate this further, and hope someone in the future finds something.

closed 2024-07-03 by italocjs