espressif/arduino-esp32 #8205: Light Version of Bluetooth Serial only for receiving serial data

marschalla7, 2023-05-15:
### Related area

ESP32 BT

### Hardware specification

AZDelivery ESP32 NodeMCU Module WLAN WiFi Dev Kit C Development Board with CP2102 

### Is your feature request related to a problem?

I'm working with the ESP32 Dev Module. I want to  use the Bluetooth Serial Library but there is not enough program storage free on my ESP32 Dev Module.

### Describe the solution you'd like

I thought about a version of the library only for receiving serial data from antoher device.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-05-16:
You can try using IDF 4.4 calls directly instead of the Arduino SPP Layer (BT Serial).

Example:
https://github.com/espressif/esp-idf/tree/54576b7528b182256c027de86eb605a172bc2821/examples/bluetooth/bluedroid/classic_bt/bt_spp_initiator

IDF documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_spp.html

SuGlider, 2023-05-16:
>but there is not enough program storage free on my ESP32 Dev Module.

What is the size of the Flash? 4MB?
What is the size of the final binary?

Have you tried to change the partition to `Huge APP (3MB no OTA/1MB SPIFFs)` ?

[image]

marschalla7, 2023-05-17:
Hello,
I've changed the Partition Scheme to Huge APP not it's working, thanks for your help.
The final binary is 1511005 bytes.

closed 2023-05-17 by marschalla7