espressif/arduino-esp32 #8207: XIAO ESP32-C3: I2S takes a MASSIVE amount of time to come back to life

zackees, 2023-05-16:
### Board

XIAO ESP32-C3 with battery

### Device Description

https://www.amazon.com/dp/B0B94JZ2YF?psc=1&ref=ppx_yo2ov_dt_b_product_details

I'm using Platform io. Here is my ini file

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c3]
platform = espressif32@^6.2.0
board = seeed_xiao_esp32c3
framework = arduino

; change microcontroller
board_build.mcu = esp32c3
;upload_speed = 115200

; change WiFi firmware
board_build.variant = esp32c3

monitor_speed = 115200

board_build.f_cpu = 80000000L

build_flags = 
    -D CONFIG_PM_ENABLE
    -D CONFIG_PM_USE_RTC
    -D CONFIG_PM_LIGHTSLEEP_RTC_OSC_CAL_INTERVAL=8
```

I have a INMP441 mems mic:

https://www.amazon.com/AITRIP-Omnidirectional-Microphone-Precision-Interface/dp/B092HWW4RS/ref=sr_1_3?keywords=INMP441&sr=8-3

For the life of me I cannot get the IS2 peripheral to power back on quickly. I have to wait about 1/3rd of a second before the microphone stabilizes. I've tried so many different things: changing the clock signal. Changing the board frequency. Nothing seems to work to make the microphone better.

### Hardware Configuration

```C++
#ifndef _DEFS_H_
#define _DEFS_H_

#include <iostream>
using std::cout;
using std::endl;

#define MAX_ANALOG_READ 1023

#define LED_PIN 9  // Pin is held high on startup.
#define LED_PIN_IS_SINK true
#define LED_0 0  // Led channel name
#define LED_PWM_FREQ 2000
#define LED_PWM_RESOLUTION 14  // Max on ESP32-c3 XIOA

#define TIME_PWM_CYCLE_MS  3 // Flickers at 1ms
#define TIME_PWM_TRANSITION_MS 3  // 60 fps

[... 342 lines cut ...]
  // gpio_hold_dis(PIN_AUDIO_PWR);

  // i2s_start(I2S_NUM_0);
  i2s_audio_init();
  //delay(160);
  //audio_sample_t buffer[IS2_AUDIO_BUFFER_LEN] = {0};
  //uint32_t future_time = millis() + 180;
  //while (millis() < future_time) {
  //  read_raw_samples(buffer);
  //}
  //delay(180);
  //delay(POWER_ON_TIME_MS * 2);
  garbage_buffer_count = 14; // For some reason it takes a few buffers to get the audio going again.
}
```

### Debug Message

```plain
None. But the recorded sound levels are through the roof for the first few buffers then decrease.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-05-16:
Since you are using ESP-IDF's API, I suggest you file an issue in their repo. You can say that you are running on ESP-IDF v4.4.4

SuGlider, 2023-05-16:
My wild guess about it is that entering/returning from light sleep changes the source frequency of the I2S peripheral.
Thus, the first 300ms, the I2S peripheral may be running in the wrong frequency and it looks like reading garbage data up to the system starts back to run in the correct APB Freq.

Maybe (just a guess) if the I2S is set to use a Clock Source that is not APB dependent, it may not be affected by this change.

`      .use_apll = true,`  instead of false?

The new IDF 5.1 I2S driver documentation talk about using APLL... (but remember that Arduino 2.0.9 uses IDF 4.4.4).
This is just for reading and understanding my "wild guess".
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#i2s-clock

SuGlider, 2023-05-16:
Anyway.... this exactly same issue also happens with UART when returning from Light Sleep. The first bytes read are bad...
It is documented. Maybe.... it just needs to document that I2S has a similar behavior, like the one you are experimenting.

zackees, 2023-05-17:
I used `.use_apll = true` as well and used a fixed clock frequency. I also tried different CPU frequencies. It seems that 80 mhz is better and I'm able to minimize the number of 1k buffers I have to discard down to 10 (1024@44.1khz, single channel, 16 bit) to about 10.

It's a shame, because the MEMS mic performs so well otherwise. I tried putting a medium sized ceramic decoupling capacitor at the power pins of the MEMS, but this didn't do anything.

If I want to get proper light sleep I'll have to use an analog microphone and sample it with an ISR.

SuGlider, 2023-05-17:
I guess that those 300ms after returning from light sleep are very important for the project.
Why using an analog microphone would make it work?

Can you imagine a process/work around that fits better the needs of the project?
Stopping the I2S before going into light sleep and starting it after returning would work?

zackees, 2023-05-17:
It works because I can just keep the microphone on in a powered state while the CPU sleeps and then immediately start reading again after the processor wakes. While the `INMP441` mems mic has a ~2 mA operation current, the common `MAX9814` eletret mic uses just 50 uA.

This will allow me to microsleep the MCU in the main loop and immediately execute the next loop iteration without a delay.

SuGlider, 2023-05-17:
I'm not sure that ESP32 ADC will not also take some time to read accurate data... 
It uses SAR algorithm and it may also depend on the clock source.

A potential reference: https://github.com/espressif/esp-idf/issues/8287

SuGlider, 2023-05-17:
A possible alternative strategy to go low power and get MIC data could be to use the ULP to read the data... but the C3 has no ULP.
It could be possible with an S3 instead, using its RISC-V ULP.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/ulp-risc-v.html

zackees, 2023-05-17:
It's not possible to read the I2S under 80 mhz.

SuGlider, 2023-05-17:
> It's not possible to read the I2S under 80 mhz.

With some work to code it... RMT may do it. Not easy... but it may be possible to decode an I2S MIC signal...
Anyway, it may be a huge challenge and not work properly after a big effort to code it. Too risky, maybe.

zackees, 2023-05-17:
Sounds complicated. Reading from an analog condenser mic is well known and the components are cheap. I only need it for loudness and some fft. While the mems produced excellent audio, the condenser mic will work *well enough*.

zackees, 2023-05-17:
> > It's not possible to read the I2S under 80 mhz.
> 
> With some work to code it... RMT may do it. Not easy... but it may be possible to decode an I2S MIC signal... Anyway, it may be a huge challenge and not work properly after a big effort to code it. Too risky, maybe.

Do you know of any repositories that have a I2S implementation in software?

SuGlider, 2023-05-17:
> Do you know of any repositories that have an I2S implementation in software?

No. I think that there may be some MCU emulator software that does it, but it may not be applicable within an MCU.

craigyjp, 2023-07-04:
Can you confirm you are using a Seeed Studio ESP32-C3, I have been told that I2S is not supported as the pins do not exist, but it seems you are mapping them onto pins 2,3,4,5 for i2s

zackees, 2023-07-04:
Confirmed, I am using an ESP32-C3 Xiao

VojtechBartoska, 2023-08-18:
same here, can you please help with the triage @me-no-dev? Thanks

zackees, 2023-08-18:
Code repo:

https://github.com/zackees/xiao-inmp441-test

# Audio.h

```
#ifndef _AUDIOSENSOR_H_
#define _AUDIOSENSOR_H_

#include <stdint.h>
#include "defs.h"
#include <cstring>

#if AUDIO_BIT_RESOLUTION == 16 && AUDIO_CHANNELS == 1
typedef int16_t audio_sample_t;
#else
#error "Unsupported audio configuration"
#endif

typedef audio_sample_t audio_buffer_t[IS2_AUDIO_BUFFER_LEN];

struct audio_state_t {
  float db = 0.f;
  uint32_t updated = 0;
  audio_buffer_t samples = {0};
  audio_state_t(float _db, uint32_t _updated, const audio_buffer_t& samples) : db(_db), updated(_updated) {
    memcpy(this->samples, samples, sizeof(this->samples));
  }
};

void audio_init(bool wait_for_powerup = true);
void audio_shutdown();
// The last loudness_dB set is returned.
audio_state_t audio_update();

void audio_buffer_clear();
void audio_loudness_test();

float audio_loudness_dB();

// Power functions

void audio_enter_light_sleep();
void audio_exit_light_sleep();

#endif // _AUDIOSENSOR_H_
```

# Audio.cpp
```cpp
#include <iostream>

#include "audio.h"
#include "defs.h"
#include <Arduino.h>
#include <stdint.h>
#include <driver/i2s.h>
#include "ringbuffer.hpp"
#include "alloc.h"
#include "buffer.hpp"
#include "task.h"
#include <limits>
#include "time.h"
#include <stdio.h>
#include <atomic>

#include "alloc.h"

using namespace std;
[... 244 lines cut ...]
  gpio_hold_en(PIN_I2S_WS);
  gpio_hold_en(PIN_IS2_SD);
  gpio_hold_en(PIN_I2S_SCK);
  gpio_hold_en(PIN_AUDIO_PWR);
}

void audio_exit_light_sleep()
{
  gpio_hold_dis(PIN_I2S_WS);
  gpio_hold_dis(PIN_IS2_SD);
  gpio_hold_dis(PIN_I2S_SCK);
  // gpio_hold_dis(PIN_AUDIO_PWR);
  delay(160);
  // i2s_start(I2S_NUM_0);
  i2s_audio_init();
  // audio_sample_t buffer[IS2_AUDIO_BUFFER_LEN] = {0};
  // uint32_t future_time = millis() + 180;
  // while (millis() < future_time) {
  //   read_raw_samples(buffer);
  // }
  // delay(180);
  // delay(POWER_ON_TIME_MS * 2);
  //  garbage_buffer_count = 16;
}
```

zackees, 2023-08-19:
Sorry for the wall of text. Bug reproduction can be hard and I wanted to make sure it was as easy as possible with a working codebase that reproduces the issue clearly.

me-no-dev, 2023-08-22:
@zackees @VojtechBartoska given that all code is actually the ESP-IDF API, I will suggest that you contact the IDF team in their repository for help. Not sure which version ESP-IDF you are using, but 2.0.9/2.0.10 use IDF 4.4.4 and 2.0.11 use IDF 4.4.5

zackees, 2023-08-22:
This is my platform io .ini file:

```
[env:esp32c3]
platform = espressif32@^6.3.2
board = seeed_xiao_esp32c3
framework = arduino
```

What is the proper place to file this bug?

VojtechBartoska, 2023-10-23:
Hi, you can 3.0.0 alpha2 development release as well, there is new implementation of I2S library.

zackees, 2023-10-23:
Can I confirm this is getting fixed then?

VojtechBartoska, 2023-12-12:
Yes, there is new I2S library so you can try it our. Please let us know if it fixes your issue. Thanks

zackees, 2023-12-12:
link?

VojtechBartoska, 2023-12-13:
It is the part of https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3 release. Related Pull Request is here: https://github.com/espressif/arduino-esp32/pull/8714

zackees, 2023-12-15:
Thank you.

Is this available on platform io? The following  platformio.ini file fails to build:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c3]
platform = https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
board = seeed_xiao_esp32c3
framework = arduino

upload_speed = 921600

; change microcontroller
board_build.mcu = esp32c3
;upload_speed = 115200

; change MCU frequency
; board_build.f_cpu = 160000000L

; change WiFi firmware
board_build.variant = esp32c3

monitor_filters = 
	default
	esp32_exception_decoder

monitor_speed = 115200

board_build.f_cpu = 80000000L

build_flags = 
    -D CONFIG_PM_ENABLE
    -D CONFIG_PM_USE_RTC
    -D CONFIG_PM_LIGHTSLEEP_RTC_OSC_CAL_INTERVAL=8
    -D FIXED_POINT=16
    -D KISS_FFT_USE_ESP32_PSRAM=0
```

Do you know what the solution is? I'll test this out as soon as I have access.

zackees, 2024-01-25:
I'm trying to test this fix but I have at this point re-write my app into the format used by the idf.py tool first.

If anyone else is having trouble installing `idf.py` in a cross-platform way then I recommend a tool I made that make the process easy and works under Windows/Linux/Mac.

https://github.com/zackees/idf-install

However, it looks like the expressif arduino package is about to land with support for 5.1, which from the changelist includes this fix above. However currently I can't test it out because of a missing `platform.json`.

Here's the bug report:
https://github.com/espressif/arduino-esp32/issues/9175

If this turns out to be an easy fix then I'd love to test out the I2S fix before it goes live.

zackees, 2024-01-27:
Good news, there is an installable platformio version of IDF v5.1 with the I2S fix.

get it with this pull request. I verified it my self and now I have to port ledc, I2C etc 

https://github.com/platformio/platform-espressif32/pull/1281

zackees, 2024-01-28:
IDF toolchain for platformio just became available thanks to a pull request.

Add these to your ini file:

'''
platform = https://github.com/sgryphon/platform-espressif32.git#sgryphon/add-esp32-arduino-libs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
'''

zackees, 2024-01-30:
Okay so follow up. I installed the unofficial IDF 5.1 libs for platform IO and then went through each device driver and ported it to the the new v5.1 API.

LEDC ported fine. But for the life of me, I cannot get the IS2 to work with the INMP411 with the new API. It works FINE with the 4.x legacy IS2 API.

At first I did a straight port. The data signal is delayed by one frame as defined by the Philips standard, which apparently was the default for the legacy API. The straight port did not work. So I spent about another 4-6 hours diving deep, look at the source code and trying every combination I could think of of clock, pin modes, etc.

Now that I've gone deep, I have absolutely no idea how the legacy drivers were even working. It appears that every copy and paste of the 4.x IS2 api appears wrong. No joke. INMP441 works with 16 bit audio, but it's clear it using 24 bit audio in a 32 bit WS frame, delayed by one clock cycle.

The 4.x api allows me to specify 16 bit, which makes no sense at all since this the data sheet of the INMP441 clearly says it's a 24 bit data frame. I've tried putting the 5.1 api into I2S master mode, slave mode. My INMP441 is set to be right channel only. I saw that in the 4.x lib there was a time where the L/R channels were swapped. So I've tried both but neither work.

## BUG?

What's really strange is that when i query the info state of the channel like this:

```C++
err = i2s_channel_init_std_mode(rx_chan, &i2s_std_cfg_rx);
ESP_ERROR_CHECK(err);
i2s_chan_info_t info;
err = i2s_channel_get_info(rx_chan, &info);
ESP_ERROR_CHECK(err);
delay(1000);
Serial.printf("I2S channel info: %d, %d, %d, %d, %d\n", info.id, info.role, info.dir, info.mode, info.pair_chan);
delay(500);
```
    
 I see that `info.dir` is `1`

So the direction is in TX mode instead of RX mode. Not sure if this is significant.

I'm going to try and get a repro case in a public github. For now, I just have this brain dump of the halfway ported I2S `i2s_device.cpp` file.

```C++

#include <iostream>

#include "audio.h"
#include "defs.h"
// #include <driver/i2s.h>
#include <driver/i2s_std.h>
#include "i2s_device.h"
#include "driver/gpio.h"
namespace
{
  bool g_is_initialized = false;
  static_assert(AUDIO_BIT_RESOLUTION == 16, "Only 16 bit resolution is supported");
  static_assert(AUDIO_CHANNELS == 1, "Only 1 channel is supported");
  static_assert(sizeof(audio_sample_t) == 2, "audio_sample_t must be 16 bit");
  i2s_chan_handle_t     rx_chan = NULL;
  i2s_chan_handle_t tx_chan = NULL;
  //i2s_chan_config_t     i2s_chan_cfg_rx = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER); // stores I2S channel values
  i2s_chan_config_t  i2s_chan_cfg_rx = {
    .id = I2S_NUM_0,
    .role = I2S_ROLE_SLAVE,
    .dma_desc_num = 6,
    .dma_frame_num = 240,
    .auto_clear = false,
  };
  i2s_std_config_t      i2s_std_cfg_rx = {
    //.clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(AUDIO_SAMPLE_RATE),
    .clk_cfg = {
        .sample_rate_hz = 44100ul,
        .clk_src = i2s_clock_src_t(I2S_CLK_SRC_PLL_160M),
        .mclk_multiple = I2S_MCLK_MULTIPLE_384,
    },

    //.slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_24BIT, I2S_SLOT_MODE_MONO),
    /*
#define I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(bits_per_sample, mono_or_stereo) { \
    .data_bit_width = bits_per_sample, \
    .slot_bit_width = I2S_SLOT_BIT_WIDTH_AUTO, \
[... 186 lines cut ...]
      // cout << "Bytes read: " << bytes_read << endl;
      const size_t count = bytes_read / sizeof(audio_sample_t);
      return count;
    }
  }
  return 0;
  #else
  size_t bytes_read = 0;
  esp_err_t err = i2s_channel_read(rx_chan, buffer, sizeof(buffer), &bytes_read, 0);
  if (err == ESP_OK)
  {
    if (bytes_read > 0)
    {
      // cout << "Bytes read: " << bytes_read << endl;
      const size_t count = bytes_read / sizeof(audio_sample_t);
      return count;
    }
  }
  if (err != ESP_ERR_TIMEOUT) {
    ESP_ERROR_CHECK(err);
  }
  return 0;
  #endif
}
```

zackees, 2024-02-07:
Okay, I've gone ahead and created a minimal repo that reproduces this bug using the alpha release 3.0.0 5.1 libraries that were unofficially released.

For the life of me, I cannot get the INMP441 to work in the new v5.1 drivers, but the could in the v4.4 drivers. The legacy i2s drivers work with the new v5.1 alpha release 3.0.0, fyi.

https://github.com/zackees/esp32-v51-inmp441

zackees, 2024-02-09:
Okay, here's the code to get IS2 to work in v5.1.

```C++

/*
Uses the new idf 5.1 i2s driver..
*/

#include <iostream>
#include <Arduino.h>
#include "defs.h"
#include <driver/i2s_std.h>
#include "i2s_device.h"
#include "driver/gpio.h"

#define SD_PULLDOWN_ENABLED 0

namespace
{
  static_assert(AUDIO_BIT_RESOLUTION == 24, "Only 16 bit resolution is supported");
  static_assert(AUDIO_CHANNELS == 1, "Only 1 channel is supported");
  static_assert(sizeof(audio_sample_t) == 4, "audio_sample_t must be 32 bit");

  struct I2SContext
  {
    i2s_chan_handle_t rx_chan;
    i2s_chan_config_t i2s_chan_cfg_rx;
    i2s_std_config_t i2s_std_cfg_rx;
  };

  I2SContext g_i2s_context;
  I2SContext make_inmp441_context() {
    I2SContext ctx;
    i2s_chan_handle_t rx_chan = NULL;
    i2s_chan_handle_t tx_chan = NULL;
    i2s_chan_config_t i2s_chan_cfg_rx = {
        .id = I2S_NUM_0,
        .role = I2S_ROLE_MASTER,
        .dma_desc_num = AUDIO_DMA_BUFFER_COUNT,
        .dma_frame_num = IS2_AUDIO_BUFFER_LEN,
        .auto_clear = false,
    };
    i2s_std_config_t rx_std_cfg = {
        .clk_cfg  = I2S_STD_CLK_DEFAULT_CONFIG(AUDIO_CHANNEL_SAMPLE_RATE),
        .slot_cfg = {
                .data_bit_width = I2S_DATA_BIT_WIDTH_24BIT,
                .slot_bit_width = I2S_SLOT_BIT_WIDTH_32BIT,
                .slot_mode = I2S_SLOT_MODE_MONO,
                .slot_mask = I2S_STD_SLOT_RIGHT,
                .ws_width = 32,
                .ws_pol = false,
                .bit_shift = true,
                .left_align = true,
                .big_endian = false,
                .bit_order_lsb = false,
        },
        .gpio_cfg = {
                .mclk = I2S_GPIO_UNUSED,
                .bclk = PIN_I2S_SCK,
                .ws   = PIN_I2S_WS,
                .dout = I2S_GPIO_UNUSED,
                .din  = PIN_IS2_SD,
                .invert_flags = {
                        .mclk_inv = false,
                        .bclk_inv = true,
                        .ws_inv   = false,
                },
        },
    };
    ctx = {rx_chan, i2s_chan_cfg_rx, rx_std_cfg};
    return ctx;
  }

  void init_i2s_pins()
  {
    //g_i2s_context = get_i2s_context();
    g_i2s_context = make_inmp441_context();
    esp_err_t err = i2s_new_channel(&g_i2s_context.i2s_chan_cfg_rx, NULL, &g_i2s_context.rx_chan);
    ESP_ERROR_CHECK(err);
    err = i2s_channel_init_std_mode(g_i2s_context.rx_chan, &g_i2s_context.i2s_std_cfg_rx);
    ESP_ERROR_CHECK(err);
    i2s_chan_info_t info;
    err = i2s_channel_get_info(g_i2s_context.rx_chan, &info);
    ESP_ERROR_CHECK(err);
    delay(1000);
    Serial.printf("I2S channel info: %d, %d, %d, %d, %d\n", info.id, info.role, info.dir, info.mode, info.pair_chan);
    delay(500);

    ESP_ERROR_CHECK(err);
    err = i2s_channel_enable(g_i2s_context.rx_chan);
    ESP_ERROR_CHECK(err);

    gpio_num_t gpio_pin = PIN_IS2_SD; // Replace with your GPIO pin

// Data sheet recommends a pulldown on the input pin when connecting stero microphones.
// However this may also be necessary in mono mode because of the tri state nature of the
// I2S bus.
// Configure the pin as an input
#if SD_PULLDOWN_ENABLED
    init_pulldown_datapin();
#endif
  }

  void init_pulldown_datapin()
  {
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_INTR_DISABLE;       // Disable GPIO interrupts
    io_conf.mode = GPIO_MODE_INPUT;              // Set as Input mode
    io_conf.pin_bit_mask = (1ULL << PIN_IS2_SD); // Bit mask of the pin
    io_conf.pull_down_en = GPIO_PULLDOWN_ENABLE; // Enable pull-down resistor
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;    // Disable pull-up resistor
    gpio_config(&io_conf);
  }

} // namespace

void i2s_audio_init()
{
  pinMode(PIN_AUDIO_PWR, OUTPUT);
  digitalWrite(PIN_AUDIO_PWR, HIGH); // Power on the IS2 microphone.
  init_i2s_pins();
}

// esp_err_t i2s_channel_read(i2s_chan_handle_t handle, void *dest, size_t size, size_t *bytes_read, uint32_t timeout_ms)

void i2s_audio_shutdown()
{
  // i2s_stop(I2S_NUM_0);
  // i2s_driver_uninstall(I2S_NUM_0);
  i2s_del_channel(g_i2s_context.rx_chan);
}

void i2s_audio_enter_light_sleep()
{
  // digitalWrite(PIN_AUDIO_PWR, LOW); // Power off the IS2 microphone.
  // hold pin engaged
  digitalWrite(PIN_AUDIO_PWR, HIGH); // Power on the IS2 microphone.
  gpio_hold_en(PIN_AUDIO_PWR);
  // i2s_stop(I2S_NUM_0);
  // i2s_driver_uninstall(I2S_NUM_0);
}

void i2s_audio_exit_light_sleep()
{
  // digitalWrite(PIN_AUDIO_PWR, HIGH); // Power on the IS2 microphone.

  // i2s_start(I2S_NUM_0);
  // i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
}

size_t i2s_read_raw_samples(audio_sample_t (&buffer)[IS2_AUDIO_BUFFER_LEN])
{

  size_t bytes_read = 0;
  esp_err_t err = i2s_channel_read(g_i2s_context.rx_chan, buffer, sizeof(buffer), &bytes_read, 0);
  if (err == ESP_OK)
  {
    if (bytes_read > 0)
    {
      // cout << "Bytes read: " << bytes_read << endl;
      const size_t count = bytes_read / sizeof(audio_sample_t);
      return count;
    }
  }
  if (err != ESP_ERR_TIMEOUT)
  {
    ESP_ERROR_CHECK(err);
  }
  return 0;
}
```

zackees, 2024-02-12:
# Update4: Oscilloscope time!

  * ESP32 shuts down the clock and word select in light sleep mode
    * I suspect this puts the microphone in a low power mode
  * The clock/ws signal resumes
  * The microphone outputs noise (decreasing) for ~100 ms

Should the ESP32 keep the clock/ws pin alive and feed into the DMA buffer? Or at least strobe them and discard the data. I'ml going to look into feeding in a clock signal during sleep and see if this solves the issue.

zackees, 2024-02-12:
So it looks like DMA is NOT enabled during light sleep. Which is the reason I2S stops. My next trick will be to try and attach an LEDC driver in low bit mode and try to strobe the clk signal to keep the INMP441 from going into sleep mode.

Here is some code that will configure a pin to use LEDC during light sleep:

https://github.com/espressif/arduino-esp32/issues/3534

zackees, 2024-02-13:
So confirmed working via the oscilloscope in 5.1 alpha 3.0.0.

Closing this bug. It's the INMP441 microphone that is the culprit, which might be auto-sleeping when the I2S halts it's clock during light sleep.

A potential work-around is to use ledc on the RTC clock to create a clock signal that keeps the INMP441 microphone from going to sleep. I'm having difficulty for I2S and LEDC to share a pin so I think I'm just going to tie the pins together and see how it goes.

closed 2024-02-13 by zackees

zackees, 2024-02-27:
## Okay update: I've mostly got it working.

There are two issue here as I went deep into this issue.

  1. In v5.1 and v4.4, stopping an I2S channel and resuming it DOES cause a noise for 250ms. It seems to do this without even sleeping. I'm going to confirm this with deeper analysis and to see if disable/resume with no delay still causes the issue. I was disabling the channel and resuming it on wake from light sleep so this makes sense. The solution is to NOT do anything to the I2S channel before or after sleep. Simply let the whole system go to light sleep and resume on it's own.
  2. Driving a Psuedo I2S signal via the LEDC library works very well. I have to drive both the SCK and WS pins on the INMP441. To do this I have to allocate one of the neighbor pins to these pseudo signals and tie it to the real pin. When the chip wakes up I immediately stop the LEDC driver and then put it into INPUT mode. Simply driving it with psuedo SCK pin does not work. Both pins have to be emulated and only worked when I closely matched the real signals via an oscilloscope.
 3. The I2S must not stop/delete the channel before sleep. Nothing special must be done after awake from light sleep. Upon wake, the I2S DMA buffer will contain some garbage. For my 512 sample buffers this was a big problem, but I found that I could reduce the DMA buffer size to 16 samples and this contains the noise to just a few small samples which I can discard quickly. Yes, it does this for different sampling speeds. Reducing the INMP441 clock (sample rate) has no difference.
 
 Yes, I'm locking the APB frequency on setup().
 
I suspect still that it's the INMP441 that's causing the issue, but I can't rule out the esp32 driver yet.

## Next steps:

  * Deeper analysis with the oscilloscope: what happens when the I2S starts for the first time? I have to trigger catch the first about 250 ms of data on first WS click and manually inspect the SCK/WS and data-in signals to determine whether the ESP32C3 is indeed generating the right signals and whether the ESP32C3 is generate good looking data during a quite period.
  
  
## Other notes:

Wow, it's really hard to get that LEDC driver to operate in light sleep mode! The documentation is missing the following requirements:

  * `ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC8M, ESP_PD_OPTION_ON));`
  * `ESP_ERROR_CHECK(gpio_sleep_sel_dis(PIN_LEDC)); // Needed for light sleep.`
  
 I've posted my working code sample [here](https://github.com/espressif/arduino-esp32/issues/3534#issuecomment-1965779153)