espressif/arduino-esp32 #8210: The last part of WiFiMulti.run(), is spurious or incorrect

mahboud, 2023-05-16:
### Board

ESP32 Dev Kit

### Device Description

Any ESP32 device

### Hardware Configuration

Not important.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

MacOS 13.3.1

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

libraries/WiFi/src/WiFiMulti.cpp

This is the last part of WiFiMulti.run():

```
    } else {
        // start scan
        log_d("[WIFI] delete old wifi config...");
        WiFi.disconnect();

        log_d("[WIFI] start scan");
        // scan wifi async mode
        WiFi.scanNetworks(true);
    }

    return status;
}
```

The results of that last scan are never placed anywhere, and this function returns. If the function is called again, these results are not used, and are merely discarded. 

Basically, it seems to me that the async call to scanNetworks is doing nothing here. Please tell me if I am wrong.

### Sketch

```cpp
/*
 *  This sketch trys to Connect to the best AP based on a given list
 *
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

    Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop()
{
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```

### Debug Message

```plain
No debug messages. Just looking a the code, it looks like the expected behavior that the last scan will lead to useful actions, such as the selection of a strong SSID.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-16:
Considering that `WiFi.scanNetworks()` can return a zero/positive number (of found WiFi SSIDs) or a negative value as below:

#define WIFI_SCAN_RUNNING   (-1)
#define WIFI_SCAN_FAILED        (-2)

In Summary this code executes like this:

``` cpp
    // Line 103 of libraries/WiFi/src/WiFiMulti.cpp

    scanResult = WiFi.scanNetworks();
    if(scanResult == WIFI_SCAN_RUNNING) {  // returned -1 (WIFI SCAN  is still RUNNING...)
        // scan is running
        return WL_NO_SSID_AVAIL;
    } else if(scanResult >= 0) {  // <===== it can return -2 when there is a WiFi scanning problem
      // Code for dealing with WiFi scan Results...
      // Lines 108 to 192
    } else {                                // <===== at this point we had a WiFi Scan Failure!
        // start scan
        log_d("[WIFI] delete old wifi config...");
        WiFi.disconnect();
                                             // <====== Start WiFi again and try to scan it again...
        log_d("[WIFI] start scan");
        // scan wifi async mode
        WiFi.scanNetworks(true);
    }

    return status;
}
```

mahboud, 2023-05-17:
So, you're saying that when an error happens, then there is an async scan triggered. Then maybe when the next time `WiFiMulti.run()` happens from the `loop()` then the results are ready and waiting?

The problem is that each time you do a `scanNetworks()`, the last results are cleared, no?

```
int16_t WiFiScanClass::scanNetworks(bool async,
                                    bool show_hidden,
                                    bool passive,
                                    uint32_t max_ms_per_chan,
                                    uint8_t channel,
                                    const char* ssid,
                                    const uint8_t* bssid) {
  if (WiFiGenericClass::getStatusBits() & WIFI_SCANNING_BIT) {
    return WIFI_SCAN_RUNNING;
  }

  WiFiScanClass::_scanTimeout = max_ms_per_chan * 20;
  WiFiScanClass::_scanAsync = async;

  WiFi.enableSTA(true);

  scanDelete();  // <===== results are cleared
  ```

That means that the last `WiFi.scanNetworks(true);` isn't doing anything useful.

SuGlider, 2023-05-17:
WiFi works in separated tasks and ISRs. It depends on FreeRTOS to work.
Some processes work in the "background". In this case, this is the idea.

mahboud, 2023-05-17:
Even if in the background or in parallel, if you only call `WiFiMulti.run()` once and it happens to execute that second `WiFi.scanNetworks(true);` it doesn't do anything, right? That second call will asynchronously run, but the results will sit waiting to be pulled out. Are we in agreement about this? 

Now let's move to the case where you call `run()` once, and it happens to run the last `scanNetworks()` async. 

That's where the second or next calls to `WiFiMulti.run()` are supposed to come in and unload the scanned info from that async scan prior. However, if we call `WiFi.scanNetworks();` again, it goes and deletes the results of any previous scan - and given that we're not locking any other threads or using volatiles, we can't assume that in between ending a scan, and starting the next one, that we just happened to read the info out before it was deleted before the next scan. 

To get around it, we need to not call `WiFi.scanNetworks();` until we have assured that there is no data in there waiting for us to read it. We can do this by calling `int n = WiFi.scanComplete();` which should get us the results that are waiting to be read. If `n` is 0 then we can do another `WiFi.scanNetworks();`; if `n` is positive, then we have results to process.

SuGlider, 2023-05-17:
OK, I see you point. I agree that the last part of WiFiMulti.run() could be removed from the code.

mahboud, 2023-05-17:
I'm going to do a PR soon as a possible consideration. Without changing the way things work, it might be a way to make this do what it is supposed to do.

SuGlider, 2023-05-24:
@mahboud - closing it. Feel free to reopen it, whenever necessary.

closed 2023-05-24 by SuGlider