espressif/arduino-esp32 #8224: HWCDC::end() does not set tx_lock to NULL, causing crashes

TD-er, 2023-05-18:
### Board

ESP32-C3 / ESP32-S3

### Device Description

Any ESP32-C3 or -S3 using HWCDC

### Hardware Configuration

Any ESP32-C3 or -S3 board using HWCDC

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

HWCDC::end() does not reset the semaphore.
So after a call to `end()` and a new call to `begin()` the `tx_lock` semaphore is invalid.
Thus trying to write something to that port again will cause a crash.

https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/cores/esp32/HWCDC.cpp#L190-L205

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-18:
Thanks @TD-er - I'll check it and fix it.

SuGlider, 2023-05-24:
Thanks again @TD-er - PR #8247 will fix it.

PR #8247 (2023-05-31): Fixes HWCDC::end()
## Description of Change

If we call `serial.end()` followed by `serial.begin` using JTAG/CDC interface, it will never initialize the semaphore `tx_lock` and it will reset the board when data is written.
There is a missing `tx_lock = NULL;` in the `HWCDC::end()` function.

## Tests scenarios
Tested with C3 and S3.

## Related links
Fixes #8224
closed 2023-05-31 by me-no-dev