espressif/arduino-esp32 #8225: Std::round has not been declared

yeesk23, 2023-05-18:
### Board

ESP Wrover Module

### Device Description

Arduino UNO

### Hardware Configuration

.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have run through a very basic of examples of esp32 camera which is CameraWebServer. However after following exactly youtube steps by steps, i get an error that mention that Std::Round has not been declared. What am i suppose to do?

### Sketch

```cpp
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
//
//            You must select partition scheme from the board menu that has at least 3MB APP space.
//            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15 
//            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well

// ===================
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
// ** Espressif Internal Boards **
#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD

#include "camera_pins.h"

// ===========================
// Enter your WiFi credentials
// ===========================
const char* ssid = "Redmi Note 10 Pro";
const char* password = "123456789";

void startCameraServer();
void setupLedFlash(int pin);

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_UXGA;
  config.pixel_format = PIXFORMAT_JPEG; // for streaming
  //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){
      config.jpeg_quality = 10;
      config.fb_count = 2;
      config.grab_mode = CAMERA_GRAB_LATEST;
    } else {
      // Limit the frame size when PSRAM is not available
      config.frame_size = FRAMESIZE_SVGA;
      config.fb_location = CAMERA_FB_IN_DRAM;
    }
  } else {
    // Best option for face detection/recognition
    config.frame_size = FRAMESIZE_240X240;
#if CONFIG_IDF_TARGET_ESP32S3
    config.fb_count = 2;
#endif
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1); // flip it back
    s->set_brightness(s, 1); // up the brightness just a bit
    s->set_saturation(s, -2); // lower the saturation
  }
  // drop down frame size for higher initial frame rate
  if(config.pixel_format == PIXFORMAT_JPEG){
    s->set_framesize(s, FRAMESIZE_QVGA);
  }

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

#if defined(CAMERA_MODEL_ESP32S3_EYE)
  s->set_vflip(s, 1);
#endif

// Setup LED FLash if LED pin is defined in camera_pins.h
#if defined(LED_GPIO_NUM)
  setupLedFlash(LED_GPIO_NUM);
#endif

  WiFi.begin(ssid, password);
  WiFi.setSleep(false);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // Do nothing. Everything is done in another task by the web server
  delay(10000);
}
```

### Debug Message

```plain
In file included from C:\Users\User\AppData\Local\Temp\arduino\sketches\E6114CD6F7299D255006F3CD4C244B8F\sketch\CameraWebServer.ino.cpp:1:0:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:194:12: error: 'std::round' has not been declared
 using std::round;
            ^

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-05-18:
@SuGlider any clues on the random? @yeesk23 `CAMERA_MODEL_AI_THINKER` is the camera config you need to uncomment

SuGlider, 2023-05-18:
>C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:194:12: error: 'std::round' has not been declared
 using std::round;
            ^
>
>exit status 1

`std::round()` is fixed a while ago in https://github.com/espressif/arduino-esp32/pull/6726 - since Core 2.0.4
I have compiled the same camera example with Arduino IDE 1.8.15 and Arduino Core 2.0.9.
I got not errors.

@yeesk23 - I think that the Arduino Development Environment is corrupted.
Please uninstall ESP32 Arduino and update it to the Arduino Core to 2.0.9 and this issue is solved.

SuGlider, 2023-05-18:
I just installed the Arduino Core 2.0.7 and compiled the CameraWebServer.ino example.
I got no errors...

SuGlider, 2023-05-18:
@yeesk23 - what is the Arduino IDE that you are using?

SuGlider, 2023-05-19:
> @SuGlider any clues on the random? @yeesk23 `CAMERA_MODEL_AI_THINKER` is the camera config you need to uncomment

@me-no-dev - The error message is about `Arduino.h` with `using std::round;` -- doesn't make sense, otherwise we would have problems with CI.

yeesk23, 2023-05-19:
> @SuGlider any clues on the random? @yeesk23 `CAMERA_MODEL_AI_THINKER` is the camera config you need to uncomment

Ya i have select this camera model but it's still the same issue happening

yeesk23, 2023-05-19:
> I've just installed the Arduino Core 2.0.7 and compiled the CameraWebServer.ino example. I've got no errors...

im using arduino ide 2.1.0

yeesk23, 2023-05-19:
> CI

is there any suggestion to try on?

yeesk23, 2023-05-19:
What is meant by arduino core ya?

yeesk23, 2023-05-19:
[image]
i face this error when i want to update my version

SuGlider, 2023-05-19:
I see... You use Core 2.0.7 and can't update to 2.0.9, using the Arduino IDE 2.1.0. 
I agree that it is a coincidence of version numbering...

Let me try to replicate the issue.

SuGlider, 2023-05-24:
> [image] i face this error when i want to update my version

This screenshot doesn't look like from IDE 2.1.0... Are you sure that you are not using 1.8.x IDE version?
This error may be related to manually removing the `esp32` folder from ARDUINO15 installation folder.
Go to ARDUINO15 folder and verify if there is any problem there such as 2 different esp32 versions still installed etc.
https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder

IDE 2.1.0 board manager looks like this:
[image]

SuGlider, 2023-05-24:
Anyway, I could not reproduce the issue and I could upgade from 2.0.7 to 2.0.9 with no errors, using IDE 2.1.0.

podaen, 2023-06-10:
In Arduino.h

```
// Use float-compatible stl abs() and round(), we don't use Arduino macros to avoid issues with the C++ libraries
using std::abs;
using std::isinf;
using std::isnan;
using std::max;
using std::min;
using std::round;
```

C99

 double round  (double x);      float roundf (float x);long double roundl (long double x);

 C11

 double round (double x);      float round (float x);long double round (long double x);     double round (T x);

So in the current C99 using a float is roundf. Why not using C11 in platform.txt?

SuGlider, 2023-07-20:
>Why not using C11 in platform.txt?

@podaen - The Arduino Core currently uses C17, thus, it is even better.

podaen, 2023-07-20:
I use the reference in Arduino-ESP platform.txt. 

https://github.com/espressif/arduino-esp32/blob/21b88659b9ded3fcc1082f23f498bc2a04cd4f1b/platform.txt#L40C1-L41

You probably mean that the Arduino IDE uses C17...

SuGlider, 2023-07-22:
Sorry, my bad, actually Arduino 2.0.x uses C99 and C++11.
I think that it is because of backward compability with older IDF 4.x source code.

But for the Arduino Core 3.0.0 + IDF 5.1 we will use C17 and C++2b (C++23 with extensions)
https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32/flags/c_flags
https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32/flags/cpp_flags

podaen, 2023-07-26:
@me-no-dev When can we aspect that?

VojtechBartoska, 2023-12-12:
Hi, this should be covered in current 3.0.0-Alpha releases, closing the ticket.If needed, you can reopen.

closed 2023-12-12 by VojtechBartoska