espressif/arduino-esp32 #8226: OTAWebUpdater.ino: The login page is really insecure, username and password are visible in the page source code

rtek1000, 2023-05-19:
### Related area

WIFI, OTA Web Updater

### Hardware specification

ESP32

### Is your feature request related to a problem?

Login page is insecure, username and password are visible in the page source code:

line 48:

`"if(form.userid.value=='admin' && form.pwd.value=='admin')"`

https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino

I understand it's just an example, but certainly a lot of people have used it.

### Describe the solution you'd like

I found this code which could perhaps serve as the basis for an update to the OTAWebUpdater.ino example.

The username and password are only on the server side (ESP32),
line 49:
`    if (server.arg("USERNAME") == "admin" &&  server.arg("PASSWORD") == "admin" ){`

https://github.com/zhouhan0126/WebServer-esp32/blob/master/examples/SimpleAuthentification/SimpleAuthentification.ino

As an example, help text is displayed as well, which should be ignored for the sss example,
line 60:
`  String content = "<html><body><form action='/login' method='POST'>To log in, please use : admin/admin<br>";`

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2023-05-19:
Thanks for reporting the issue @rtek1000.

@PilnyTomas Can you please help with triage? Thanks

surya0701, 2023-07-13:
I want https OTA ESP32 insecure method instead of using server certificate

PilnyTomas, 2023-08-11:
Thanks for the detailed request. In my opinion, if the Espressif teams were to implement this it would take a long this since there are tasks with higher priority.
Espressif code is open source and anyone is welcome to contribute. If you are comfortable with it, I would like to ask you to implement this and open a PR.

closed 2025-01-16 by Parsaabasi