espressif/arduino-esp32 #8227: bad bad ADC! error: invalid conversion from 'int' to 'adc_unit_t'

josef2600, 2023-05-19:
### Board

esp32-c3 dev

### Device Description

Board: "ESP32C3 Dev Module, Disabled, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 256000, Info, Enabled"

### Hardware Configuration

Board: "ESP32C3 Dev Module, Disabled, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 256000, Info, Enabled"
with or without code! . compiling issue. 

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

win 10

### Flash frequency

80 or 40

### PSRAM enabled

no

### Upload speed

256000

### Description

funny thing. i was trying to make a calibration code for ADC1. pay attention , i was trying, no code yet. i was just trying to add library's to sketch to see what library is in Arduino and what is not. while adding only headers, i heat this wall and was amazed by how wonderful Arduino is! try it and be amaze! just add `#include "hal/adc_hal.h"` to anything! and Arduino  make a ton of error for its own library! amazing!

### Sketch

```cpp
//----ADC include--------
#include <string.h>
#include <stdint.h>
#include <stdbool.h>
#include <sys/cdefs.h>
#include <esp_types.h>
#include <sdkconfig.h>
#include <esp_err.h>
#include <esp_log.h>
#include <esp_check.h>
#include <esp_heap_caps.h>
#include <hal/adc_types.h>
#include <esp_bit_defs.h>
#include "esp_pm.h"
#include "freertos/FreeRTOS.h"
#include "freertos/ringbuf.h"
#include "hal/adc_types.h"
#include "hal/adc_hal.h"
```

### Debug Message

```plain
Board: "ESP32C3 Dev Module, Disabled, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 256000, Info, Enabled"

In file included from C:\Users\Josef\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/adc_hal.h:17,
[... 116 lines cut ...]

C:\Users\Josef\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/adc_ll.h:358:75: error: invalid conversion from 'int' to 'adc_unit_t' [-fpermissive]

         filter->adc_unit = (APB_SARADC.filter_ctrl0.filter_channel1 >> 3) & 0x1;

                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~

C:\Users\Josef\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/adc_ll.h:359:67: error: invalid conversion from 'int' to 'adc_channel_t' [-fpermissive]

         filter->channel = APB_SARADC.filter_ctrl0.filter_channel1 & 0x7;

                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~

C:\Users\Josef\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/adc_ll.h:360:48: error: invalid conversion from 'long unsigned int' to 'adc_digi_filter_mode_t' [-fpermissive]

         filter->mode = APB_SARADC.filter_ctrl1.filter_factor1;

                        ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~

```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-05-19:
@P-R-O-C-H-Y Please help with triage, thanks

lbernstone, 2023-05-19:
It is a C library from ESP-IDF, not actually part of arduino.  You probably actually want to use driver/adc.h rather than the hal, but your choice. Encapsulate it in an extern:
```
extern "C" {
#include "hal/adc_hal.h"
}

me-no-dev, 2023-05-19:
@VojtechBartoska as @lbernstone said, this is NOT part of Arduino and is in fact a low-level HAL of ESP-IDF. There is nothing we can do here (and @lbernstone gave the basic "fix") so I am closing this

closed 2023-05-19 by me-no-dev