espressif/arduino-esp32 #8229: ESP32-CAM-MB rebooting

777Raim, 2023-05-19:
### Board

ESP32-CAM-MB

### Device Description

ESP32-CAM on an ESP32-CAM-MB

### Hardware Configuration

No

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Fedora 34

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have 2 similar `ESP32-CAM` modules. Try to upload the `CameraWebServer` from examples. First board works great, but second always rebooting.

I use this settings:

[image]

### Sketch

```cpp
`CameraWebServer` from examples
```

### Debug Message

```plain
15:56:12.425 -> 
15:56:12.425 -> assert failed: do_core_init startup.c:326 (flash_ret == ESP_OK)
15:56:12.425 -> 
15:56:12.425 -> 
15:56:12.425 -> Backtrace: 0x40083acd:0x3ffe3aa0 0x40091089:0x3ffe3ac0 0x40097265:0x3ffe3ae0 0x400e456a:0x3ffe3c10 0x400831d2:0x3ffe3c40 0x40079276:0x3ffe3c90 |<-CORRUPTED
15:56:12.458 -> 
15:56:12.458 -> 
15:56:12.458 -> 
15:56:12.458 -> 
15:56:12.458 -> ELF file SHA256: 4c6bfd5b00ffa84a
15:56:12.458 -> 
15:56:12.458 -> E (545) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
15:56:12.458 -> Rebooting...
15:56:12.458 -> ets Jun  8 2016 00:22:57
15:56:12.458 -> 
15:56:12.458 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:56:12.458 -> configsip: 0, SPIWP:0xee
15:56:12.491 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:56:12.491 -> mode:DIO, clock div:1
15:56:12.491 -> load:0x3fff0030,len:1344
15:56:12.491 -> load:0x40078000,len:13924
15:56:12.491 -> ho 0 tail 12 room 4
15:56:12.491 -> load:0x40080400,len:3600
15:56:12.491 -> entry 0x400805f0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-21:
Known issue with core flash dump. Try "erase all flash" and reprogram.

777Raim, 2023-05-21:
How to do this `erase all flash`?

Tried `python -m esptool --chip esp32 erase_flash` then upload again: same issue.

SuGlider, 2023-05-25:
@777Raim - It is possible to erase the flash before uploading the sketch by just enabling the option in the IDE Menu:

[image]

SuGlider, 2023-05-25:
Please let us know if the issue is solved by erasing the flash, in order to close this issue. Thanks!

VojtechBartoska, 2023-08-18:
closing as expired

closed 2023-08-18 by VojtechBartoska

tuantvtlove, 2023-10-11:
[image]

tuantvtlove, 2023-10-11:
help

mrengineer7777, 2023-10-11:
@tuantvtlove You have not provided enough info to diagnose the problem.  You do have something corrupt in the filesystem.  Try erasing it and reprogramming.  The steps are noted above.  If you continue to have problems then open a new issue and provide details.