espressif/arduino-esp32 #8232: Light sleep hangs with ESP32-C3

NathanSweet, 2023-05-19:
### Board

Seeed Studio XIAO ESP32C3

### Device Description

[Seeed Studio XIAO ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html) with nothing attached except USB via the USB-C connector.

### Hardware Configuration

Nothing.

### Version

v2.0.9

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

When I call `esp_sleep_enable_timer_wakeup` with a value like 95000 and then call `esp_light_sleep_start`, it never returns. I found if I use 15 then it does return every time. I increased it up to 850 microseconds and it returns a few times, then it hangs.

### Sketch

```cpp
#include <Arduino.h>
#include <esp_sleep.h>

void setup () {
	Serial.begin(115200);
}

void loop () {
	uint64_t sleepMicros = 850;

	delay(100);
	esp_sleep_enable_timer_wakeup(sleepMicros);
	delay(100);

	Serial.print("Sleeping: ");
	Serial.println(sleepMicros);
	delay(100);

	uint32_t s = micros();
	esp_light_sleep_start();
	uint32_t e = micros();

	Serial.print("Awake: ");
	Serial.println(e - s);

	delay(100);
}
```
```

### Debug Message

```plain
Example output:

Sleeping: 850
Awake: 958
Sleeping: 850
Awake: 958
Sleeping: 850
Awake: 957
Sleeping: 850
Awake: 958
Sleeping: 850
```

Nothing is ever printed after that. Sometimes it goes on for longer before hanging.
```

### Other Steps to Reproduce

My `platformio.ini`:

```
[platformio]
src_dir = .

[env:test]
platform = espressif32
board = seeed_xiao_esp32c3
framework = arduino

monitor_speed = 115200

build_flags =
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-23:
@NathanSweet - You are using ESP32-C3 with USB CDC interface.
When the USB Host (computer) enumerates the interface, it will be available for connection and it is possible to use a Serial Terminal or any other Serial Monitor application to read the ESP32-C3 output.

USB Host is the unique connection controlling entity, which sends requests to the USB Device (ESP32-C3).
The USB Device must reply back to the USB Host in order to keep the channel open and the Host considers that the Device has not been disconnected.

When the ESP32-C3 goes into Sleep mode, its USB interface doesn't replay back to the Host, which thinks that the device is unplugged. 

The C3 doesn't hang. The USB CDC connection is the one that is halted by the USB Host side (computer) because the C3 can't reply back in a certain timeout to the USB Host.

A way to verify it is by using this sketch. It may or not hang the CDC interface, but the LED will continue to blink, indicating that the ESP32-C3 is still running.

``` cpp
#include <Arduino.h>
#include <esp_sleep.h>

#define RGB_IO 8 // set it to your board RGB LED GPIO or use some LED+resistor connected to thse GPIO

void setup () {
  Serial.begin(115200);
  delay(1000);
  Serial.print("Testing USB + light sleep.\nPC USB Host may consider that the interface has gone off...\n\n");
}

void loop () {
  uint64_t sleepMicros = 500;   // use 2000 to force the USB HOST to end the CDC connection
  static bool LED_ON = true; 

  delay(100);
  esp_sleep_enable_timer_wakeup(sleepMicros);
  delay(100);

  // this example works with the C3 devkit. 
  // for the SEED XIAO C3, it is possible to just use a regula LED and a resitor 
  // connected to any GPIO - in this case, just use pinMode() and digitalWrite() to see it blinking.
  if (LED_ON) neopixelWrite(RGB_IO, 0, 16, 0); // LED goes Green
  else neopixelWrite(RGB_IO, 0, 0, 0);         // LED goes OFF

  Serial.print("Sleeping: ");
  Serial.println(sleepMicros);
  delay(100);

  uint32_t s = micros();
  esp_light_sleep_start();
  uint32_t e = micros();

  Serial.print("Awake: ");
  Serial.println(e - s);

  delay(100);
  LED_ON = !LED_ON;
}

```

NathanSweet, 2023-05-23:
@SuGlider Thank you for taking a look. I appreciate your thorough description of what is happening.

I used your script with my Xiao ESP32-C3 and an LED and verified you are correct: with a 2500us light sleep the LED continues to blink even though the prints in the USB monitor cease.

In retrospect, I should have thought to try this. I was thrown off by a couple things:
1. The USB device still appears connected, eg it continues to be shown in Windows' Device Manager. 
2. My ESP32 does not show the same behavior. I ran the same sketch and the USB prints are shown. That is true for 2500us but also when light sleeping for 5 seconds every loop.
My ESP32 is a wESP32 which uses a Silicon Labs CP2102N for USB. I suppose that is significantly different and I should not expect it to behave the same as an ESP32-C3 using USB CDC.

I don't know if there is still an issue here. If USB CDC can continue to function during all light sleep, that would be great. If that's not possible then this issue can be closed and I apologize for my misunderstanding.

SuGlider, 2023-05-24:
@NathanSweet - the ESP32 using a CP2102 chip for UART-USB integration is very different from the C3 using its hardware JTAG-CDC USB interface.
When the ESP32 goes into sleep mode, the CP2102 is not sleeping and this chip can reply back to the USB Host (Computer), keeping the connection alive.
The SEED XIAO C3 has not such chip. The CDC is internal to the C3 and when it goes into sleep mode, the CDC also stops working. Therefore, C3 doesn't reply back to the USB Host and there is no other hardware chip to do it.

I have a C3 devkit board that has both, the USB CDC interface and the CP2102 connected to the C3 UART. If I upload the sketch with the CDC option off, using the UART + CP2102 of the board instead, the sketch works fine, because the `Serial` is now the UART and not the USB JTAG-CDC interface of the C3.

Think that the CP2102 is a full chip that takes care of the USB Device<--->Host communication that also has an UART port on it.
It never sleeps... always consume energy.

I think that there is no issue here.

NathanSweet, 2023-05-24:
That makes good sense. Thanks again for all your help! I'll close this issue.

closed 2023-05-24 by NathanSweet