espressif/arduino-esp32 #8233: timer_isr_callback_add - register interrupt failed.  Platformio project

bdewalch, 2023-05-19:
### Board

Nodemcu ESP32 wroom 32

### Device Description

Custom prototype board for security device

### Hardware Configuration

switches, mechanical and electronic 

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have a timer that I am trying to attach an interrupt.  The interrupt service routine should be called after the timer runs but it is never called.  The message "E (725442) timer_group: timer_isr_callback_add(236): register interrupt service failed" is output to the terminal window.  

### Sketch

```cpp
timer_1 = timerBegin(1, 80, true);  
.
.code above in a function that calls the function with the code below
.
timerAttachInterrupt(timer_1, &isr_Comm20msLow, true);
timerAlarmWrite(timer_1, 200, false); // 20ms
timerAlarmEnable(timer_1);
.
.code for ISR below
.
void IRAM_ATTR isr_Comm20msLow()
{
    // set mCommLowOk to true to indicate that at least 20ms have passed before a rising edge has occurred
    mCommLowOk = true;
    timerDetachInterrupt(timer_1);
}
```

### Debug Message

```plain
E (725442) timer_group: timer_isr_callback_add(236): register interrupt service failed
```

### Other Steps to Reproduce

The code was running fine, with no problems.  I have been doing development on a nodemcu ESP32 dev board.  When I tried to load the code on the production board, I had problems loading the code so I figured that environment variables needed to be added to support the new board.

I added a new environment variable entry in the platformio.ini file and using the platformio commands I updated the dependencies.  It was after these changes that the code stopped working.

My assumption is that the previous configuration (probably with earlier framework files) was working.  When I made the change, I began to get the error message.

I traced the command to the esl32-hal-timer.c file and in looking at the commit history noticed that there have been some changes made to " Fix timerAttachInterrupt() and timerDetachInterrupt() for migration to IDF 4.4" . 

The error message that I receive mentions the group parameter and my investigation led me to suspect that it may not be getting set properly.  I do not have a convenient way to debug the code at the moment so this is just a thought.

After extensive searching, I have not been able to find others having issues with the timer.  Possibly my configuration is bad but after deleting and reloading platformio and the libraries, I am unsure what is going on.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-20:
Please show your platformio.ini.

bdewalch, 2023-05-20:
Here is the latest.  I have tried several configurations without success/change.

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200
upload_speed = 921600

I am including the commented out code to illustrate some of the other parameters that I have tried without success:

; ;platform = espressif32@3.3.2;later versions of this library seem to have problems. v3.2 works
; ;not sure when problem begins.  Problems with interrupt handlers

 I was able to get the code to respond with the statement above and then it stopped after the first or second attempt without making any changes.

; platform = espressif32
; ;platform = https://github.com/platformio/platform-espressif32.git
; ;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

; ;platform = espressif32@5.3.0

; board = nodemcu-32s
; framework = arduino
; ;framework = espidf

; monitor_speed = 115200
; upload_speed = 921600
; use ESP-IDF 4.2.0
;platform_packages = framework-espidf@4.4.4
;lib_deps = fbiego/ESP32Time@^2.0.0
;debug_tool =esp-prog
;debug_init_break = tbreak setup

Additional Information::
This is a separate library but possibly related.  I am getting the message below before I get the message regarding the timer interrupt:

E (5031) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first

However, the gpio interrupts are responding and possibly the service installation is getting called in the library code.  In other words, the gpio interrupts seem to be triggering properly in spite of the message.

Many thanks for your help!

bdewalch, 2023-05-20:
Correction:  "This is a separate library (FILE not library) but possibly related."

mrengineer7777, 2023-05-21:
I've had trouble with "upload_speed = 921600". I believe my project uses half that speed.

What is the purpose of your timer? Can you use Ticker instead? Ticker has worked perfectly for me.

I'm not familiar with "fbiego/ESP32Time@^2.0.0". Based on the name I wouldn't expect it to access any gpios.

bdewalch, 2023-05-22:
Thanks, I will try the lower speed.  The timer is used to do several timing chores.  The most important is to control transmission period for a custom serial communication.  It transmits 200+ bits without syncrnronization and requires sub microsecond resolution to maintin sync over the entire transmission.  I will take a look at Ticker to see if it might work.  It seems that this behavior is not correct.  Particularly since the code worked prior to updating the libraries.  Additionally, do you think that the upload speed would affect the registration of the interrupt?

mrengineer7777, 2023-05-22:
>  It transmits 200+ bits without syncrnronization and requires sub microsecond resolution to maintin sync over the entire transmission. I will take a look at Ticker to see if it might work

I believe Ticker is 1ms resolution, so it won't work for your situation.

> do you think that the upload speed would affect the registration of the interrupt?

No.  Upload speed is only for programming of the module.  It has no effect on runtime code.

bdewalch, 2023-05-22:
Thanks for the help.  Do you think that the behavior that I am experiencing is caused by something that I am doing?  It seems like other people would have problems with it as well but I could not find any solutions after a fair amount of searchine.

mrengineer7777, 2023-05-22:
It seems that timerAttachInterrupt has had issues in the last few releases.  I'm not one of the maintainers, so unfortunately I don't know what changed.  Perhaps looking at the history will help? https://github.com/espressif/arduino-esp32/commits/master/cores/esp32/esp32-hal-timer.c

me-no-dev, 2023-05-23:
HW Timer is actually one of the things that is being tested by CI on real hardware: https://github.com/espressif/arduino-esp32/blob/master/tests/timer/timer.ino

I would say, the issue is not in our core, but rather some misconfiguration/cache/etc issue in your environment. I am not very familiar with PIO versioning, so I am not sure how `espressif32@3.3.2` corresponds to our version for example. Getting the version of `framework-arduinoespressif32` could help figuring out what your env might be.

bdewalch, 2023-06-03:
Thanks for the help.  After extensive effort, my conclusion was that there seems to be a problem with the library at least with the nodemcu-32 board.

I solved the problem by just using the ESP32 General Purpose Timer calls directly in my code.  These are the same calls that esp32-hal-timer.c uses.  When I substituted the calls directly in my code using the documentation in on Espressif's web site: https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/timer.html.  Everything worked perfectly - no problems.  This was done with the same configuration/cache/etc in my environment.

So, possibly there was something not configured just right in my environment that caused the problem.  It may be worth a look at what could be causing this level of tweakyness to help other programmers such as myself that are not as up to speed as the experts to get their code to work.  The library code in the esp32-hal-timer.c has some conveniences that would be nice, it is a shame not to be able to use it.  And, timers are so important for real time applications such as the product that we are developing.

bdewalch, 2023-06-03:
Also, the code here: helped to get the Espressif code working quickly.
https://embeddedexplorer.com/esp32-timer-tutorial/

closed 2023-08-18 by VojtechBartoska