espressif/arduino-esp32 #8235: WebUSB example link broken

zjwhitehead, 2023-05-20:
### Board

ESP32S3

### Device Description

NA

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

NA

### Operating System

NA

### Flash frequency

NA

### PSRAM enabled

yes

### Upload speed

NA

### Description

I cant find a working WebUSB HTML/CSS example. 
The one linked in the docs here https://github.com/espressif/arduino-esp32/blob/master/docs/source/api/usb.rst#webusburl (https://espressif.github.io/arduino-esp32/webusb.html) returns a 404. 

I noticed that the example Arduino sketch links refers to `localhost/webusb`
https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBVendor/USBVendor.ino#L171

Can we update the docs with a working sample webusb site that is accessible?
FWIW I found https://example.tinyusb.org/webusb-serial/index.html that we could base it off of.

### Sketch

```cpp
NA
```

### Debug Message

```plain
404 not found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JpEncausse, 2023-07-13:
Thanks  @zjwhitehead 
+1 looking for a working example (Arduino and Web)  to make a communication between ESP32-S3 and a web page.

The [USBVendor code](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBVendor/USBVendor.ino) seems to work. At least the device is recognized by windows and web browser

But on the web side I don't know what to do. This [blog article](https://medium.com/@gerybbg/webusb-by-example-b4358e6a133c) is providing some sample  but the Arduino code is different and rely on a custom made library

VojtechBartoska, 2024-02-20:
@lucasssvaz Can you please also take a look on this, probably we can fix/improve this for 3.0.0

PR #9349 (2024-03-11): Improve USBVendor Example
## Description of Change
Improves USBVendor example by adding the flush method and adding WebUSB console hosted inside the documentation.

## Tests scenarios
Tested with ESP32-S3 DevKit

## Related links
Closes #8235
closed 2024-03-11 by me-no-dev