espressif/arduino-esp32 #8237: ESP32-S3 USB frustrates me, you too?

frankcohen, 2023-05-22:
### Board

Reflections ESP32-S3-Mini custom board, similar to Adafruit Feather S3

### Device Description

I'm building an ESP32-based wrist watch to show movies of my kids. It's an open-source platform (https://github.com/frankcohen/ReflectionsOS). I'm using the built-in USB support in ESP32-S3 and building code using Arduino IDE 2.1.0. I haven't written my own board definition yet, so I'm using the Adafruit Feather ESP32-S3 No PSRAM board. 
[image]

### Hardware Configuration

USB port is connected to the normal ESP32-S3 pins for communication, power, and reset.

### Version

v2.0.9

### IDE Name

Arduino IDE 2.1.0

### Operating System

MacOS 13.3.1 (Ventura)

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Most of the time I have to go through these steps to get the board to accept a new code upload:

- Hold Boot down, press and release Reset (Reset is closer USB port)
- From pop-up "Select other board and port", choose Adafruit Feather ESP32-S3 No PSRAM and USB port
- Upload
- Press Boot, then press Reset
- Re-open the Serial Monitor

esp[tool.py](https://esptool.py/) reports "Hard resetting via RTS pin..." yet I need to press Boot, then press Reset to get the sketch to run. Espressif appears aware of this problem and has no solution yet. https://github.com/espressif/arduino-esp32/issues/6762

CDC support is incompatible with TinyUSB. That doesn't bother me yet. I'm expecting I will soon enable Over The Air OTA uploads and will use USB only for providing power.

I'd like to move on to [Platform.IO](https://platform.io/) and JTAG debugging. That just seems like another can of worms. The [Platform.IO](https://platform.io/) interface seems confusing to me, and I'm thinking I'll have the same USB problems.

What are you using? Am I missing something?

-Frank

### Sketch

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  long time = millis();
  while (!Serial && ( millis() < time + 5000) ); // wait up to 5 seconds for Arduino Serial Monitor  
  delay(2000);
  Serial.println(" ");
  Serial.println("Test");

}

int i = 0;

void loop() {
  Serial.println( i++ );
  delay(500);
}
```

### Debug Message

```plain
Sketch uses 359541 bytes (17%) of program storage space. Maximum is 2097152 bytes.
Global variables use 43028 bytes (13%) of dynamic memory, leaving 284652 bytes for local variables. Maximum is 327680 bytes.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodem14301
Waiting for upload port...
No upload port found, using /dev/cu.usbmodem14301 as fallback
"/Users/frankcohen/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool" --chip esp32s3 --port "/dev/cu.usbmodem14301" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/D4A47B7A0DB94325FA4038B7D88EA811/AccelGestureDTW.ino.bootloader.bin" 0x8000 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/D4A47B7A0DB94325FA4038B7D88EA811/AccelGestureDTW.ino.partitions.bin" 0xe000 "/Users/frankcohen/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/D4A47B7A0DB94325FA4038B7D88EA811/AccelGestureDTW.ino.bin" 0x410000 "/Users/frankcohen/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/variants/adafruit_feather_esp32s3_nopsram/tinyuf2.bin"
esptool.py v4.5.1
Serial port /dev/cu.usbmodem14301
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:f5:7e:14
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00067fff...
Flash will be erased from 0x00410000 to 0x00438fff...
Compressed 22624 bytes to 14245...
Writing at 0x00000000... (100 %)
Wrote 22624 bytes (14245 compressed) at 0x00000000 in 0.4 seconds (effective 417.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 136...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.1 seconds (effective 487.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 638.0 kbit/s)...
Hash of data verified.
Compressed 359904 bytes to 209872...
Writing at 0x00010000... (7 %)
Writing at 0x0001a3be... (15 %)
Writing at 0x0002740f... (23 %)
Writing at 0x0002ca1b... (30 %)
Writing at 0x000327c2... (38 %)
Writing at 0x0003818e... (46 %)
Writing at 0x0003d919... (53 %)
Writing at 0x00042ed4... (61 %)
Writing at 0x00047f74... (69 %)
Writing at 0x0004d21a... (76 %)
Writing at 0x00052749... (84 %)
Writing at 0x0005ac36... (92 %)
Writing at 0x000634e7... (100 %)
Wrote 359904 bytes (209872 compressed) at 0x00010000 in 3.9 seconds (effective 744.8 kbit/s)...
Hash of data verified.
Compressed 165536 bytes to 106746...
Writing at 0x00410000... (14 %)
Writing at 0x00418667... (28 %)
Writing at 0x0041dbce... (42 %)
Writing at 0x0042459a... (57 %)
Writing at 0x0042a572... (71 %)
Writing at 0x0042feab... (85 %)
Writing at 0x004352c0... (100 %)
Wrote 165536 bytes (106746 compressed) at 0x00410000 in 1.9 seconds (effective 686.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### Other Steps to Reproduce

I also tried to run esptool.py directly, same result... the cannot get a response from the ESP32-S3.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-05-22:
You are using the OTG USB peripheral, instead of the integrated CDC+JTAG. That port can not reset from download mode. If you do not need the USB for things other than serial port and JTAG, I suggest you switch to use that mode. You can also use the `ESP32-S3 Dev Board` definition, because it offers all options and does not use a custom bootloaders, etc., like the Adafruit boards. They are mostly meant to be used with MSC UF2 (drop a file in a USB storage device)

frankcohen, 2023-05-24:
Thanks for the reply @me-no-dev. I cannot get the board to upload code. I get an error. Here is the behavior I see in Arduino IDE 2.1:

```
Sketch uses 258301 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 19496 bytes (5%) of dynamic memory, leaving 308184 bytes for local variables. Maximum is 327680 bytes.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodem1234561
Waiting for upload port...
No upload port found, using /dev/cu.usbmodem1234561 as fallback
"/Users/frankcohen/Library/Arduino15/packages/esp32/tools/esptool_py/4.5.1/esptool" --chip esp32s3 --port "/dev/cu.usbmodem1234561" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/7E6873437A0CB0B0CF5921C73E32951B/sketch_may21a.ino.bootloader.bin" 0x8000 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/7E6873437A0CB0B0CF5921C73E32951B/sketch_may21a.ino.partitions.bin" 0xe000 "/Users/frankcohen/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/8y/z6t4zfcs2611ddcwsm1cg1080000gn/T/arduino/sketches/7E6873437A0CB0B0CF5921C73E32951B/sketch_may21a.ino.bin" 
esptool.py v4.5.1
Serial port /dev/cu.usbmodem1234561
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```
I see this in the Serial monitor:

[0;32mI (2121188) example: log -> USB[0m
example: print -> stdout
example: print -> stderr
[0;32mI (2122688) example: log -> USB[0m
example: print -> stdout
example: print -> stderr
[0;32mI (2124188) example: log -> USB[0m
example: print -> stdout
example: print -> stderr

How do I upload a sketch and let it run without pressing buttons?

-Frank

dsilletti, 2023-05-24:
please follow this guide, for the 1st upload you must use put the board in DFU mode manually https://www.hackster.io/hanguyen92205/program-debug-esp32s3-via-usb-serial-onchip-with-arduino-eeead3

SuGlider, 2023-05-25:
@frankcohen - In order to make the S3 work seamless for uploading and reseting the board after uploading, please use the USB HW (JTAG/CDC) configuration. It can be selected in the IDE Menu:

USB Mode ==> Hardware CDC and JTAG
USB CDC On Boot ==> Enabled
Upload Mode ==> UART0 / Hardware CDC

Upload any sketch to activate it for the next time.
`Serial` will work in the USB CDC port of the board.
`Serial0` is UART 0, `Serial 1` is UART 1 and so on. 

[image]

frankcohen, 2023-05-26:
Thank you Rodrigo, very kind to post these instructions.

May I ask you another question: I have an Espressif ESP-Prog board for JTAG debugging. I connect it to GPIO39, 40, 41, 42. What are the Arduino IDE settings to use that?

-Frank

SuGlider, 2023-05-26:
@frankcohen - The ESP32-S3 doesn't need the ESP-PROG because it already has the JTAG USB.
Please follow the instructions that can be found in the documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-builtin-jtag.html

As for the new IDE 2.1.0, it is necessary to:
1) Select the ESP32-S3 DevKit board
2) Select the USB port 
3) Enable CDC on Boot (IDE Board Menu)
4) Select "Upload Mode" as "UART0 / Hardware CDC"
5) Select "USB Mode" as "Hardware CDC and JTAG"
6) Select "JTAG Adapter" as "Integrated USB JTAG"
7) Select and load the Sketch that will be debugged
8) Build and Upload it
9) Press the Debug "play" button to open the DEBUG window
10) Press the Green "play" button and wait for OpenOCD and dbg to start in the "Debug Console"
11) Add the BREAKPOINTS
12) Use the DEBUG UI as desired

Some Images of my IDE may help:

Menu Options:
[image]

Screenshot of the Debugger in Action:
[image]

SuGlider, 2023-05-26:
@frankcohen - I hope that the ESP32-S3 USB stops frustrating you now. Let me know.

sblantipodi, 2023-05-26:
@SuGlider ESP32-S3 USB works really bad...
it destroyed the usability with tools like [ESP Web Tools](https://github.com/esphome/esp-web-tools).

SuGlider, 2023-05-26:
> @SuGlider ESP32-S3 USB works really bad... it destroyed the usability with tools like [ESP Web Tools](https://github.com/esphome/esp-web-tools).

I'll test the Web USB uploading. 
Beside it, could you use the USB JTAG as uploading port, like as with a plain UART?
Did it work as JTAG debugging tool within the Arduino IDE?
All this information will help the community and also will be used as base for a future FAQ about the S3.

SuGlider, 2023-05-26:
@frankcohen - About [ESP Web Tools](https://github.com/esphome/esp-web-tools):

1) I can select the USB JTAG port and connect to the S3, seeing the logs, including debug log messages (see screenshots)
[image]
[image]

2) I can also reset the device from the ESP Web Tools:
[image]

3) I also can't upload Tasmota for the S3 using their ESP WEB TOOL.
[image]

When using the USB JTAG port, I see that it can't detect correctly the device.
This can be an issue with the TS scripts... 

4) By other hand, I can use the same USB JTAG to upload a C3, which uses exactly the same digital design for this USB port from the S3. I have used an Espressif Tool: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml

[image]

SuGlider, 2023-05-26:
> @SuGlider ESP32-S3 USB works really bad... it destroyed the usability with tools like [ESP Web Tools](https://github.com/esphome/esp-web-tools).

It seems that ESP Home Web Tools may need some patches to work in the same way that Espressif Launchpad does in order to be able to upload using the USB JTAG port.

sblantipodi, 2023-05-26:
@SuGlider to be able to upload using USB JTAG you need to do the following:
Oress the "0" button, 
keep it pressed, 
press "RST" you hear a sound from windows, 
release "RST", you need to hear another sound from windows so don't do it too fast,
reload the web flasher page, 
then select the flash procedure, 
select com port, etc. and when it is showing the progress meter with at least a few % you can release the "0" button.

This works on S3 and C3. Does not work on S2.

When using TinyUSB (that it's obviously far better for our firmwares) the procedure is even more tricky...

SuGlider, 2023-05-26:
I see, but as I could verify, Espressif Lauchpad (Web Uploading) does something different that works with no button pressing.
I think that the team from ESP Web Tool could check it out and try to implement something similar there.

https://github.com/espressif/esp-launchpad

sblantipodi, 2023-05-26:
@SuGlider esp launchpad does not see TinyUSB devices at all...
if I put the ESP in CDC mode it does not connect on my S2.

[image]

SuGlider, 2023-05-26:
Yes, there is a problem with TinyUSB (S2 and S3). Something to investigate...
Anyway, I think we have covered the S3 + USB JTAG pretty well.

SuGlider, 2023-05-26:
> @SuGlider to be able to upload using USB JTAG you need to do the following: press the "0" button, keep it pressed, press "RST" you hear a sound from windows, release "RST", you need to hear another sound from windows so don't do it too fast, reload the web flasher page, then select the flash procedure, select com port, etc. and when it is showing the progress meter with at least a few % you can release the "0" button.
> 
> This works on S3 and C3. Does not work on S2.
> 
> When using TinyUSB (that it's obviously far better for our firmwares) the procedure is even more tricky...

Please try this other web site:
https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml

It can connect to the USB JTAG from the S3 with no issues and no need for pressing buttons.
When flashing, the tool seems to do not test the ChipID and therefore it tries to flash a wrong firmaware into the S3.

SuGlider, 2023-05-26:
Another importante detail: In order to make USB JTAG upload and reset automatially, it is necessary to upload a first sketch with the USB JTAG/CDC configuration as defines in https://github.com/espressif/arduino-esp32/issues/8237#issuecomment-1563497918

frankcohen, 2023-05-26:
This is great and thank you. Now I’m wondering about cables. Reading the docs you reference I see these ESP pins and USB signals:

ESP32-S3 Pin
USB Signal
GPIO19
D-
GPIO20
D+
5V
V_BUS
GND
Ground
My custom board followed the data sheet spec and put the JTAG header pins off GPIO 39, 40, 41, 42, with GPIO 39 =  as TCK, 40 as TDO, 41 as TDI, and 42 as TMS.

￼

from the data sheet:

￼

Seems like a custom USB cable is needed:

￼

Right?

-Frank

frankcohen, 2023-05-26:
Seems that Github doesn't like embedded images from email. Here is my messsage again.

This is great and thank you. Now I’m wondering about cables. Reading the docs you reference I see these ESP pins and USB signals:

[image]

My custom board followed the data sheet spec and put the JTAG header pins off GPIO 39, 40, 41, 42, with GPIO 39 =  as TCK, 40 as TDO, 41 as TDI, and 42 as TMS.

[image]

from the data sheet:
'
[image]

Seems like a custom USB cable is needed:
[image]

Right?

SuGlider, 2023-05-26:
The ESP32-S3 USB JTAG is all that you need for debugging. Just GPIO 19 and 20 (D- / D+).
GPIOs 39 to 42 are meant for using with the ESP-PROG.
Those are 2 different and independet ways to using JTAG.

SuGlider, 2023-05-26:
Arduino IDE has a few option in S3 the Menu-->JTAG Adapter:
`Integrated USB JTAG` --> use the regular USB (GPIO19 D- and GPIO20 D+) as JTAG interface - implemented in HW
`FTDI Adapter` --> When there is an FTDI chip with JTAG (USB) in the board, like the ESP-PROG
`ESP USB Bridge` --> Using utilizing an ESP32-S2 or an ESP32-S3 chip to create a JTAG bridge - it's a software JTAG.

These option will define an OpenOCD configuration file to be used within the IDE debugger.

SuGlider, 2023-05-26:
@frankcohen -- Important Note from Doumentation:
[image]
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html

frankcohen, 2023-05-27:
Hi @SuGlider thanks! I created a USB cable for the JTAG support. 

[image]

This connects to the ESP32-S3-DevKitMini-1 I just bought. I selected ESP32-S3-Box in Arduino IDE 2.1 on Macbook Pro. I could find no ESP32-S3-DevKitMini board definition in Arduino IDE 2.1. The upload speed compared to Arduino IDE 2.1 over USB was surprising - much faster on the JTAG interface.

[image]

[image]

Next for me:
1) On Arduino 2.1 how to see output to the Serial Monitor over JTAG? Or, do I need to connect a second USB cable to my Powerbook to see  the Serial.println(" "); logging?
2) How to do step/breakpoints on Arduino 2.1 on JTAG?
3) How to configure this board and JTAG cable on Platform IO? 

I'm not finding useful search results on the above issues on Google nor the Espressif Web site search.

-Frank

SuGlider, 2023-05-27:
>On Arduino 2.1 how to see output to the Serial Monitor over JTAG? Or, do I need to connect a second USB cable to my >Powerbook to see the Serial.println(" "); logging?

The same USB JTAG works as USB CDC (serial communication over USB). When `USB CDC On Boot` is set  `Enabled` in the Arduino IDE, the software will set the object `Serial` to the `HWCDC` class and it will work out of the box, using the JTAG/CDC USB port for any `Serial.print()`.

No need to connect any second cable.

`log_d()` also works, but it is necessary to tell Arduino to use the `HWCDC` interface about it.
`Serial.setDebugOutput(true)` will do it when  `USB CDC On Boot` is set  `Enabled` - in this case `Serial0` is the UART0
`USBSerial.setDebugOutput(true)` will do it when `USB CDC On Boot` is set  `Disabled` - in this case `Serial` is the UART0

>How to do step/breakpoints on Arduino 2.1 on JTAG?

First of all, it is necessary to set the Arduino IDE 2.1 and activate the Debugging mode.
Please refer to https://github.com/espressif/arduino-esp32/issues/8237#issuecomment-1563746127
After that, in the Debug Window there is a `Breakpoints` session - click on `+` to add functions or go to the Source Code window and click on the left border to the Code Line to add a breakpoint there.

>How to configure this board and JTAG cable on Platform IO?

For the Hardware, there is no change. It is the same as explained above.
The PIO IDE can open a `/dev/....` (linux, Mac)  or `COMxx` (Windows) and set it for the Output.

SuGlider, 2023-05-27:
@frankcohen - The S3 DevKit already exposes the GPIOx 19 and 20 as USB. Maybe you alreade know it.
[image]

frankcohen, 2023-05-27:
@SuGlider thank you Rodrigo! I appreciate your experience and willingness to share. I plan to follow your instruction for Arduino IDE 2.1 and Platform.io. I feel a tutorial video for YouTube and Reddit article coming from this experience to help others. Also, I will let [Andreas Speiss ](https://www.youtube.com/@AndreasSpiess) know about this, perhaps he will be inspired to make a video too. -Frank

Jason2866, 2023-05-27:
> @frankcohen - About [ESP Web Tools](https://github.com/esphome/esp-web-tools):
>
> 1. I can select the USB JTAG port and connect to the S3, seeing the logs, including debug log messages (see [...]

@SuGlider  The issue is known. But actually no one has an idea how to solve. Open Issue https://github.com/esphome/esp-web-tools/issues/331

frankcohen, 2023-05-27:
Hi @SuGlider, Rodrigo, I'm seeing Serial Monitor logging over JTAG USB, thanks! 

When I try to enable the debugger in Arduino IDE 2.1 with the Espressif ESP32-S3-DevKitMini-1 over JTAG USB using Adafruit Feather S3 No PSRam board. I don't think I have everything configured correctly. I'm on a Mac Powerbook, running 13.3.1 (a). I do not have OpenOCD installed.

[image]

Should I install OpenOCD with HomeBrew?

SuGlider, 2023-05-28:
> Should I install OpenOCD with HomeBrew?

No, it is installed with the ESP32 Pacakage at `/Users/{username}/Library/Arduino15/packages/esp32/tools/openocd-esp32`
Have you selected the `JTAG Adapter` option in the menu as `Integrated USB JTAG`?
Press the Green "Play" button, it shall load OpenOCD and activate the debugging.

SuGlider, 2023-05-28:
> @SuGlider The issue is known. But actually no one has an idea how to solve. Open Issue [esphome/esp-web-tools#331](https://github.com/esphome/esp-web-tools/issues/331) Webtools is using esptool.js v2.1 which is the exact same version as in LaunchPad

From what I could test, I found out that the S3 must have CDC/JTAG already active. This can be done by uploading, for example an empty sketch, with CDC/JTAG setup prior to using LaunchPad. 
This firmware will "turn on CDC/JTAG" after booting, setting the conditions to the S3 to work with LaunchPad. 
The new uploaded firmware shall also turn on CDC/JTAG for the following upload, and so on...

I had a similar issue with other MCUs when dealing with USB bootloaders... Since, it is activated by software, it need to have a uploaded proper firmware in order to work fine. The MCU boots, sets the proper USB mode, it may also need a running active USB driver when WebUSB tries to upload a new firmware. 

TinyUSB mode seems to fail when uploading using LaunchPad. 

I agree that it is not easy and quite confusing...

SuGlider, 2023-05-28:
The S3 native USB has 3 modes: (1) CDC/JTAG || (2) OTG - TinyUSB driver || (3) disabled
Each of these 3 modes are activated by the running firmware of the S3, right after booting.

USB Web tools and other uploading modes work best with the (1) CDC/JTAG.
It is possible to activate this mode by uploading an empty (or small) firmware/sketch that sets the CDC/JTAG mode.
But if the next uploaded firmware is created/programmed to use (2) or (3), the following uploads using USB will fail.

Therefore, S3 demands that CDC/JTAG is activated by the running firmware and that all next firmwares keep this mode always active.

frankcohen, 2023-05-28:
I confirmed the openOCD is installed in the correct path on my Mac. I set a breakpoint in the code and Debug seems to ignore it.

[image]

The program runs, and I see results in Serial Monitor, and nothing else, no breakpoints, etc.

Also, I am not seeing JTAG Adapter as an option when selecting ESP32-S3-Mini.

frankcohen, 2023-05-28:
And, of course, I wasn't running the sketch with the "Start Debugging" icon. Doing so I see the error: 
[image]

Jason2866, 2023-05-28:
> The S3 native USB has 3 modes: (1) CDC/JTAG || (2) OTG - TinyUSB driver || (3) disabled Each of these 3 modes are activated by the running firmware of the S3, right after booting.
>
> USB Web tools and [...]

If I understand your post correct, you are saying there is no possibility to flash a brand new S3 (which has no additional UART chip) reproducible correctly with the target productive firmware?
If this is the case it is a big hardware bug, which has to be mentioned in all S3 data sheets!

sblantipodi, 2023-05-28:
@Jason2866 PlatformIO and Arduino IDE are able to flash these chips in the way you want with the firmware you want. Sometimes you need to put the device in CDC mode before flashing by doing an absurd sequence of steps like pressing the IO button, then pressing the RST one, wait a bit, release the RST button leaving the IO pressed, than start flashing, then releasing the IO button.

It's clear that new S2 and S3 has some serious flaws on this.
The proof is the even the official web tools from espressif isn't able to flash their own devices.

Using CDC on S3 is a non sense since you loose all the benefits of having a native USB. CDC creates lags and some weird issues when pushing it with data.
TinyUSB works very well when streaming data on it but it's very difficult to flash the ESP once set in TinyUSB mode.

Jason2866, 2023-05-28:
@sblantipodi Yes, but as @SuGlider correctly mentioned only because there is a firmware before which enables the S3 to be correctly flashed.
It is a bug. The use case decides whether it needs all the USB features. For many use cases it is a one time need only. -> Flashing the firmware

SuGlider, 2023-05-28:
@sblantipodi @Jason2866 

>If I understand your post correct, you are saying there is no possibility to flash a brand new S3 (which has no additional UART chip) reproducible correctly with the target productive firmware?
If this is the case it is a big hardware bug, which has to be mentioned in all S3 data sheets!

All said is exclusively about using Web tools. WebUSB is different from the other tools.
For Win/Linux/Mac Espressive command line tools, it works fine.

Jason2866, 2023-05-28:
@SuGlider Thats good news. So just esptool.js needs to be fixed

SuGlider, 2023-05-28:
There many ways to flash a new firmware, including Wireless, SPI, UART (or JTAG), DFU, any other can be "invented".
Uploading a firmware or any user application data means that some piece of software will store a block of data in the Flash, using the internal ROM functions.

I "could invent a new way" that takes I2C data and store it in the flash. It is up the application/software. 
ESP32 has ROM functions (software) that help the user to achieve it more easily using, for instance, UART.
ROM functions also check if BOOT GPIO is pressed right after a RESET, starting an upload mode, as defined in the ROM code.
This is documented. 
https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/dfu.html#
https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/flashing-firmware.html

SuGlider, 2023-05-28:
Check this video: https://www.youtube.com/watch?v=Zqqj8GC0Q9Q
BOOT + RESET activates S2 TinyUSB ROM layer and then the computer is able to enumerate it.

Jason2866, 2023-05-28:
@SuGlider Thx for the links how to flash. Having no issues with flashing any espressif device local.
Espressif is providing https://github.com/espressif/esptool-js which is the base of ALL Web tools 
for flashing via a Browser. This is a great way for users which "just" want to install a ready made firmware.
Without installing any tools or knowing how this all works. Plug and Play.
Dont get me wrong i appreciate the great work you and your colleagues are doing here.
It is a bit frustrating that the issue (wherever it is) prevents the easy flashing of the S3 this way, as it is possible for every other espressif Arduino supported device.

frankcohen, 2023-05-31:
I created a companion video for the ESP32-S3 (and why you should stay away from the others). View it here [https://youtu.be/vQBxMgNvwZI](https://www.youtube.com/watch?v=vQBxMgNvwZI) -Frank

closed 2023-06-01 by SuGlider

VojtechBartoska, 2023-06-01:
Changing the title as the original one will no one find in the future, even though it's still generic.

frankcohen, 2023-06-01:
> Thanks @frankcohen <https://github.com/frankcohen> for the discussion and
> questions.
> I'll close this issue given that most doubts are answered.
> Please feel free to reopen it whenever needed.

Thanks

frankcohen, 2023-06-04:
I posted the JCUSB solution (my name for the S3 + USB + CDC + OpenOCD + Arduino IDE solution) to Reddit at
https://www.reddit.com/r/esp32/comments/140nc29/esp32_usb_is_frustrating_try_jcusb_for_s3_usb_cdc/

BotSpace, 2023-06-21:
Thank you for sharing this. I have been very frustrated by esp32S3 USB upload, but now JCUSB works very well for me. I can upload code and use the serial monitor, all without having to press buttons or change settings.

I have just one issue, when I unplug and replug my board, ArduinoIDE does not remember my port. And in the drop down my device shows up as TAMC Termod S3?
[image]
It is easy enough to select the port again but Im just curious to hear if anyone has thoughts on this.

frankcohen, 2023-06-21:
I have the same problem. I haven’t looked into Arduino IDE to see if the source is available, if it is I propose a contribution that does a better job at remembering the previously chosen board and port.

-Frank

me-no-dev, 2023-06-21:
Arduino.cc are aware of this issue and will fix it. Can not give timeframe yet :)

me-no-dev, 2023-06-21:
> That's great, thank you. Is there a ticket number or URL to the issue? Do
they take software contributions?

Issue is mentioned in a few places, but we have spoken to them directly to explain everything in detail. They do take contributions, though it might take some time for an external contribution to be accepted. I am fairly certain that this happens in `arduino-cli` and not the IDE itself

frankcohen, 2023-06-21:
Thank you, I will wait. I appreciate your advocacy and reply. -Frank

mukarram1999, 2023-07-28:
> @frankcohen - The ESP32-S3 doesn't need the ESP-PROG because it already has the JTAG USB. Please follow the instructions that can be found in the documentation: [...]

Could you please give full detailed instructions on how to enable debugging of esp32s3 through Arduino IDE 2.x? I am messing up somewhere but I have no idea where.

frankcohen, 2023-07-28:
Thank you for the detailed instructions, I appreciate your effort. I wrote an article pulling together all that I learned. I posted it to Reddit and got a good dialogue going. I'm calling the technique JCUSB. I am hosting a how-to document here: https://github.com/frankcohen/ReflectionsOS/blob/main/Docs/JCUSB%20using%20JTAG%2C%20CDC%2C%20USB%20for%20debugging.md

-Frank

frankcohen, 2023-07-28:
Here is the Reddit posting https://github.com/espressif/arduino-esp32/issues/8237

mukarram1999, 2023-07-29:
> Thank you for the detailed instructions, I appreciate your effort. I wrote an article pulling together all that I learned. I posted it to Reddit and got a good dialogue going. I'm calling the [...]

Thank you so much for that. However, this is where I am stumped:
1. What should launch.json config be to enable debugging? Debug does not happen without editing that but my Arduino IDE is not able to save any configurations in the first place.
2. I got OpenOCD installed using the official ESP-IDF tools installer, how do I get Arduino IDE to use it to enable debugging?
3. Is debugging plus uploading all possible with one cable? If not, how do I get the Arduino IDE to use a cable for uploading and the other for debugging?

frankcohen, 2023-07-29:
> What should launch.json config be to enable debugging? Debug does not happen without editing that but my Arduino IDE is not able to save any configurations in the first place.
Using Arduino IDE 2.1 Nightly Build on Mac OS, launch.json is ignored. Arduino IDE does some internal calculation to figure out where to find OpenOCD. It can get confused because OpenOCD uses Python 3 and there is some path strangeness on Mac. I found this resolved by using the Nightly Build. I click Upload, it’s builds and uploads, then I click the Debug button (upper left list of icons), and the debug panel appears.
> I got OpenOCD installed using the official ESP-IDF tools installer, how do I get Arduino IDE to use it to enable debugging?
Use the Arduino IDE nightly build and it should just work.
> Is debugging plus uploading all possible with one cable? If not, how do I get the Arduino IDE to use a cable for uploading and the other for debugging?
Yes, debugging plus uploading happens using the one cable (that I’m calling JCUSB).

-Frank

mukarram1999, 2023-08-02:
> Using Arduino IDE 2.1 Nightly Build on Mac OS, launch.json is ignored. Arduino IDE does some internal calculation to figure out where to find OpenOCD. It can get confused because OpenOCD uses Python [...]

I am using Arduino 2.1.1 IDE on Windows and unfortunately those stops don't work for me. Please help.

> Yes, debugging plus uploading happens using the one cable (that I’m calling JCUSB).

That is awesome.

Thanks so much for all your help on this, Frank.

frankcohen, 2023-08-02:
"I am using Arduino 2.1.1 IDE on Windows and unfortunately those stops
don't work for me. Please help."

Unfortunately I don't have a Windows machine. Have you installed the
nightly build? Have you tried posting to the Arduino IDE bug report or
discussion? They are responsive to me, and I bet would be able to help you
too.

Frank

AMttos, 2023-08-07:
All,

could any of you (that successfully can run the ESP32-S3 USB JTAG debugging in the Arduino IDE 2.1X) share your launch.json / debug_custom.json files, therefore I can troubleshoot my setup?  I posted at the endo of this tread the errors I am getting.  Am I missing something?  @SuGlider or @frankcohen

USB programming and reboot works fine.

I am using the following:

ESP32-S3-WROOM-2 with GPIO 19 and 20 as USB CDC communications and programing.
Arduino IDE 2.1.1
esp32 by Espressif 2.011, which installs OpenOCD and other tools 
- checked "Optimize for debugging" -  (IDE Sketch Menu)
- Enabled "CDC on Boot" -  (IDE Board Menu)
- Selected "Upload Mode" as "UART0 / Hardware CDC" -  (IDE Board Menu)
- Selected "USB Mode" as "Hardware CDC and JTAG" -  (IDE Board Menu)
- Selected "JTAG Adapter" as "Integrated USB JTAG" -  (IDE Board Menu)  

Here is my launch.json settings (as the Start Debugging button is pressed)
{
  "version": "0.2.0",
  "configurations": [
    {
      "cwd": "${workspaceRoot}",
      "name": "Arduino on ESP32-S3",
      "request": "attach",
      "type": "cortex-debug",
      "executable": "C:\\Users\\USER\\AppData\\Local\\Temp\\arduino\\sketches\\D0059C980E0F01DD711B407D02FD1697/Handheld_S3.ino.elf",
      "servertype": "openocd",
      "serverpath": "C:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/bin/openocd",
      "armToolchainPath": "C:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp-elf-gdb\\11.2_20220823/bin/",
      "configFiles": [
        "debug.cfg"        
      ],
        "toolchainPrefix": "xtensa-esp32s3-elf",
      "svdFile": "debug.svd",
      "overrideAttachCommands": [
        "set remote hardware-watchpoint-limit 2",
        "monitor reset halt",
        "monitor gdb_sync",
        "thb setup",
        "c"
      ],
      "overrideRestartCommands": [
        "monitor reset halt",
        "monitor gdb_sync"
      ]
    }
  ]
}

MY ERRORS
OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.

gdb-server
[2023-08-07T20:56:20.472Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\USER\\Documents\\Electronics\\01-Proj\\Handheld_S3" -f "C:/Program Files/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f debug.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: esp_usb_jtag: could not find or open device!

[2023-08-07T20:56:20.802Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

Thanks in Advance

me-no-dev, 2023-08-08:
@AMttos your configuration is correct, but openocd can not find the board.

@SuGlider is there anything additional that needs to be done on Windows?

AMttos, 2023-08-09:
@frankcohen and @me-no-dev 

Thank you very much for replying to my questions and confirming my settings are correct.  the weird thing is I can program and get serial monitor on the USB without issues. I do not understand why OpenOCD cannot find the board.  I will do more digging and also try to use another PC and see if the issue is reproducible.  I will let you know my findings.
Thanks again.

[1 more reply]