espressif/arduino-esp32 #8245: LWIP ALTCP linking error

HamzaHajeir, 2023-05-24:
### Board

ESP32 Devkit1

### Device Description

ANY

### Hardware Configuration

Independent

### Version

v2.0.8

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi there, I'm trying to develop TLS inside the working solid [H4AsyncTCP ](https://github.com/HamzaHajeir/H4AsyncTCP) using lwip raw APIs.

LWIP offers slight change from tcp functions to [Application Layered TCP](https://www.nongnu.org/lwip/2_1_x/group__altcp__api.html), which seems very slight development and I can get both TCP and TLS dynamically.

However, compiling the source code yielded `undefined reference to `altcp_XXX' linking errors, even if that I've defined LWIP_ALTCP in my platformio.ini.

Can I do something to compile against espressif/esp-lwip natively in PlatformIO? and How?
How can someone else compile fine within Arduino IDE?

Thanks for your help

### Sketch

```cpp
// This sketch doesn't produce the link-error, where altcp_xxx functions are mapped to tcp_xxx when LWIP_ALTCP == 0

#include "lwip/altcp.h"

void setup(){
    altcp_close((altcp_pcb *)nullptr);
}
void loop(){
}
```

### Debug Message

```plain
NO
```

### Other Steps to Reproduce

NO

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

HamzaHajeir, 2023-05-24:
Tried adding `espressif/esp-lwip` library to the environment in PlatformIO to see if it solves the issue, got an exception immediately after connecting to WiFi.
It seems to me that there're two separate sources got linked to the binary...

SuGlider, 2023-05-24:
@HamzaHajeir - ESP32 Arduino has some already precompiled libraries, including LwIP.
It is very possible that, as you have seen, the libraries don't have ALTCP functions.

I think that the only way to use ALTCP would be by moving your project to become a Arduino as IDF Component project.
It is possible to configure IDF to work the way it is necessary.

Arduino as Component is explained in https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

mrengineer7777, 2023-05-25:
@HamzaHajeir That's a nicely documented library.  Note to others: it specifically prohibits commercial use.

https://github.com/me-no-dev/ESPAsyncWebServer
https://github.com/me-no-dev/AsyncTCP
Are you already familiar with these?  I've read that ESPAsyncWebServer currently doesn't support HTTPS Webserver.  However it may give you a clue on how to structure your project to get it to compile.

If you are looking for LWIP functions that aren't compiled into arduino-esp32, then you'll need to switch to the IDF framework as Suglider suggested, OR fork [arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) and use it to build your own binaries..

HamzaHajeir, 2023-05-25:
> @HamzaHajeir That's a nicely documented library. Note to others: it specifically prohibits commercial use.
>
> https://github.com/me-no-dev/ESPAsyncWebServer https://github.com/me-no-dev/AsyncTCP Are [...]

Thanks for your notes, I'm sad to inform you that Phil has passed away one year ago, we were friends and we've discussed the license issue when he was changing from MIT, he told me that it wont affect friends.
The license issue is a next step to solve to unlock it to the public, we've failed to get his paperworks from his family, and last new we heard that they sold all his hardwares :\. We could be contacting them regarding license change perhaps we could get an agreement.

The noted libraries are sadly ineffictive and have serious bugs, that's why Phil had decided to build one.
You might watch his videos where he illustrated why:
- https://www.youtube.com/watch?v=O2W8uyC-Gio 
- https://www.youtube.com/watch?v=txpVm1ftGOI (shows real tests)

For the second choice (arduino-lib-builder), I think I can setup an environment on PlatformIO to point out to a customized libraries, can you guide me what steps needed to be done there? 
Is it adding LWIP_ALTCP to [defconfig.common](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.common) then execute `/build.sh`?
After building, Is it correct that I'll be able to specify them at PlatformIO?

Thanks

HamzaHajeir, 2023-05-25:
> @HamzaHajeir <https://github.com/HamzaHajeir> - I just compiled the
> example using Arduino Core 2.0.9 and the Arduino IDE 1.8.15.
> It has created the binary with no linker problems. It looks like an
> PlatformIO building problem instead...
>
> Sketch uses 269844 bytes (20%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 16972 bytes (5%) of dynamic memory, leaving 310708 bytes for local variables. Maximum is 327680 bytes.
>
> It seems clear that the precompiled LwIP Arduino libraries already have
> all necessary support for ALTCP.

Thanks for your trial, please confirm that you've added a compiler
definition -DLWIP_ALTCP=1, because it'll compile and link due to
preprecessors falling it back to normal tcp_xxx APIs.

It can be done easily in PlatformIO by this line in `platformio.ini`:
`build_flags = -DLWIP_ALTCP=1`

SuGlider, 2023-05-25:
> Thanks for your trial, please confirm that you've added a compiler definition -DLWIP_ALTCP=1, because it'll compile and link due to preprecessors falling it back to normal tcp_xxx APIs. It can be done easily in PlatformIO by this line in `platformio.ini`: `build_flags = -DLWIP_ALTCP=1`

No, I have not added this `build_flags`. I have not used PlatformIO.
I have used the Arduino IDE instead - with Arduino Core 2.0.9.

HamzaHajeir, 2023-05-25:
All altcp_xxx functions got fall backed to tcp_xxx ones, which could be
found easily in the precompiled headers.

See:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/altcp.h#L145

HamzaHajeir, 2023-05-29:
Went through the process of building customized arduino core by esp32-arduino-lib-builder by adding definitions to LWIP_ALTCP and LWIP_ALTCP_TLS to [defconfig.common](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.common), Replacing the [output of the build](https://github.com/HamzaHajeir/ESP32-Arduino-Builds) with [arduino-esp32 fork](https://github.com/HamzaHajeir/arduino-esp32/tree/lwip-tls), then used it in PlatformIO project.

What I've just realized is that these configurations I've just inserted don't make any effect within esp-lwip library!
As it doesn't inherit any ESP-IDF configuration macro ...
See:
https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/opt.h#L1525

If no solution here, where should I seek for a consistent solution? ESP-IDF? Arduino lib builder? esp-lwip?

SuGlider, 2023-05-29:
Another option would be to use Arduino as Component. It allows to build the whole IDF system.

HamzaHajeir, 2023-05-30:
> Another option would be to use Arduino as Component. It allows to build the whole IDF system.

It's a good option and I'll go for it for my project at some point in the future, but I can't enforce library users for this direction...

SuGlider, 2023-05-30:
Yes, it is correct. User Arduino Libraries depend on the precompiled IDF LwIP built with LibBuilder.
There is not other way than using ALTCP, when trying to maintain `H4AsyncTCP`?

Is it possible to build a wrapper around current Arduino "version" of IDF LwIP in order to provide the necessary functionalities?

HamzaHajeir, 2023-05-31:
> Yes, it is correct. User Arduino Libraries depend on the precompiled IDF LwIP built with LibBuilder. There is not other way than using ALTCP, when trying to maintain `H4AsyncTCP`?
>
> Is it possible to [...]

Thanks for your seek to help.
I'd like to inform you with resutls, I've successfully built a working core libraries containing ALTCP functions, the essential problem was those files are not fed into binaries, excluded from compiling by lwip CMakeLists file, I've successfully overriden them and I now have ALTCP functions inside.

Thanks for your help, I'm going to close this issue as I've found the solution, and BTW, I've requested the feature in [esp-idf](https://github.com/espressif/esp-idf/issues/11556).

closed 2023-05-31 by HamzaHajeir