espressif/arduino-esp32 #8246: ../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8 bytes - DRAM segment data does not fit - collect2.exe: error: ld returned 1 exit status    ?

dsyleixa, 2023-05-24:
### Board

Adafruit ESP32 Feather, Board ore 1.0.6

### Device Description

Adafruit ESP32 Feather, Board ore 1.0.6

### Hardware Configuration

standard, default

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Wundows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

upade issue from Arduino 1.8.9. portable  to 1.8.19 portable (identical portable directory)
Arduino 1.8.9.: all is fine.

BUT:
Aduino 1.8.19:
d:/arduino/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8 bytes     ?
DRAM segment data does not fit       ?  
collect2.exe: error: ld returned 1 exit status    ?

what does that mean and how to fix it?

### Sketch

```cpp
https://github.com/dsyleixa/Arduino/tree/master/ESP32_GBox/ESP32_Box031i
```

### Debug Message

```plain
Arduino 1.8.9.: all is fine:
...

Linking everything together...
"D:\\Arduino189\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LD:\\Arduino189\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/lib" "-LD:\\Arduino189\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/ld" -T esp32_out.ld -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u esp_app_desc -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\sketch\\ESP32_Box031i.ino.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\SD\\SD.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\SD\\sd_diskio.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\SD\\sd_diskio_crc.c.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\FS\\FS.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\FS\\vfs_api.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit-GFX-Library\\Adafruit_GFX.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit-GFX-Library\\Adafruit_SPITFT.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit-GFX-Library\\glcdfont.c.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_HX8357_Library\\Adafruit_HX8357.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_STMPE610-1-0-1\\Adafruit_STMPE610.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Wire\\Wire.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_ImageReader_Library\\Adafruit_ImageReader.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_ADS1X15\\Adafruit_ADS1X15.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_BusIO\\Adafruit_BusIO_Register.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_BusIO\\Adafruit_I2CDevice.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\Adafruit_BusIO\\Adafruit_SPIDevice.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\KalmanFilter\\Kalman.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\I2Cdev\\I2Cdev.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\MPU6050\\MPU6050.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\ETH.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFi.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiAP.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiClient.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiGeneric.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiMulti.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiSTA.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiScan.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiServer.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WiFi\\WiFiUdp.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WebServer\\Parsing.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WebServer\\WebServer.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\libraries\\WebServer\\detail\\mimetable.cpp.o" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553\\core\\core.a" -lgcc -lesp_websocket_client -lwpa2 -ldetection -lesp_https_server -lwps -lhal -lconsole -lpe -lsoc -lsdmmc -lpthread -llog -lesp_http_client -ljson -lmesh -lesp32-camera -lnet80211 -lwpa_supplicant -lc -lmqtt -lcxx -lesp_https_ota -lulp -lefuse -lpp -lmdns -lbt -lwpa -lspiffs -lheap -limage_util -lunity -lrtc -lmbedtls -lface_recognition -lnghttp -ljsmn -lopenssl -lcore -lfatfs -lm -lprotocomm -lsmartconfig -lxtensa-debug-module -ldl -lesp_event -lesp-tls -lfd -lespcoredump -lesp_http_server -lfr -lsmartconfig_ack -lwear_levelling -ltcp_transport -llwip -lphy -lvfs -lcoap -lesp32 -llibsodium -lbootloader_support -ldriver -lcoexist -lasio -lod -lmicro-ecc -lesp_ringbuf -ldetection_cat_face -lapp_update -lespnow -lface_detection -lapp_trace -lnewlib -lbtdm_app -lwifi_provisioning -lfreertos -lfreemodbus -lethernet -lnvs_flash -lspi_flash -lc_nano -lexpat -lfb_gfx -lprotobuf-c -lesp_adc_cal -ltcpip_adapter -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553/ESP32_Box031i.ino.elf"
"D:\\Arduino189\\portable\\packages\\esp32\\tools\\esptool_py\\3.0.0/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553/ESP32_Box031i.ino.bin" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553/ESP32_Box031i.ino.elf"
esptool.py v3.0-dev

"D:\\Arduino189\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/gen_esp32part.exe" -q "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553/partitions.csv" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_196553/ESP32_Box031i.ino.partitions.bin"
Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: D:\Arduino189\portable\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Nicht benutzt: D:\Arduino189\portable\sketchbook\libraries\SD
[... 42 lines cut ...]
Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Nicht benutzt: D:\Arduino\libraries\SD
 Nicht benutzt: D:\Arduino\portable\sketchbook\libraries\SD
Bibliothek TFT_HX8357 im Ordner: D:\Arduino\portable\sketchbook\libraries\TFT_HX8357 (legacy) wird verwendet
...
Bibliothek WebServer in Version 1.0 im Ordner: D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

```
d:/arduino/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8 bytes    ?
DRAM segment data does not fit     ?
collect2.exe: error: ld returned 1 exit status    ?

what does that mean and how to fix it?
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-05-24:
@dsyleixa we do not support 1.0.6 at all anymore. It does seem like you have way too many libs there. The success compile does not show the last line, which shows how much memory your sketch is using.

closed 2023-05-24 by me-no-dev

dsyleixa, 2023-05-25:
1.0.6 is crucial because 2.x  is not backwards-compatible at all.
as to IDE 1.8.9 which still works:

Der Sketch verwendet 861538 Bytes (65%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 124572 Bytes (38%) des dynamischen Speichers, 203108 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.

So there is enough memory left AFAIK.

dsyleixa, 2023-05-25:
1.0.6 is crucial because 2.x  is not backwards-compatible at all and cannot be compiled by either IDE any more.

As to IDE 1.8.9 which still works:

Der Sketch verwendet 861538 Bytes (65%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 124572 Bytes (38%) des dynamischen Speichers, 203108 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.

So there is enough memory left AFAIK.

However, It is far beyond me why a IDE upgrade would matter anyway.

me-no-dev, 2023-05-25:
> ESP32 core 1.0.6 is crucial because 2.x is not backwards-compatible at all and my Sketches cannot be compiled by either IDE any more.

I understand that your particular project compiles only with particular IDE and Core versions, but we can not and will not support anything 1.0.x related. It is very old, IDF that is inside is also not supported for quite some time.

To me is also very strange why one IDE version will work and the other not, but you should keep in mind that if you are getting in such situations, that means that your firmware is way too big/busy/memory intensive.

dsyleixa, 2023-05-25:
why memory-intensive? there is enough left using 1.8.9
```
Der Sketch verwendet 861538 Bytes (65%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 124572 Bytes (38%) des dynamischen Speichers, 203108 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
```
so why less and dont not work with .19 ??

dsyleixa, 2023-05-25:
BTW, the developers should really make sure that core updates are always absolutely backwards compatible in every respect!!