espressif/arduino-esp32 #8250: which alphanumerical keyboard to conect to ESP32 for using getc, scanf, getline?

dsyleixa, 2023-05-25:
### Related area

ESP32 keyboard

### Hardware specification

ESP32

### Is your feature request related to a problem?

which alphanumerical keyboard to conect to ESP32 for using getc, scanf, getline?

### Describe the solution you'd like

102 keyboard usage  like with common USB keyboards, stdio/iostream/stdin compatible, directly attached to PC or RaspberryPi (no WiFi or internal BT required)

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-05-25:
@dsyleixa - I'm sorry, but I didn't understand the question. 
Do you want to use a Keyboard attached to the computer that will be used to send data to the ESP32 using UART?
Please detail it further.

dsyleixa, 2023-05-25:
no, a sort of common 102 keyboard directly attached to the ESP32, for stand-allone usage.
Sorry for my poor English.

SuGlider, 2023-05-25:
ESP32 has no  USB HOST, thus, the only way to attach a keyboard directly to an ESP32 is by using a PS/2 keyboard that requires 2 GPIOs (DATA and CLK) plus VCC and GND.

There are a few libraries that can help with it:
https://github.com/techpaul/PS2KeyAdvanced   -- It has support to scanning most key codes.
https://github.com/michalhol/ps2kbdlib

SuGlider, 2023-05-25:
About supporting functions ( getc, scanf, getline, including stdio/iostream/stdin), it would require creating them by scanning the KeyCodes, forming the C strings and then passing it to the standard C functions.

It would be like having a PS/2 driver in a first low level layer and then a bounding with a C level equivalent functions created for this purpose.

dsyleixa, 2023-05-25:
as stated, directly using C/C++ getc, scanf, getline functions is crucial, at least the PS2KeyAdvanced is absolutely crappy.

SuGlider, 2023-05-25:
I see. At this point we will need community help to work on it due to lack of time.

dsyleixa, 2023-05-25:
perhaps possibly  also a USB-Hardware-Interface-Device, to be put in between the USB connector and the USB keyboard?

dsyleixa, 2023-05-25:
Thanks, but unfortuantely I only use Arduino IDE 1.8.x and arduino-esp32 API. OTOH, I don't recognize how ESP32 Soft USB HID Host Library https://github.com/tobozo/ESP32-USB-Soft-Host works with getc, scanf, getline exactly via Arduino IDE.

SuGlider, 2023-05-25:
Using Tobozo's library would need a software layer to support the standard C functions getc, scanf, getline etc.
@dsyleixa - you may try to reach @tobozo and ask for help.

@tobozo - can you add some comment to this talk?

dsyleixa, 2023-05-25:
BTW, forgot to mention: 
I'm visually impaired so I can't solder, so it really should be plug and play...

SuGlider, 2023-05-25:
> BTW, forgot to mention: I'm visually impaired so I can't solder, so it really should be plug and play...

For that case, maybe, it is better to use the S2 or S3. The DevKits have both USB (UART CP2102 and Native USB Host) plugs for connecting a OTG cable directly (no soldering). 

Or, maybe, using some protoboard and dupont cables?!

SuGlider, 2023-05-25:
OK. The possible solution seems to be using your Adafruit Feather ESP32, that seems to have at aleast 20 GPIOs, to connet the SPI TFT and use 2 GPIOs for the Tobozo's Soft USB Host library. All connected using protoboard and dupont cables.

It requires some coding for the `getc(), scanf() and getline()` functions to be integrated into the pipeline of the Soft USB Host library.

dsyleixa, 2023-05-25:
actually, tbh, I would like to connect the keyboard directly to the Feather USB port (used for Sketch upload and USB power)...
Or a BT dongle attached to it, and then using a BT keyboard...

dsyleixa, 2023-05-25:
actually, tbh, I would like to connect the keyboard directly to the Feather USB port (used for Sketch upload and USB power)...
Or a BT dongle attached to it, and then using a BT keyboard (plus a Y-switch or hub for external USB power cable)...

SuGlider, 2023-05-25:
> actually, tbh, I would like to connect the keyboard directly to the Feather USB port (used for Sketch upload and USB power)... Or a BT dongle attached to it, and then using a BT keyboard (plus a Y-switch or hub for external USB power cable)...

Sorry, but this is impossible given that the Feather USB port is actually an UART-USB chip. It can't work as USB HOST.

SuGlider, 2023-05-25:
Thanks @dsyleixa. 
I'll close this issue given that it won't be necessary anymore.