espressif/arduino-esp32 #8251: ESP32-S3 Serial.print() calls when not connected to USB cause delays.

sblantipodi, 2023-05-25:
### Board

Lolin ESP32-S3

### Device Description

Lolin ESP32-S3 and UM S3

### Hardware Configuration

no GPIO connected

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Win11

### Flash frequency

80MHzù

### PSRAM enabled

no

### Upload speed

115200

### Description

This is a duplicate issue of
https://github.com/espressif/arduino-esp32/issues/6983

the workaround described in that issue does not work on S3 but the issue is the same.

### Sketch

```cpp
#include <Wire.h>
#include <Adafruit_NeoPixel.h>

#define LED_PIN 6
#define NEOPIXEL_PIN LED_PIN
Adafruit_NeoPixel pixels(1, NEOPIXEL_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

}

void loop() {
  // put your main code here, to run repeatedly:

  pixels.begin();  // INITIALIZE NeoPixel strip object (REQUIRED)
  
  Serial.print("Red... ");
  // drawMessage("Test Red LED");
  pixels.setPixelColor(0, pixels.Color(64, 0, 0));
  pixels.show();
  delay(500);

    Serial.print("Green... ");
// drawMessage("Test Green LED");
  pixels.setPixelColor(0, pixels.Color(0, 64, 0));
  pixels.show();
  delay(500);

    Serial.print("Blue... ");
// drawMessage("Test Blue LED");
  pixels.setPixelColor(0, pixels.Color(0, 0, 64));
  pixels.show();
  delay(500);
  
    Serial.print("White... ");
// drawMessage("Test Whitish LED");
  pixels.setPixelColor(0, pixels.Color(64, 64, 64));
  pixels.show();
  delay(500);

  Serial.print("Off... ");
  pixels.clear();
  pixels.show();
  delay(500);
  Serial.println("Done. Looping");
}
```

### Debug Message

```plain
no debug msg
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sblantipodi, 2023-05-25:
I'm using espressif32@6.3 that uses 
Arduino Release v2.0.9 based on ESP-IDF v4.4.4

the issue is there, the fix reduces the lag but not eliminates it.

sblantipodi, 2023-05-25:
I am driving 100 LEDs at 100FPS with Luciferin.
https://github.com/sblantipodi/glow_worm_luciferin/tree/reduced_footprint

if I Serial.print() various messages every 10 seconds, the framerate goes down to an average of 85FPS.

if I attach the Serial console while driving the LEDs, the average framerate goes back to 100FPS.

This does not happen on S2.

SuGlider, 2023-05-25:
@sblantipodi - I can replicate the issue here when I use the HW CDC USB peripheral of the S3.
The S3 has 2 different CDC peripheral for the USB port: HW JTAG/CDC and USB HOST (the same of the S2).

There is a way to eliminate the delay completely from the HW JTAG/CDC USB port is by caling `Serial.setTxTimeoutMs(0);` before `Serial.begin(115200);`.

When using the USB HOST CDC port of the S3, the delay issue never happens, like in the S2.

``` cpp
void setup() {
  // put your setup code here, to run once:
  Serial.setTxTimeoutMs(0);
  Serial.begin(115200);
}
```

Please try it and let me know.

SuGlider, 2023-05-25:
You can also set the USB CDC peripheral that you want to use as your `Serial` in the sketch.
In order to do it, you need to select the right option in the IDE Menu:

1) HW JTAG/CDC USB:

USB Mode ==> Hardware CDC and JTAG
USB CDC On Boot ==> Enabled
Upload Mode ==> UART0 / Hardware CDC

2) USB OTG with TinyUSB (like in the ESP32-S2):

USB Mode ==> USB-OTG (TinyUSB)
USB CDC On Boot ==> Enabled
Upload Mode ==> USB-OTG CDC (TinyUSB)

This option may require the board to be manually reset after uploading a new sketch, by pressing the RESET or EN button of the board.

sblantipodi, 2023-05-25:
@SuGlider this fixed my issue when using HW JTAG/CDC USB.
thank you very much for the kind answer, I really appreciate it. 

closing the issue.

closed 2023-05-25 by sblantipodi