espressif/arduino-esp32 #8256: Using ESP32 with esp_himem.h on V 2.0.9 crash on boot

BennyShani, 2023-05-26:
### Board

ESP32 Dev Module with psram

### Device Description

Esp32-Wrover-B

### Hardware Configuration

No HW

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Linux and Windows

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

If I write a small program calling to esp_himem_get_phys_size() the program compiles but it crash on boot.
The same program works well in 1.0.6

### Sketch

```cpp
#include "esp_himem.h"

int c = 0;

void setup() {
  Serial.begin(115200);
  Serial.printf("Himem physical size:   %u\r\n", esp_himem_get_phys_size());
  Serial.printf("Himem free size:       %u\r\n", esp_himem_get_free_size());
}

void loop() {
  Serial.printf("--%d\r\n", c++);
  delay(1000);                      // wait for a second
}
```

### Debug Message

```plain
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (129) spiram: SPI RAM not initialized
abort() was called at PC 0x400dc6b8 on core 0
Backtrace: 0x4008352d:0x3ffe3ae0 0x400893ad:0x3ffe3b00 0x4008e5fd:0x3ffe3b20 0x400dc6b8:0x3ffe3ba0 0x400dc6f7:0x3ffe3bc0 0x400dc5e3:0x3ffe3be0 0x400d93a3:0x3ffe3c10 0x40082c4d:0x3ffe3c40 0x40079276:0x3ffe3c90 |<-CORRUPTED
ELF file SHA256: bd2121d07f2db85c
E (149) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

### Other Steps to Reproduce

Compile and upload the sketch

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-05-26:
Where are you getting esp_himem.h from?  In 2.0.9, you should use the himem library included from IDF 4.4 (`#include <himem.h>`).  The docs are at https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/system/himem.html

SuGlider, 2023-05-26:
@BennyShani  - The sketch example does not compile and build using Arduino Core 2.0.9.
I got this messages:
```
#warning esp_himem.h has been replaced by esp32/himem.h, please include esp32/himem.h instead

Linking everything together...
undefined reference to `esp_himem_get_phys_size'
undefined reference to `esp_himem_get_free_size'
```

SuGlider, 2023-05-26:
Arduino 2.0.x uses IDF 4.4.
In such case, as pointed by @lbernstone, it may be better to review it with the proper documentation that he mentioned.

> Where are you getting esp_himem.h from? In 2.0.9, you should use the himem library included from IDF 4.4 (`#include <himem.h>`). The docs are at https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/system/himem.html

SuGlider, 2023-05-26:
@BennyShani - As you said when calling esp_himen_ functions, it crashes.
This is because the PSRAM is managed in a different way with IDF 4.4 (used by Arduino Core 2.0.x).

In order to get the free PSRAM space available, please use the functions from `Esp.h` as pointed out by @mrengineer7777 
https://github.com/espressif/arduino-esp32/blob/82161bebe357b78d1343fa0878e17c1fc8c3b315/cores/esp32/Esp.h#L67-L77

SuGlider, 2023-05-26:
Arduino 2.0.9 uses `heap_caps_` functions and any `malloc()` will automatically use PSRAM if the size of the allocated block is bigger than `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384` (16KB) as defined in the `sdkconfig`. A lower than 16KB block is allocated in the HEAP.

lbernstone, 2023-05-26:
@suglider I don't have a wrover in front of me.  Can you try disabling psram and see if the himem functions work?  The himem library is generally a replacement for psram on devices with >4MB psram.

SuGlider, 2023-05-26:
When selecting the WROVER, the option for disabling the PSRAM doesn't show up.
When selecting the ESP32 Dev Module (but flashing a WROVER) and disabling the PSRAM in the menu, I get this:

```
E (124) spiram: SPI RAM not initialized

abort() was called at PC 0x400dc414 on core 0

Backtrace: 0x400833dd:0x3ffe3ae0 0x40087f2d:0x3ffe3b00 0x4008d17d:0x3ffe3b20 0x400dc414:0x3ffe3ba0 0x400dc42f:0x3ffe3bc0 0x400dc33f:0x3ffe3be0 0x400d9117:0x3ffe3c10 0x40082afd:0x3ffe3c40 0x40079276:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 4173d373e31f13bf

E (144) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

When enabling PSRAM, I get this:
```
E (130) spiram: SPI RAM not initialized

abort() was called at PC 0x400dc6b8 on core 0

Backtrace: 0x4008352d:0x3ffe3ae0 0x400893ad:0x3ffe3b00 0x4008e5fd:0x3ffe3b20 0x400dc6b8:0x3ffe3ba0 0x400dc6f7:0x3ffe3bc0 0x400dc5e3:0x3ffe3be0 0x400d93a3:0x3ffe3c10 0x40082c4d:0x3ffe3c40 0x40079276:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 9841458b9dd14000

E (150) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

SuGlider, 2023-05-26:
> @SuGlider I don't have a wrover in front of me. Can you try disabling psram and see if the himem functions work? The himem library is generally a replacement for psram on devices with >4MB psram.

As the results above, it doesn't work anyway...

SuGlider, 2023-05-26:
@BennyShani - Please try this sketch:

``` cpp
void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.printf("PSRAM Total Size: %d\n", ESP.getPsramSize());
  Serial.printf("PSRAM Free Space: %d\n", ESP.getFreePsram());
}

void loop() {
  // memory leaking for consuming the PSRAM!
  char *ptr = (char *) malloc(100 * 1024);
  for (int i = 0; i < 1024; i++) ptr[i] = i;
  
  Serial.printf("PSRAM Free Space: %d\n", ESP.getFreePsram());
  delay(500);
}
```

### Output:
```
[   461][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
PSRAM Total Size: 4192123
PSRAM Free Space: 4192123
PSRAM Free Space: 4089707
PSRAM Free Space: 3987291
PSRAM Free Space: 3884875
PSRAM Free Space: 3782459
PSRAM Free Space: 3680043
PSRAM Free Space: 3577627
PSRAM Free Space: 3475211
PSRAM Free Space: 3372795
PSRAM Free Space: 3270379
PSRAM Free Space: 3167963
PSRAM Free Space: 3065547
PSRAM Free Space: 2963131
PSRAM Free Space: 2860715
PSRAM Free Space: 2758299
PSRAM Free Space: 2655883
PSRAM Free Space: 2553467
PSRAM Free Space: 2451051
PSRAM Free Space: 2348635
PSRAM Free Space: 2246219
PSRAM Free Space: 2143803
PSRAM Free Space: 2041387
PSRAM Free Space: 1938971
```

BennyShani, 2023-05-26:
> 1.x and 2.x are based on different versions of IDF and are very different.
>
> Does one of the functions in Esp.h give you the information you want?
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h<https://eu-central-1.protection.sophos.com?d=github.com&u=aHR0cHM6Ly9naXRodWIuY29tL2VzcHJlc3NpZi9hcmR1aW5vLWVzcDMyL2Jsb2IvbWFzdGVyL2NvcmVzL2VzcDMyL0VzcC5o&p=m&i=NjQ2YjNlNDYwYmZiODAxMWY0Y2FlYjI0&t=clNka2VqZ2dOKzkwT2liSDdmVmhyN0lpdWs4Y0xIdmNpMnpmU0RPMDBsND0=&h=e76cb83838a446f0a6a388b341a9672d&s=AVNPUEhUT0NFTkNSWVBUSVY6K0T94M0wPyDMzKN9BocAHrXlbSFJ3THbbw-AzQ3TEw>

The code is a small part of a lager program. When ever I include the himem function it crash on boot.
Thank you
Benny

BennyShani, 2023-05-26:
I have started a clean install on a Linux machine.
The file is located in:
.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32/include/esp_hw_support/include/soc/esp_himem.h

Benny

BennyShani, 2023-05-26:
Hi

I am compiling it with 2.0.9
***@***.***

Benny

BennyShani, 2023-05-26:
Hi

I have disabled the PSRAM in the IDE but it did not solve the broblem

Benny

BennyShani, 2023-05-26:
Hi

This works for me but I need the himem functions, my application needs to use more the 4MByts accessible by the ps_malloc

Benny

SuGlider, 2023-05-26:
>This works for me but I need the himem functions, my application needs to use more the 4MByts accessible by the ps_malloc

No problem. Whatever size is available in PSRAM, it will work as well.

BennyShani, 2023-05-26:
I have 8 MByes of PSRAM but the normal ps_malloc can access only 4Mbytes. I need more memory so I am using the himem.

I have compiled the example in:
https://raw.githubusercontent.com/espressif/esp-idf/c2123056461ae265c4b5d59c9cc31d49b41df322/examples/system/himem/main/himem_example_main.c

And modified it to the Arduino environment with the same result, crash:
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (154) spiram: SPI RAM not initialized

abort() was called at PC 0x400dcfe4 on core 0

Backtrace: 0x40083541:0x3ffe3ae0 0x40089409:0x3ffe3b00 0x4008e659:0x3ffe3b20 0x400dcfe4:0x3ffe3ba0 0x400dd023:0x3ffe3bc0 0x400dc82b:0x3ffe3be0 0x400d95eb:0x3ffe3c10 0x40082c61:0x3ffe3c40 0x40079266:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 965cd3840ca70965

E (175) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

BennyShani, 2023-05-26:
I used the example from:
https://github.com/espressif/esp-idf/tree/c212305646/examples/system/himem
In the Arduino environment it crash but in the ESP-IDE it worked well.
I suspect that there is a problem in the Arduino configuration

SuGlider, 2023-05-26:
The problem here is that Arduino 2.0.x doesn't initalize the PSRAM. It is necessary to use Himen functions.
It would be necessary to move this project to Arduino as IDF Component and set all necessary configurations in the `sdkconfig`.

There are the missing setting in the Arduino 2.0.9 `sdkconfig`:
`CONFIG_SPIRAM_BOOT_INIT=y`
`CONFIG_SPIRAM_IGNORE_NOTFOUND=n`
`CONFIG_SPIRAM_MEMTEST=y`

SuGlider, 2023-05-26:
Arduino 2.0.x uses precompiled IDF libraries and the `sdkconfig` parameters can't be changed.
Therefore, it can't be used for himem and BANKSWITCH as you want.
It is necessary to build the Arduino project as `Arduino as Component` using the IDF tools.

SuGlider, 2023-05-26:
I tested a ESP32-S3-WROOM-2 that has 32MB Flash and 8MB PSRAM (both are OPI - octal SPI).
Because its is OPI, it has no 4MB limit as the ESP32.
I can allocate the whole 8MB from PSRAM with a single `malloc()`.

This may be an alternative for the project as well, but it means changing the SoC to the S3-WROOM-2

BennyShani, 2023-05-26:
Thank you very much,

This may be the best solution for me since it will eliminate the memory switching

Benny

SuGlider, 2023-05-26:
@BennyShani - 
These are the S3 with 8MB OPI PSRAM datasheets:
https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf
https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf

8MB OPI PSRAM DevKit Specification (look for the ESP32-S3-WROOM-1-N4R8|N8R8|N16R8 or the ESP32-S3-DevKitC-1U-N4R8|N8R8|N16R8 or ESP32-S3-DevKitC-1-N8R8|N16R8V|N32R8V):
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

SuGlider, 2023-05-26:
I just tested it and I can `malloc(8250032)` at once using the ESP32-S3 with 8MB OPI PSRAM.

``` cpp
void setup() {
#define M1 (1024 * 1024)
#define PSRAM_SIZE (M1 * 7 + 910000)  // about 8,250,000 bytes
  // ESP32-S3 with OPI 8MB  PSRAM!
  char *ptr = (char *) malloc(PSRAM_SIZE);
  if (ptr) {
    for (int i = 0; i < PSRAM_SIZE; i++) ptr[i] = i & 0xFF;
    for (int i = 0; i < PSRAM_SIZE; i++) {
      if (ptr[i] != (i & 0xFF)) Serial.printf("E%d ", i);
    }
  }
  
  Serial.printf("PSRAM Free Space: %d\n", ESP.getFreePsram());
}

void loop() {}
```

SuGlider, 2023-05-27:
Actually, I just read that the S3 can address up to 32MB of PSRAM, OPI or Quad SPI, it doesn't matter. 
The S2 can address up to 10.5MB of PSRAM. The C3 has no support to PSRAM.
Threfore, the need for HIMEM (4MB Pages) is for the ESP32 only.

BennyShani, 2023-05-27:
Thank you very much

Benny

Jason2866, 2023-05-27:
Can confirm S2 and S3 are working fine with PSRAM >4Mbyte without doing any changes in code (using since S3 is supported in Arduino)

SuGlider, 2023-05-27:
Closing the issue. Please feel free to reopen it whenever necessary.

closed 2023-05-27 by SuGlider