espressif/arduino-esp32 #8260: flash memory partition allocation

reltkaine, 2023-05-28:
### Board

lolin lite,lolin classic

### Device Description

flash memory is weirdly partition at a much lower memory like lolin lite were it should be 4MB which i confirm in ESP flash donload tools....but the option on the partition selection can only maximize around 2MB 

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

Arduino 2

### Operating System

window

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

un-utilize memory.....

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

used the esp32 flash download tool to verify flash memory size and check the file boards.txt to match flash size

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-28:
Most partition definitions support Ota, which divides the flash space roughly in half. E.g. https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv
Do you see an option in the ide to select partition type?

lbernstone, 2023-05-28:
https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html

mrengineer7777, 2023-05-30:
Does this answer your question?

VojtechBartoska, 2023-08-18:
closing as expired

closed 2023-08-18 by VojtechBartoska