espressif/arduino-esp32 #8261: Update.h refuses to change partition

Zombieschannel, 2023-05-28:
### Board

ESP32-S3

### Device Description

Attached to an SD card mount on pins D12 - D15an

### Hardware Configuration

D27 connected to a button, D12 - D15 connected to a SD card 

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

so I have a program on esp32 which loads another program from a sd card (I'm using Update.h)
immediately after booting into the new program, I do an `esp_ota_set_boot_partition` so that it can get back to the original program after another reboot
this usually all works... once
usually, when I upload a new sketch to the esp32 this process works once, if not, I get an error 3 (`UPDATE_ERROR_READ`), no idea why
every attempt after it working results in no errors or warnings whatsoever 
though after esp restarts, it boots into the same partition
after further inspection, it seems like when it works `esp_ota_get_boot_partition` returns app1 and when it doesn't work it returns app0
unfortunately, even after checking if `esp_ota_get_boot_partition` is set to app0 and setting the boot partition to app1 manually (also `esp_ota_set_boot_partition` returns 0 every time which should mean it should have changed the boot partition?) it still refuses to change the partition
not sure if I'm missing something, but as seen in the debug messages, it seems to correctly write the whole app (bytes written to match with the app size)

as mentioned in the documentation,
> If esp_ota_set_boot_partition() has been called, the partition which was set by that function will be returned.

though for some reason after "successfully" completing the upload, if the following code is called:
```
Serial.println(esp_ota_get_next_update_partition(NULL)->label);
Serial.println(esp_ota_set_boot_partition(esp_ota_get_next_update_partition(NULL)));
Serial.println(esp_ota_get_boot_partition()->label);
```
the output is this:
```
app1
0
app0
```
no idea why, the behavior is quite strange
I've been stuck on this for a while 😅 so literally any help is appreciated

### Sketch

```cpp
//Sketch 1 - Slightly modified SD_Update example
#include <Update.h>
#include <FS.h>
#include <SD.h>
#include <esp_ota_ops.h>

void updateFromFS()
{
   File updateBin = SD.open("/APP.BIN");
   if (updateBin)
   {
      size_t updateSize = updateBin.size();
      if (updateSize > 0)
      {
        if (Update.begin(updateSize))
        {      
          size_t written = Update.writeStream(updateBin);
          if (written == updateSize)
              Serial.println("Written : " + String(written) + " successfully");
          else
              Serial.println("Written only : " + String(written) + "/" + String(updateSize) + ". Retry?");
          if (Update.end())
          {
              Serial.println("OTA done!");
              if (Update.isFinished())
              {
                Serial.println("Update successfully completed. Rebooting.");
                Serial.println(esp_ota_get_boot_partition()->label);
                ESP.restart();
              }
              else
                Serial.println("Update not finished? Something went wrong!");
          }
          else
              Serial.println("Error Occurred. Error #: " + String(Update.getError()));
        }
        else
          Serial.println("Not enough space to begin OTA");
      }
      else
         Serial.println("Error, file is empty");
      updateBin.close();
   }
   else
      Serial.println("Could not load from sd root");
}

SPIClass* hspi;
void setup() {
  hspi = new SPIClass(HSPI);
  hspi->begin(14, 12, 13, 15);
   Serial.begin(115200);
   pinMode(27, INPUT);
}
void loop()
{
  if (!digitalRead(27))
  {
    if (!SD.begin(15, *hspi))
        Serial.println("Card Mount Failed");
    uint8_t cardType;
    cardType = SD.cardType();

    if (cardType == CARD_NONE)
        Serial.println("No SD_MMC card attached");
    else
        updateFromFS();
    delay(500);
  }
}

//Sketch 2 - the one on the sd card as a .bin file
#include <esp_ota_ops.h>
void setup()
{
  Serial.begin(115200);
  esp_err_t err = esp_ota_set_boot_partition(esp_ota_get_next_update_partition(NULL));
  if (err)
  {
    Serial.print("Failed to return to main partition: ");
    Serial.println(err);
  }
  pinMode(2, OUTPUT);
}
void loop()
{
  digitalWrite(2, HIGH);
  delay(100);
  digitalWrite(2, LOW);
  delay(100);
  digitalWrite(2, HIGH);
  delay(200);
  digitalWrite(2, LOW);
  delay(200);
  digitalWrite(2, HIGH);
  delay(100);
  digitalWrite(2, LOW);
  delay(1000);
}
```

### Debug Message

```plain
//Case 1 - error 3 (UPDATE_ERROR_READ) occurs
[ 21421][D][Updater.cpp:133] begin(): OTA Partition: app1
Written : 266464 successfully
Error Occurred. Error #: 3

//Case 2 - no error, everything works the first time after uploading a sketch to esp32, and every attempt after that still does not throw out any errors. It just does not boot into the correct partition?
//When it works
[  6574][D][Updater.cpp:133] begin(): OTA Partition: app1
Written : 266464 successfully
OTA done!
Update successfully completed. Rebooting.
app1

//When it does not work
[  4464][D][Updater.cpp:133] begin(): OTA Partition: app1
Written : 266464 successfully
OTA done!
Update successfully completed. Rebooting.
app0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-05-28:
Arduino has some "do the right thing" [code for updates included](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L219-L233).  You can add your own own override by defining a verifyRollbackLater function.

Zombieschannel, 2023-05-28:
Thanks for the response!
This code is inside a `#ifdef CONFIG_APP_ROLLBACK_ENABLE`, so does that mean if CONFIG_APP_ROLLBACK_ENABLE was not defined, there would not have been any issues? If yes, how can I remove this #define? I've seen the sdkconfig files and this is mentioned there however it says it should not be edited.

Zombieschannel, 2023-05-28:
> Arduino has some "do the right thing" [code for updates included](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L219-L233). You can add your own own override by defining a verifyRollbackLater function.

Wait, why would the issue be in this "do the right thing" code?

lbernstone, 2023-05-28:
CONFIG_XX macros are part of ESP-IDF.  This is compiled into the embedded sdk in arduino.  You can [rebuild the libraries](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html), or just define the function as I suggested.  I didn't look too closely at your code, so I don't know your flow.  That's why I included the link to the code, so you could see exactly what it is doing.  The code in initArduino, as the name would suggest, runs just before Setup().

Zombieschannel, 2023-05-28:
Thanks for the feedback!
Do you maybe know what is the cause of error 3 (UPDATE_ERROR_READ) that I mentioned earlier? Update.end() still fails randomly with this error even though Update.writeStream() returns the correct size of the file.

lbernstone, 2023-05-28:
There are a number of things that throw that error.  Put some log_d statements into [Updater.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L38-L58) and pin down what fails.

Zombieschannel, 2023-05-30:
I changed verifyOta in the second sketch which is uploaded from an sd card to always return false so that once uploaded it immediately returns back to the original program.
After some more testing, it seems like after uploading the main sketch from the Arduino ide, this updating always works from an sd card. After uploading, it restarts and goes to the second partition and since verifyOta returns false, it restarts back into the main program. Every time after that if I upload the second sketch from the sd card, everything I described already, happens normally. Though for some reason when the esp32 is in the main program and it loses power or if I press the reset button, there is like a 5% chance uploading from the sd card will still work. Either it says it uploaded everything but does not switch partitions or I get that error 3.
I still do not understand how the "do the right thing" code has anything to do with the sketch on the sd card when the issue is clearly in the main program which is doing the uploading.

Zombieschannel, 2023-05-31:
I am aware that the goal of OTA is to update the firmware over something like WiFi, so I'm not so sure if it should work completely fine in my case (using OTA as a way to have an "OS" that can run different applications in the second partition). 

Like are there any checks on whether the uploaded .bin file has previously been uploaded?

That could explain some part of the issue but still does not explain why `esp_ota_set_boot_partition` refuses to change the partition although the return code is 0 (success).

Since as I mentioned already, this uploading usually works the first time after uploading the sketch from the IDE, but if ESP32 has to restart in any way (power loss or manually pressing the reset button), it won't work anymore.

Zombieschannel, 2023-06-05:
Ok so the issue seems to be that I was using HSPI for the SD card (or just specific pins that did not work, not sure). No idea why, but using VSPI instead (which is the default) seems to work perfectly. Quite a strange issue.

closed 2023-06-05 by Zombieschannel