espressif/arduino-esp32 #8263: xtensa/config directory seems to be in the wrong place in ESP32 framework

roleroz, 2023-05-29:
### Board

esp32-s3

### Device Description

esp32-s3-devkitc-1, although this should happen with most ESP32 boards

### Hardware Configuration

Nothing outside of the dev board

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

Linux

### Flash frequency

Whatever

### PSRAM enabled

yes

### Upload speed

Whatever

### Description

When including Arduino.h in a file, IntelliSense highlights a problem with it (it cannot include ../hal.h, needed by Arduino.h). After following the whole transitive inclusion I realized that the problem is `framework-arduinoespressif32/tools/sdk/esp32s3/include/xtensa/esp32s3/include/xtensa/config/core.h` is trying to include both `../hal.h` and `../xtensa-versions.h`, which doesn't exist

While looking into this, I realized that it seems like the `xtensa/config` directory has a bogus second mentioning of `esp32s3` in the path name, as both expected files exist in `framework-arduinoespressif32/tools/sdk/esp32s3/include/xtensa/include/xtensa/` instead of `framework-arduinoespressif32/tools/sdk/esp32s3/include/xtensa/esp32s3/include/xtensa/` (see the extra `esp32s3`), and only `xtensa/config` is under this directory, all the rest of xtensa is outside of the `framework-arduinoespressif32/tools/sdk/esp32s3/include/xtensa/esp32s3/include/xtensa/` directory

I'm not sure how does the compiler find `../hal.h` and `../xtensa-versions.h`, but IntelliSense correctly cannot find them

Is this expected?, and if not, can we get this fixed?

Thanks

Joso

### Sketch

```cpp
#include <Arduino.h>

void setup() {}

void loop() {}
```

### Debug Message

```plain
[{
	"resource": "/home/joso/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/joso/git/lego_robotics/firmware/libs/gamepad/gamepad.cc).",
	"source": "C/C++",
	"startLineNumber": 33,
	"startColumn": 1,
	"endLineNumber": 33,
	"endColumn": 31
}]
[{
	"resource": "/home/joso/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"../hal.h\" (dependency of \"freertos/FreeRTOS.h\")",
	"source": "C/C++",
	"startLineNumber": 33,
	"startColumn": 1,
	"endLineNumber": 33,
	"endColumn": 31
}]
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-30:
>Is this expected?, and if not, can we get this fixed?

Yes, this is how it should be.
Fixing Intelisense may require configuring the correct include path in the VSCode setup.
https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference

SuGlider, 2023-05-30:
Maybe the PlatformIO github issues may help answering your question better.

roleroz, 2023-05-30:
How is this expected?, `core.h` is including files that don't exist (`../hal.h` is one of them)

SuGlider, 2023-05-30:
Is there any compilation error due to it?
As far as I know CMake can build its own include path and can find all necessary files.

The problem is only with Intelisense.

VojtechBartoska, 2023-08-18:
Closing as expired, if needed you can reopen.

closed 2023-08-18 by VojtechBartoska