espressif/arduino-esp32 #8265: E (340) esp_core_dump_flash: Core dump data check fX

davidbcf, 2023-05-30:
### Board

ESP32-C3-S2

### Device Description

ESP32-C3

### Hardware Configuration

nothing

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

win 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

961200

### Description

I have many esp32-c3 modules, but one of them work not normally, it seems cant save local paramems to flash, however others can do it with the same program.
the bad one i can reflash the program, however the result still bad.

### Sketch

```cpp
save_info(SYS_STATE,M_STATE,"configing");
save_info(SYS_STATE,M_STATIC_IP,"");
```

### Debug Message

```plain
[15:22:41.359]收←◆BLE disconnected!
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382164
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x24e4
SHA-256 comparison failed:
Calculated: 96afbbd20dbc7ff192673fa3f3deb5ddf1fc3246ecffab8fb4721155a22e1e6d
Expected: f5c4688f0d8b92118d0d83a48c5e632b32a44cca8081e4e081f4fdf86d0a195f
Attempting to boot anyway...
entry 0x403cc710

[15:22:41.700]收←◆E (340) esp_core_dump_flash: Core dump data check fX赩ER:0.0.3.8
loading wifi info
4354
4354
4354
wifi info load erro!
AeroSense Wavve_A0:76:4E:59:CA:68
E (390) BT_OSI: config_new unable to open NVS namespace 'bt_config.conf'

E (391) BT_OSI: config_save, err_code: 0x2

E (402) BT_OSI: config_save, err_code: 0x2

E (402) BT_OSI: config_save, err_code: 0x2

E (403) BT_OSI: config_save, err_code: 0x2

E (403) BT_OSI: config_save, err_code: 0x2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-05-30:
Could be a side effect of the not correct bootloader handling of the old core 2.0.2
Update to actual core 2.0.9 and try again. If the problem is still there most probably device is defect.

SuGlider, 2023-05-30:
Yes, it can be one of the Flash chips that had its access fixed recently.
@davidbcf - Please update to Arduino Core 2.0.9 and try again running the sketch.

closed 2023-08-18 by VojtechBartoska