espressif/arduino-esp32 #8267: M5stack - Update.h - invalid segment length 0xffffffff

Dan-Kiefer, 2023-05-30:
### Board

ESP32-D0WDQ6-V3

### Device Description

- M5stack Core Basic 2.6
- COM-LTE SIM7600G-H

### Hardware Configuration

Serial between ESP32 and SIM Module: RX=15, TX=13

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

With the M5stack Core Basic V2.6 and the SIM module SIM7600G, I successfully send data to a database on a server.
Now I would like to update the firmware via OTA, from that same server. While downloading the new code works fine, something goes wrong when writing to the flash of the ESP32, using 'Update.h'. (I have checked the bytes I received via OTA, and can safely say that the transfer is working correctly).
I always get the message: '**esp_image: invalid segment length 0xffffffff**'.

I have tried several boards without success.

Does somebody have an idea what I may be doing wrong?

Thanks and kind regards, Dan

### Sketch

```cpp
// Based on :
// https://github.com/felmue/MyM5StackExamples/tree/main/M5Stack/OTAoverLTE

// Based on TinyGSM - FileDownload example
// https://github.com/vshymanskyy/TinyGSM/blob/master/examples/FileDownload/FileDownload.ino

#define TINY_GSM_MODEM_SIM7600
#define UPDATE_SIZE_UNKNOWN 0xFFFFFFFF

#define PIN_MODEM_RX    15
#define PIN_MODEM_TX    13
#define GSM_PIN         "1234"

const char apn[]      = "internet";
const char gprsUser[] = "";
const char gprsPass[] = "";

const char server[]   = "myServer.com";

const int  port       = 80;
const char resource[] = "/Hello.bin";

#define SerialMon Serial
#define SerialAT Serial1

#if !defined(TINY_GSM_RX_BUFFER)
#define TINY_GSM_RX_BUFFER 1024
#endif

#define TINY_GSM_DEBUG SerialMon
#include <TinyGsmClient.h>
#include <Update.h>

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

bool bReboot = false;

void setup()
{
  SerialMon.begin(115200);
  delay(10);

  SerialAT.begin(115200, SERIAL_8N1, PIN_MODEM_RX, PIN_MODEM_TX);
  delay(2000);

  SerialMon.println("Initializing modem...");
  modem.restart();

  // Wait for unsolicited messages e.g.
  modem.testAT(30000);

  if(GSM_PIN && modem.getSimStatus() != 3) { modem.simUnlock(GSM_PIN); }
}

void loop()
{
  SerialMon.print(F("Waiting for network..."));
  if(!modem.waitForNetwork())
  {
    SerialMon.println(F(" fail"));
    delay(10000);
    return;
  }
  SerialMon.println(F(" success"));

  if(modem.isNetworkConnected()) { SerialMon.println(F("Network connected")); }

  // GPRS connection parameters are usually set after network registration
  SerialMon.print(F("Connecting to "));
  SerialMon.print(apn);
  if(!modem.gprsConnect(apn, gprsUser, gprsPass))
  {
    SerialMon.println(F(" fail"));
    delay(10000);
    return;
  }
  SerialMon.println(F(" success"));

  if(modem.isGprsConnected()) { SerialMon.println(F("GPRS connected")); }

  SerialMon.print(F("Connecting to "));
  SerialMon.print(server);
  if(!client.connect(server, port)) {
    SerialMon.println(F(" fail"));
    delay(10000);
    return;
  }
  SerialMon.println(F(" success"));

  client.print(String("GET ") + resource + " HTTP/1.0\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Connection: close\r\n\r\n");

  uint32_t timeElapsed = millis();

  SerialMon.println(F("Waiting for response header"));

  const uint32_t clientReadTimeout   = 60000;
  uint32_t       clientReadStartTime = millis();
  String         headerBuffer;
  bool           finishedHeader = false;
  uint32_t       contentLength  = 0;

  while(!finishedHeader)
  {
    int nlPos;

    if(client.available())
    {
      clientReadStartTime = millis();
      while(client.available())
      {
        char c = client.read();
        headerBuffer += c;

        if (c < 16)     {SerialMon.print('-');}
        if (isprint(c)) {SerialMon.print((char)c);}
        else            {SerialMon.print(' ');}        

        if(headerBuffer.indexOf(F("\r\n")) >= 0) break;
      }
    }
    else
    {
      if(millis() - clientReadStartTime > clientReadTimeout)
      {
        SerialMon.println(F(">>> Client Timeout !"));
        break;
      }
    }

    nlPos = headerBuffer.indexOf(F("\r\n"));

    if(nlPos > 0)
    {
      headerBuffer.toLowerCase();
      if(headerBuffer.startsWith(F("content-length:")))
      {
        contentLength = headerBuffer.substring(headerBuffer.indexOf(':') + 1).toInt();
      }
      headerBuffer.remove(0, nlPos + 2);
    }
    else if(nlPos == 0)
    {
      finishedHeader = true;
    }
  }

  if(Update.begin(contentLength)) {
    Serial.println(F("Update begin ok"));
  }
  else {
    Serial.println(F("Update begin not ok"));
    while (true) { delay(1); }
  }

  uint32_t readLength = 0;

  if(finishedHeader)
  {
    SerialMon.println(F("Reading response data"));
    clientReadStartTime = millis();

    while(readLength < contentLength && client.connected() &&
           millis() - clientReadStartTime < clientReadTimeout)
    {
      while(client.available())
      {    
        uint8_t c = client.read();
        Update.write(&c, 1);
        // Serial.print(c);
        // Serial.print(".");
        readLength++;
        clientReadStartTime = millis();
      }
    }
  }

  timeElapsed = millis() - timeElapsed;
  SerialMon.println();

  if(Update.end())
  {
    Serial.println("OTA done!");
    if(Update.isFinished()) {
      Serial.println("Update PASS");
      bReboot = true;
    }
    else                    {
      Serial.println("Update FAIL");
    }
  }
  else
  {
    Serial.println("Error Occurred. Error #: " + String(Update.getError()));
  }

  client.stop();
  SerialMon.println("Server disconnected");

  modem.gprsDisconnect();
  SerialMon.println("GPRS disconnected");

  float duration = float(timeElapsed) / 1000;

  SerialMon.println();
  SerialMon.print("Content-Length: ");
  SerialMon.println(contentLength);
  SerialMon.print("Actually read:  ");
  SerialMon.println(readLength);
  SerialMon.print("Duration:       ");
  SerialMon.print(duration);
  SerialMon.println("s");

  if(bReboot == true) ESP.restart();

  while (true) { delay(1); }
}
```

### Debug Message

```plain
Here is the output received on the serial monitor:
_Waiting for network... success
Network connected
Connecting to internet success
GPRS connected
Connecting to myServer.com success
Waiting for response header
HTTP/1.1 200 OK- - Date: Tue, 30 May 2023 05:58:02 GMT- - Server: Apache- - Upgrade: h2,h2c- - Connection: Upgrade, close- - Last-Modified: Thu, 25 May 2023 04:45:04 GMT- - ETag: "5d8e0-5fc7d47ba036e"- - Accept-Ranges: bytes- - Content-Length: 383456- - Content-Type: application/octet-stream- - - - Update begin ok
Reading response data
[73149] ### Closed:  0

E (73877) esp_image: invalid segment length 0xffffffff
Error Occurred. Error #: 9
Server disconnected
GPRS disconnected

Content-Length: 383456
Actually read:  383456
Duration:       54.20s_
```

### Other Steps to Reproduce

- Similar threads concerning 'invalid segment length' don't seem to relate to my problem.
- I fully erased the flash before using the sketch above, without success.
- I tried different sizes of firmware to download, without success.
- OTA works fine when using the ESP32's WiFi functionality. But I need the use of the SIM module.
- I use 'Default' as partitioning. Attempts with other partitionings have shown the same result.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-05-30:
Suggestion:

It may be fixed by assuring a correct content-length is supplied, by the HTTP Server, in the HTTP header.

lbernstone, 2023-05-30:
Why are you not getting either of the "Update begin" messages?

felmue, 2023-05-30:
Hello @lbernstone 

it's at the end of the long line starting with `HTTP/1.1 ...`. E.g. `Update begin ok`

Thanks
Felix

SuGlider, 2023-05-30:
> Hello SuGlider why do think the content-length is false? The length of the files declared in the header is exactly (383456) the one I get when I open the file with a hex editor (HxD). Could you please explain more in detail, why you think it may be false? Although I did not check all bytes, at least the first and the last several bytes are correctly transmitted. As the content-length also was equal to the one read by the hex-editor, I believed the transmission to be ok. What else could I do to further verify the content? Any suggestions welcome. Thanks and kind regards, Dan

I don't think that it is wrong or false. I just suggested that it could be a possible reason for the failure.
If you say it is correct, therefore, the issue is something else.

SuGlider, 2023-05-30:
@Dan-Kiefer - 
There was an issue with some Flash chip that has been fixed recently. 
Please try Arduino Core 2.0.9 that has an updated IDF version.

lbernstone, 2023-05-30:
As suglider says, use the latest version. Also, change the write line to `counterWrite += Update.write(&c, 1);` with a global var and a print statement at the end.

Dan-Kiefer, 2023-05-31:
@SuGlider 
I updated to esp32 core 2.0.9. Result is still the same. 
BTW: I now have three possible boards: 'esp32', 'ESP32 Arduino' and 'M5Stack Arduino'. I tried all of them with no success. 
Which one should really be used?

[image]

Dan-Kiefer, 2023-05-31:
@lbernstone 
I added the counter as you suggested. It shows the same value as the 'Content-Length' and the 'Actually read':

E (69909) esp_image: invalid segment length 0xffffffff
Error Occurred. Error #: 9
Server disconnected
GPRS disconnected

Content-Length: 383456
Actually read:  383456
Counter write:  383456
Duration:       50.26s

How should I interpret the 0xffffffff ? If it was some real value, that would at least provide some hint. Any suggestions?

Thanks for your time and help!
Kind regards, Dan

SuGlider, 2023-06-01:
> @SuGlider I updated to esp32 core 2.0.9. Result is still the same. BTW: I now have three possible boards: 'esp32', 'ESP32 Arduino' and 'M5Stack Arduino'. I tried all of them with no success. Which one should really be used?
> 
> [image]
> 
> Thanks for your time and help! Kind regards, Dan

My question is why do you have those 3 options... `esp32` is the installed ESP32 Arduino Core using the board manager.
The other 2 are installed in the sketchbook. Didn't you install it?

SuGlider, 2023-06-01:
@Dan-Kiefer - Question: if you replace the LTE Modem, in the code, by the native ESP32 WiFi radio, does it work?

SuGlider, 2023-06-01:
@Dan-Kiefer - Please try this in `setup()`:

``` cpp
void setup()
{
  SerialMon.begin(115200);
  delay(10);

  SerialAT.setRxBufferSize(4096);
  SerialAT.begin(115200, SERIAL_8N1, PIN_MODEM_RX, PIN_MODEM_TX);
  SerialAT.setRxFIFOFull(64);
  SerialAT.setRxTimeout(2); 
  delay(2000);

```

Dan-Kiefer, 2023-06-01:
> My question is why do you have those 3 options... `esp32` is the installed ESP32 Arduino Core using the board manager. The other 2 are installed in the sketchbook. Didn't you install it?

I am not sure about the 'history' of these boards. What I am sure is, that the 'esp32' one only appeared yesterday, after updating to core 2.0.9.

Dan-Kiefer, 2023-06-01:
> @Dan-Kiefer - Please try this in `setup()`:
> 
> ```c++
> void setup()
> {
>   SerialMon.begin(115200);
>   delay(10);
> 
>   SerialAT.setRxBufferSize(4096);
>   SerialAT.begin(115200, SERIAL_8N1, PIN_MODEM_RX, PIN_MODEM_TX);
>   SerialAT.setRxFIFOFull(64);
>   SerialAT.setRxTimeout(2); 
>   delay(2000);
> ```

Thank you so much for your help. But this also didn't change anything.

Dan-Kiefer, 2023-06-02:
I tried at home with my personal notebook. That works fine! So the program is definitely ok.
Back at work, it's the same error again. I will now try to find out the differences between these two work places. If successful, I will report back here.
Thanks to all, Dan

Dan-Kiefer, 2023-06-02:
I struggled the whole day to find out why it works fine at home, but not at work. I finally got to the conclusion that it has most likely nothing to do with the 'Update' library, but with a custom partition I am using, that seems to be incompatible with the board 'esp32 core 2.0.9', but **has no problems with the 'M5stack Arduino 2.0.6' board**.

Instead of my own code I reproduced the error with the M5Stack example code 'HelloWorld.ino'. Used without custom partition all works fine. But wih my custom partition, the esp core 2.0.9 goes crazy. I get tons of error messages like this:

_23:12:51.161 -> rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
23:12:51.161 -> configsip: 0, SPIWP:0xee
23:12:51.161 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:12:51.161 -> mode:DIO, clock div:1
23:12:51.161 -> load:0x3fff0030,len:1344
23:12:51.161 -> load:0x40078000,len:13924
23:12:51.161 -> ho 0 tail 12 room 4
23:12:51.161 -> load:0x40080400,len:3600
23:12:51.161 -> entry 0x400805f0
23:12:51.161 -> ets Jul 29 2019 12:21:46_

My custom partitions is as follows:
_# Name,   Type, SubType, Offset,  Size, Flags
nvs,data,nvs,36K,20K,
otadata,data,ota,56K,8K,
app0,app,ota_0,64K,3M,
app1,app,ota_1,,3M,
spiffs,data,spiffs,,4M,
nvs1,data,nvs,,20K,
coredump,data,coredump,,64k_

I am wondering if that's also the reason behind my problems with the OTA update. Is there something wrong with my custom partition? My custom partition is copied to the sketch's root directory.

Thanks for your further comments,
Dan

felmue, 2023-06-03:
Hello @Dan-Kiefer 

I am no expert when it comes to partition tables, but depending on the total flash size I'd say it probably doesn't fit.
```
# Name,  Type, SubType, Offset, Size, Flags
nvs,     data, nvs,     36K,    20K,         = 56K
otadata, data, ota,     56K,     8K,         = 64K
app0,    app,  ota_0,   64K,     3M,         =  3M  64K
app1,    app,  ota_1,      ,     3M,         =  6M  64K
spiffs,  data, spiffs,     ,     4M,         = 10M  64K
nvs1,    data, nvs,        ,    20K,         = 10M  84K
coredump,data, coredump,   ,    64k          = 10M 148K
```
So if the flash size is about 4M, then the first partition which gets truncated is 'ota_1' - that is the partition the first OTA update is written too. And normally when the end of the flash is reached the pointer wraps, which means it probably started to overwrite `nvs` and `otdata`.

Why that works for pre 2.0.9 version I don't know.

Thanks
Felix

Dan-Kiefer, 2023-06-03:
Hi @felmue 
According to [this specification](https://docs.m5stack.com/en/core/basic_v2.6) of the M5stack V2.6, the one I use, FLASH size is 16M. So I'd expect my partition to be ok. As for my understanding 'M5stack Arduino' is NOT pre 'esp32 2.0.9', but a different library made by M5Stack. Thanks for clarification if I'm missing something.

Thanks and kind regards, Dan

felmue, 2023-06-03:
Hello @Dan-Kiefer 

at this point in time I am out of ideas. Sorry.

Thanks
Felix

SuGlider, 2023-06-04:
Where can I find the github for `M5stack Arduino 2.0.6`?
Is it different from the Arduino 2.0.6 that can be downloaded using the ESP32 board wihtin the Arduino IDE board Manager?

Dan-Kiefer, 2023-06-04:
@SuGlider 
I found it by adding 'https://docs.m5stack.com/en/arduino/arduino_development' to the additional board manager url, according to the official M5stack Arduino Tutorial, [here](https://docs.m5stack.com/en/arduino/arduino_development).

Thanks, Dan

Dan-Kiefer, 2023-06-04:
I have to correct my last post. When I wrote it, I had the following boards in my Arduino IDE:
[image]

In order to verify my response given above, I removed the 'M5Stack Arduino', and added it by the link I posted. That gave me the following boards available:
[image]

So, while the index 2.0.6 is the same, the name is not. Also the error messages I get with the 'esp by Espressif Systems', are the same on the 'M5Stack by M5Stack official'.

Sadly, I can't tell anymore, where I got the only version that worked. I am really confused about all these different boards. 

As I won't be able to solve that boards problem, I will do more tests with different partitions.

Dan-Kiefer, 2023-06-04:
partition:
If I reduce ota0, ota1 and spiffs to 1M each, it works.
If I increase them again to 2M, it fails.

lbernstone, 2023-06-04:
Find esptool.exe in your file system.  Run from the command line `esptool.exe flash_id`.  This will tell you the flash size.

Dan-Kiefer, 2023-06-04:
Here's the result:

C:\>py -m esptool flash_id
esptool.py v4.6.1
Found 3 serial ports
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:21:84:93:e9:f0
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4018
**Detected flash size: 16MB**

lbernstone, 2023-06-04:
Sure doesn't seem like it is using your custom partition.  Can you choose one of the larger built-in partitioning schemes and see if works properly?  Have you put partitions.csv in the folder with your sketch?

Dan-Kiefer, 2023-06-05:
Why do you think it doesn't use my partition? If it works or fails depending on the values I put into the partitions.csv (see my earlier post), doesn't that prove the ketch uses my partition?
Yes, partitions.csv is in the folder of my sketch.

Dan-Kiefer, 2023-06-05:
I'm afraid I can tell with much certainty: **custom partitions only work up to about 4MB total size**. 

I checked all available built-in partitions (Default, No OTA, Minimal SPIFFS). They all work. Strangely there are only three choices displayed, althouh there are many more in the espressif tools folder... 

As soon as I use a 'partitions.csv' copied to the sketch folder, problems start. To make sure I didn't screw up my partition file, I copied existing ones from the espressif tools folder (Documents/Arduino/hardware/espresif/esp32/tools/partitions).

Take the 'no_ota.csv', it works:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x200000,
spiffs,   data, spiffs,  0x210000,0x1F0000,

Increase the size of the last partition slightly, to 0x200000, it FAILS.

Do I have to change the boards.txt when using custom partitions? May that be the cause?

Thanks, Dan

Dan-Kiefer, 2023-06-08:
Dear Espressif

Can you please confirm if you have been able to reproduce the problem, or if it is just not working for me.

Thanks and kind regards, Dan

Dan-Kiefer, 2023-06-08:
The problem with the partition is solved. I didn't expect that, in order to use all FLASH available on a device, you first have to change the boards.txt file.

After changing the flash size to 16MB it worked: 
m5stack-core-esp32.build.flash_size=16MB

Not yet sure this will also solve the original problem of this thread. I will come back  for reporting.

Dan-Kiefer, 2023-06-08:
Whatever the partition, the error 'invalid segment length 0xffffffff' still persists, when trying OTA.

Dan-Kiefer, 2023-06-09:
Espressif writes about the use of custom partition schemes:

[image]

IMHO, that is not sufficent. You have also to select 'Default' as your Partition Scheme, and adjust the flash size in the boards.txt. If another scheme is selected, its '.upload.maximum_size' will be used and not the one of the board.

Having said that, my original issue still waits for ideas. I am at my wits end.

Dan-Kiefer, 2023-06-29:
I finally got it.

On my work PC, there were two 'Update.h' libraries, one in the folder: _documents/Arduino/libraries_ and the other one in the folder: _AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.6\libraries\_. The Compiler was using the first library. However, when I removed it, things improved significantly.
I am unsure how I ended up with two different libraries. The non-functioning library had some lines commented out.

closed 2023-06-29 by Dan-Kiefer