espressif/arduino-esp32 #8272: Connection to hidden SSID impossible

milefork, 2023-05-30:
### Board

ESP32 Dev Module

### Device Description

Az-Delivery ESP32-WROMM-32

### Hardware Configuration

No

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Can't connect to hidden SSID.
I'm running a Unifi Wifi Setup. The AP is broadcasting serveral SSIDs. One of them is hidden. When I try to connect to the hidden SSID it fails. The other ones are working fine. All running on same Channel(1).
I'm trying it with the default Sketch.
[image]

[image]

At first I thought it might be related to https://github.com/espressif/esp-idf/issues/2989 but it seems that it's only affecting channel 12 to 13.

### Sketch

```cpp
/*
 *  This sketch shows the WiFi event usage
 *
*/

/*
* WiFi Events

0  ARDUINO_EVENT_WIFI_READY               < ESP32 WiFi ready
1  ARDUINO_EVENT_WIFI_SCAN_DONE                < ESP32 finish scanning AP
2  ARDUINO_EVENT_WIFI_STA_START                < ESP32 station start
3  ARDUINO_EVENT_WIFI_STA_STOP                 < ESP32 station stop
4  ARDUINO_EVENT_WIFI_STA_CONNECTED            < ESP32 station connected to AP
5  ARDUINO_EVENT_WIFI_STA_DISCONNECTED         < ESP32 station disconnected from AP
6  ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE      < the auth mode of AP connected by ESP32 station changed
7  ARDUINO_EVENT_WIFI_STA_GOT_IP               < ESP32 station got IP from connected AP
8  ARDUINO_EVENT_WIFI_STA_LOST_IP              < ESP32 station lost IP and the IP is reset to 0
9  ARDUINO_EVENT_WPS_ER_SUCCESS       < ESP32 station wps succeeds in enrollee mode
10 ARDUINO_EVENT_WPS_ER_FAILED        < ESP32 station wps fails in enrollee mode
11 ARDUINO_EVENT_WPS_ER_TIMEOUT       < ESP32 station wps timeout in enrollee mode
12 ARDUINO_EVENT_WPS_ER_PIN           < ESP32 station wps pin code in enrollee mode
13 ARDUINO_EVENT_WIFI_AP_START                 < ESP32 soft-AP start
14 ARDUINO_EVENT_WIFI_AP_STOP                  < ESP32 soft-AP stop
15 ARDUINO_EVENT_WIFI_AP_STACONNECTED          < a station connected to ESP32 soft-AP
16 ARDUINO_EVENT_WIFI_AP_STADISCONNECTED       < a station disconnected from ESP32 soft-AP
17 ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED         < ESP32 soft-AP assign an IP to a connected station
18 ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED        < Receive probe request packet in soft-AP interface
19 ARDUINO_EVENT_WIFI_AP_GOT_IP6               < ESP32 ap interface v6IP addr is preferred
19 ARDUINO_EVENT_WIFI_STA_GOT_IP6              < ESP32 station interface v6IP addr is preferred
20 ARDUINO_EVENT_ETH_START                < ESP32 ethernet start
21 ARDUINO_EVENT_ETH_STOP                 < ESP32 ethernet stop
22 ARDUINO_EVENT_ETH_CONNECTED            < ESP32 ethernet phy link up
23 ARDUINO_EVENT_ETH_DISCONNECTED         < ESP32 ethernet phy link down
24 ARDUINO_EVENT_ETH_GOT_IP               < ESP32 ethernet got IP from connected AP
19 ARDUINO_EVENT_ETH_GOT_IP6              < ESP32 ethernet interface v6IP addr is preferred
25 ARDUINO_EVENT_MAX
*/

#include <WiFi.h>

const char* ssid     = "A-SSID"; // Change this to your WiFi SSID
const char* password = "aPassword"; // Change this to your WiFi password

void WiFiEvent(WiFiEvent_t event)
{
    Serial.printf("[WiFi-event] event: %d\n", event);

    switch (event) {
        case ARDUINO_EVENT_WIFI_READY: 
            Serial.println("WiFi interface ready");
            break;
        case ARDUINO_EVENT_WIFI_SCAN_DONE:
            Serial.println("Completed scan for access points");
            break;
        case ARDUINO_EVENT_WIFI_STA_START:
            Serial.println("WiFi client started");
            break;
        case ARDUINO_EVENT_WIFI_STA_STOP:
            Serial.println("WiFi clients stopped");
            break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
            Serial.println("Connected to access point");
            break;
        case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
            Serial.println("Disconnected from WiFi access point");
            break;
        case ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE:
            Serial.println("Authentication mode of access point has changed");
            break;
        case ARDUINO_EVENT_WIFI_STA_GOT_IP:
            Serial.print("Obtained IP address: ");
            Serial.println(WiFi.localIP());
            break;
        case ARDUINO_EVENT_WIFI_STA_LOST_IP:
            Serial.println("Lost IP address and IP address is reset to 0");
            break;
        case ARDUINO_EVENT_WPS_ER_SUCCESS:
            Serial.println("WiFi Protected Setup (WPS): succeeded in enrollee mode");
            break;
        case ARDUINO_EVENT_WPS_ER_FAILED:
            Serial.println("WiFi Protected Setup (WPS): failed in enrollee mode");
            break;
        case ARDUINO_EVENT_WPS_ER_TIMEOUT:
            Serial.println("WiFi Protected Setup (WPS): timeout in enrollee mode");
            break;
        case ARDUINO_EVENT_WPS_ER_PIN:
            Serial.println("WiFi Protected Setup (WPS): pin code in enrollee mode");
            break;
        case ARDUINO_EVENT_WIFI_AP_START:
            Serial.println("WiFi access point started");
            break;
        case ARDUINO_EVENT_WIFI_AP_STOP:
            Serial.println("WiFi access point  stopped");
            break;
        case ARDUINO_EVENT_WIFI_AP_STACONNECTED:
            Serial.println("Client connected");
            break;
        case ARDUINO_EVENT_WIFI_AP_STADISCONNECTED:
            Serial.println("Client disconnected");
            break;
        case ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED:
            Serial.println("Assigned IP address to client");
            break;
        case ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED:
            Serial.println("Received probe request");
            break;
        case ARDUINO_EVENT_WIFI_AP_GOT_IP6:
            Serial.println("AP IPv6 is preferred");
            break;
        case ARDUINO_EVENT_WIFI_STA_GOT_IP6:
            Serial.println("STA IPv6 is preferred");
            break;
        case ARDUINO_EVENT_ETH_GOT_IP6:
            Serial.println("Ethernet IPv6 is preferred");
            break;
        case ARDUINO_EVENT_ETH_START:
            Serial.println("Ethernet started");
            break;
        case ARDUINO_EVENT_ETH_STOP:
            Serial.println("Ethernet stopped");
            break;
        case ARDUINO_EVENT_ETH_CONNECTED:
            Serial.println("Ethernet connected");
            break;
        case ARDUINO_EVENT_ETH_DISCONNECTED:
            Serial.println("Ethernet disconnected");
            break;
        case ARDUINO_EVENT_ETH_GOT_IP:
            Serial.println("Obtained IP address");
            break;
        default: break;
    }}

void WiFiGotIP(WiFiEvent_t event, WiFiEventInfo_t info)
{
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(IPAddress(info.got_ip.ip_info.ip.addr));
}

void setup()
{
    Serial.begin(115200);

    // delete old config
    WiFi.disconnect(true);

    delay(1000);

    // Examples of different ways to register wifi events
    WiFi.onEvent(WiFiEvent);
    WiFi.onEvent(WiFiGotIP, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_GOT_IP);
    WiFiEventId_t eventID = WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info){
        Serial.print("WiFi lost connection. Reason: ");
        Serial.println(info.wifi_sta_disconnected.reason);
    }, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);

    // Remove WiFi event
    Serial.print("WiFi Event ID: ");
    Serial.println(eventID);
    // WiFi.removeEvent(eventID);

    WiFi.begin(ssid, password);

    Serial.println();
    Serial.println();
    Serial.println("Wait for WiFi... ");
}

void loop()
{
    delay(1000);
}
```

### Debug Message

```plain
22:43:53.891 -> WiFi Event ID: 3
22:43:53.891 -> [WiFi-event] event: 0
22:43:53.891 -> WiFi interface ready
22:43:53.985 -> [WiFi-event] event: 2
22:43:53.985 -> WiFi client started
22:43:53.985 -> 
22:43:53.985 -> 
22:43:53.985 -> Wait for WiFi... 
22:43:56.412 -> [WiFi-event] event: 5
22:43:56.412 -> Disconnected from WiFi access point
22:43:56.412 -> WiFi lost connection. Reason: 201
22:43:58.836 -> [WiFi-event] event: 5
22:43:58.836 -> Disconnected from WiFi access point
22:43:58.836 -> WiFi lost connection. Reason: 201
22:44:01.222 -> [WiFi-event] event: 5
22:44:01.222 -> Disconnected from WiFi access point
22:44:01.222 -> WiFi lost connection. Reason: 201
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-06-04:
Interesting. Try erasing the wifi settings with WiFi.eraseAP(). 
 https://github.com/espressif/arduino-esp32/blob/72c41d09538663ebef80d29eb986cd5bc3395c2d/libraries/WiFi/src/WiFiSTA.cpp#L376 

On some wifi radios you have to specify the BSSID rather than SSID when connecting to a hidden network.

mrengineer7777, 2023-06-14:
Stale

milefork, 2023-06-18:
Hey @mrengineer7777 sorry for the late reply I had not had time to test it.
Right now I tried eraseAP with no sucess. And the following;

My Unifi ATH (replaced real bssid):

```
ath2      IEEE 802.11ng  ESSID:"A-SSID"
          Mode:Master  Frequency:2.437 GHz  Access Point: 52:2F:14:8D:63:0B

```

So right now I run the same sketch as above. Inserting this:

```cpp
@@ -40,7 +40,6 @@

 const char* ssid     = "A-SSID"; // Change this to your WiFi SSID
 const char* password = "aPassword"; // Change this to your WiFi password
+const uint8_t bssid[6] = {82,47,20,141,99,11};

 void WiFiEvent(WiFiEvent_t event)
@@ -159,7 +158,7 @@
     Serial.println(eventID);
     // WiFi.removeEvent(eventID);

+    WiFi.begin(ssid, password,0,bssid,true);
-    WiFi.begin(ssid, password);

     Serial.println();
     Serial.println();
```

And now I get this:

```
21:55:06.671 -> [ 34994][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
21:55:06.671 -> [WiFi-event] event: 5
21:55:06.671 -> Disconnected from WiFi access point
21:55:06.671 -> WiFi lost connection. Reason: 201
21:55:09.106 -> [ 37410][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
21:55:09.106 -> [WiFi-event] event: 5
21:55:09.106 -> Disconnected from WiFi access point
21:55:09.106 -> WiFi lost connection. Reason: 201
21:55:11.491 -> [ 39825][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
21:55:11.491 -> [WiFi-event] event: 5
21:55:11.491 -> Disconnected from WiFi access point
21:55:11.491 -> WiFi lost connection. Reason: 201
```

closed 2023-06-19 by milefork