espressif/arduino-esp32 #8275: ESP32-S3 ULP-FSM SENS_ register r/w problem

rhgoll, 2023-05-31:
### Board

Heltec ESP32 Lora (V3)

### Device Description

ESP32-S3 with SX1262 Lora module

### Hardware Configuration

nothing

### Version

other

### IDE Name

PlatformIO

### Operating System

Windows 11 22H2

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

If I read or set the `SENS_SAR_PERI_CLK_GATE_CONF_REG` from ULP code, it seems not working (but I get no error message).
If I read or set it from the main CPU, it works.

I am using ULP FSM C macro in PlatformIO Arduino framework.
The framework is a custom built from esp-idf 4.4.4, with added support of ULP:
```
CONFIG_ESP32S3_ULP_COPROC_ENABLED=y
CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM=512
```

This code snippet in the sketch is only a small part of an older big project running on ESP32.
The remaining ULP-FSM code (not showed here) ported from the ESP32 is working without problem, only this new register manipulation don't work.

### Sketch

```cpp
#include "Arduino.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_io_reg.h"
#include "soc/sens_reg.h"
#include "driver/rtc_io.h"
#include "esp32s3/ulp.h"

#define _v(val) ((uint32_t)(&val-RTC_SLOW_MEM))

RTC_NOINIT_ATTR uint32_t rtc_slow_debug_sens1;
RTC_NOINIT_ATTR uint32_t rtc_slow_debug_sens2;
RTC_NOINIT_ATTR uint32_t rtc_slow_debug_sens_cpu;
RTC_NOINIT_ATTR uint32_t rtc_slow_ulp_finished;

const ulp_insn_t program[] =
{
  I_MOVI(R3,_v(rtc_slow_debug_sens1)),
  I_RD_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG,0,15),
  I_STL(R0,R3,0),
  I_RD_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG,16,31),
  I_STH(R0,R3,0),
  I_WR_REG_BIT(SENS_SAR_PERI_CLK_GATE_CONF_REG,SENS_IOMUX_CLK_EN_S,1),
  I_MOVI(R3,_v(rtc_slow_debug_sens2)),
  I_RD_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG,0,15),
  I_STL(R0,R3,0),
  I_RD_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG,16,31),
  I_STH(R0,R3,0),
  I_MOVI(R3,_v(rtc_slow_ulp_finished)),
  I_MOVI(R0,1),
  I_STL(R0,R3,0),
  I_END(),
  I_HALT()
};

size_t load_addr = 0;
size_t size = sizeof(program)/sizeof(ulp_insn_t);

String toBinary(uint32_t num);

void setup()
{
  Serial.begin(115200);
  rtc_slow_ulp_finished=0;
  Serial.println("Starting ULP...");
  ulp_process_macros_and_load(load_addr, program, &size);
  ulp_run(load_addr);   
  delay(100); //wait for ulp to finish
  Serial.printf("ULP finished: %d\n",rtc_slow_ulp_finished);
  Serial.println("PERI_REG before ULP setting: "+toBinary(rtc_slow_debug_sens1));
  Serial.println("PERI_REG after ULP setting: "+toBinary(rtc_slow_debug_sens2));
  
  rtc_slow_debug_sens_cpu=READ_PERI_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG);
  Serial.println("PERI_REG before CPU setting: "+toBinary(rtc_slow_debug_sens_cpu));
  rtc_gpio_init(GPIO_NUM_4);
  rtc_slow_debug_sens_cpu=READ_PERI_REG(SENS_SAR_PERI_CLK_GATE_CONF_REG);
  Serial.println("PERI_REG after CPU setting: "+toBinary(rtc_slow_debug_sens_cpu)); 
}

void loop()
{
}

String toBinary(uint32_t num)
{
    String binary;
    for (uint8_t i=31;i>0;i--)
    {
      binary+=bitRead(num,i);
    }
    return binary;
}
```

### Debug Message

```plain
Starting ULP...
ULP finished: 1
PERI_REG before ULP setting: 0010000000000000000000000000000
PERI_REG after ULP setting: 0010000000000000000000000000000
PERI_REG before CPU setting: 0000000000000000000000000000000
PERI_REG after CPU setting: 1000000000000000000000000000000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

wnienhaus, 2023-08-12:
I can confirm this unexpected behaviour also on an ESP32-S2. Writing to `SENS_SAR_PERI_CLK_GATE_CONF_REG` from the ULP-FSM has no effect.

I wonder if this is expected behaviour on the S2/S3, but so far I cannot find anything in the Technical Reference Manual that would explain this behaviour as expected.

wnienhaus, 2023-08-12:
Actually correction: On the ESP32S2, the register is called `SENS_SAR_IO_MUX_CONF_REG` - and that's the one I cannot write to from the ULP. But looking at [this example in the ESP-IDF](https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/examples/system/ulp_fsm/ulp/main/ulp/pulse_cnt.S#L80) it should work (I would assume they that code wouldn't be there, if it's not meant to work).

Llgok, 2024-05-02:
Yes, I encountered the same issue with ULP counting on ESP32S3 using Arduino. It was unable to count properly. However, when using examples on ESP-IDF, the ULP functionality worked correctly.

wnienhaus, 2024-06-02:
Actually... what I encountered was a bug in an earlier ESP-IDF. It was fixed in June 2023 by this PR: https://github.com/espressif/esp-idf/pull/11652.

So, for anyone still running into this, make sure your Arduino toolchain uses an ESP-IDF later than June 2023 and it should work.

(For anyone using the [micropython-esp32-ulp](https://github.com/micropython/micropython-esp32-ulp) assembler, the same bug was fixed here: https://github.com/micropython/micropython-esp32-ulp/commit/ed28d27d2c5c853c514657701ce44ccf7ec8b6d6).

closed 2025-01-16 by Parsaabasi