espressif/arduino-esp32 #8281: Reboot due to SSL error "An invalid SSL record was received"

caipifrosch, 2023-06-01:
### Board

ESP32-WROOM

### Device Description

Custom board with ESP32-WROOM

### Hardware Configuration

nothing special related to this problem, WiFi / SSL related

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Log is reporting invalid SSL record and ESP reboots due to heap problem. 

Issue happens very random. Sometime within hours, but also happens after days of operation.

Error message... 
[4814633][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
...caused by WiFiClientSecure as server connection worked for more that one hour?

Anyone solved this problem or may have ideas how to further takle this?

Probably different issue, but reducing WiFi AP upload/download speed to 10kbps or below will successful mqtt server connection and trigger wdt. PubSubClient related...?

Thanks a lot in advance for your replies!

### Sketch

```cpp
#include "MQTT.h"
#include "measure.h"

WiFiClientSecure secureClient = WiFiClientSecure();
PubSubClient mqttClient(secureClient);
```

### Debug Message

```plain
15:06:00.360 > SYSTEM: Runtime: 0d 01h 20min 05sec Free Heap: 90.86 KB
15:06:00.362 >
15:06:09.845 > [4814632][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
15:06:09.854 > [4814633][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
15:06:09.864 > CORRUPT HEAP: Bad head at 0x3ffd9240. Expected 0xabba1234 got 0x3ffcd9ec
15:06:09.870 >
15:06:09.873 > assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)
15:06:09.879 >
15:06:09.879 >
15:06:09.879 > Backtrace: 0x40083e45:0x3ffbb400 0x40095301:0x3ffbb420 0x4009af69:0x3ffbb440 0x4009abdb:0x3ffbb570 0x40084309:0x3ffbb590 0x4008d3e1:0x3ffbb5b0 0x401dc122:0x3ffbb5d0 0x401cd8b2:0x3ffbb5f0 0x400f656a:0x3ffbb610 0x400f5268:0x3ffbb630 0x400f5349:0x3ffbb650 0x400f52b4:0x3ffbb670 0x4022c34a:0x3ffbb690 0x4022bd55:0x3ffbb6c0 0x400e0a05:0x3ffbb6e0 0x400d3a43:0x3ffbb700 0x4012edcf:0x3ffbb720
15:06:09.912 >
15:06:09.912 >
15:06:09.912 >
15:06:09.912 >
15:06:09.912 > ELF file SHA256: 89e4abb211f07ab3
15:06:09.916 >
15:06:10.328 > Rebooting...
```

### Other Steps to Reproduce

1. connect to wifi
2. connect to mqtt server with pubsubclient
3. wait for invalid ssl record error message

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-06-01:
It's possible ssl_client.cpp is freeing a bac pointer.
A couple days ago I caused a bad heap in my code by freeing a pointer to a local variable instead of a pointer to malloc'd memory. Oops.

WDT reset is likely unrelated.  If you run an HTTP call on the Arduino loop(), the HTTP call can be longer than the WDT timeout and cause a reboot.

I run my HTTPS calls on a separate thread/service.  That was a lot of work to get right!

It's also possible you are running out of stack space.  An easy way to test that is to use SET_LOOP_TASK_STACK_SIZE: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/ArduinoStackSize/ArduinoStackSize.ino

mrengineer7777, 2023-06-01:
Did some Googling.  The message "An invalid SSL record was received" originates from MBED_TLS.  It can be caused by the following:
1) Packet corruption at TCP level.
2) Low memory.
3) [Solved] Memory bug when using MBED-TLS with external PSRAM and low internal memory. https://github.com/espressif/esp-idf/issues/4127
4) Incorrect Date/Time setting(?) https://www.reddit.com/r/SwitchPirates/comments/mpk681/tinfoil_network_error_35_suddenly_stopped_working/
5)  Maximum TLS message length too short(?) https://github.com/espressif/arduino-esp32/issues/2733.  You have already posted on this issue.

This issue is very similar to #2733 (duplicate?).  Needs investigation.

caipifrosch, 2023-06-02:
Thanks a lot for your investigations and hints mrengineer7777!

Yes, this error message exists for a while. There were too many other things to solve in my project, but at this stage it keeps me from releasing a stable version.

I am already running the mqtt stuff in a separate task with highest priority and will further increase stack and mqtt buffer for testing. Lets see if this helps.

caipifrosch, 2023-06-05:
Quick update: Seems like a stack increase on mqtt task did the trick here. Unit is running for more than two days now.

Thanks a lot mrengineer7777 for this valuable hint!

closed 2023-06-08 by mrengineer7777

tigramaan, 2024-12-22:
make sure you don't call mqtt send and mqtt check from different contexts. For example from Ticker, from loop and from freertos task. all mqtt and ssl tasks should be called from one context strictly. Otherwise you will get heap fragmentation, ssl error, mqtt connection failure and board reboot. All at once or separately. These will be just sporadic errors.  And dont try to send mqtt or http messages from Ticker context. it works on hardware interrupts with limited resources and doesn't tolerate blocking operations.