espressif/arduino-esp32 #8286: Slider Call back function not getting invoked

rajsimiot, 2023-06-03:
### Board

ESP32

### Device Description

ESP32 - Rainmaker code with Arduino code.  

### Hardware Configuration

ESP32 DEV MODULE

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

Rainmaker

### PSRAM enabled

yes

### Upload speed

115200

### Description

The slider callback function is not getting invoked when the slider is changed from the rainmaker App. The slider is defined with my_switch11 variable.  

### Sketch

```cpp
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include <IRremote.h>
#include <esp_now.h>
#include <esp_rmaker_standard_types.h>
#include <DHT.h> 
//#include <SimpleTimer.h>

// ESP8266 Mac address (first peer)
uint8_t mac_peer1[] = {0xE8, 0xDB, 0x84, 0x99, 0x0A, 0x5D};
//uint8_t mac_peer1[] = {0xE8, 0x9F, 0x6D, 0x94, 0x2E, 0x64};

//2C:F4:32:17:4D:FA
//esp8266 nodemcu E8:DB:84:99:0A:5D
//esp12e adapter E8:9F:6D:94:2E:64

int i = 0;
typedef struct message {
   int data1;
   int load1;
   int cdata[3];
   //int dgreen;
   //int dblue;
};
//struct message inmessage;
struct message outmessage;
[... 1021 lines cut ...]
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
      //if (Timer.isReady()) { 
        //Serial.println("Sending Sensor Data");
        //sendSensor();
        //Timer.reset();      // Reset a second timer
      //}
    }

    ir_remote(); //IR remote Control
    manual_control();
}
```

### Debug Message

```plain
Not able to get values from the UI slider
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-04:
Please cross post at https://github.com/espressif/esp-rainmaker/issues and reference this issue.

SuGlider, 2023-06-27:
@sanketwadekar - Could you please take a look into this issue? Thanks!

sanketwadekar, 2023-07-03:
@rajsimiot The sketch you've uploaded is too big. Can you provide a short example that reproduces this error? :)