espressif/arduino-esp32 #8291: Esp32 causing data corruption when delay is introduced in the first time?

shariq-azim, 2023-06-05:
### Board

esp32 dev module

### Device Description

Its not connected to anything

### Hardware Configuration

Currently no hardware is connected

### Version

v2.0.9

### IDE Name

arduino ide 2.0

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello Everyone,
I am facing this weird issue where due to some reasons. The **delay** is causing issues with the free heap memory which indirectly is causing data issues. The following is the part in void loop of the sketch. **Releasing something around 4kb** and causing data corruption to a variable of **type const char*** which is the type accepted in pubsubclient's mqtt server parameter

I was expecting it to print IP which is printed before the delay is introduced.

The complete code is much much bigger and have not been excluded fully

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
//#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>

#include <PubSubClient.h>
//#include <HTTPClient.h>
#include <ArduinoJson.h>
#include <string.h>
#include <cstring>

#include <WiFiManager.h>
#include <Preferences.h>

#define DEBUG_SW 1

const char* dnshost = "shariqESP";
char ssid[25] = "xyz xyz";

char pass[20] = "alpha beta";

[... 453 lines cut ...]
//  Serial.println(mqttServer);

192.168.1.60
3
Heap size check loop2: 252872
192.168.1.60
Heap size check loop3: 256044
 ��?6�
Wifi status: Wifi Connected
Heap size after wifi connected and before delay: 256044
 ��?6�
Wifi Connected
Heap size after wifi connected: 256044
 ��?�
connected with MQTT
Heap size check wifi on pubsub connected: 256044
 ��?�
Heap size check LOOP wifi on pubsub connected: 256604

https://i.stack.imgur.com/XTvpJ.png
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

shariq-azim, 2023-06-06:
The code could be overwhelming, but the issue is observed in the logs of void loop...

everslick, 2023-06-16:
You are walking on thin ice by doing something like `strcpy(mqttServer, mqttServerPref.c_str());` while mqttServer is pointing to a `const char *`. You are generally messing with strcpy while dealing mostly with Strings, so I bet there is some mem corruption or a dangling pointer lurking in your code.

shariq-azim, 2023-06-22:
Whats your suggestion then @everslick ?

shariq-azim, 2023-06-22:
> ```c++
> if
> ```

Part  of  Void loop..
Just after the delay is introduced  before the
WiFi.status() != WL_CONNECTED part @ibernstone

aeonSolutions, 2023-07-20:
I had to spent some additional time debugging my smart hardware electronics for a home automation wall mount switch based on the ESP32 S3 ( see it on my GitHub [here](https://github.com/aeonSolutions/AeonLabs-Smart-Power-Switch-Outlet/blob/main/README.md) ) 

[![enter image description here][1]][1]

because I was able to upload the firmware code into it however when reading serial output it only displayed "gibberish" characters. 

On the internet I could not find the answer, so I had to work harder to find the solution. Common problems found when dealing with serial output errors are:
- poor UART ground
- absence of a 500 Ohm resistor on the TX line
- UART cable to long
- track coupling and track length mismatch

are all hardware issues that cause serial communication errors and malfunctioning. However, for the ESP32 there's one missing on the list above. 
On this particular hardware, I had incorrectly soldered the RTC crystal of 32,768MHz into the MCU crystal place which is for one of 40MHz. The result was, it is possible to do firmware upload using a 32,768Mhz crystal for the MCU however, and due to the incorrect clock frequency, serial output is no longer 115200bps and displays "gibberish" characters.   

**In summary make sure the crystal has the correct clock frequency value**.

  [1]: https://i.stack.imgur.com/wACFh.png

VojtechBartoska, 2023-08-17:
Closing this issue as answered, if needed you can reopen.

closed 2023-08-17 by VojtechBartoska

VojtechBartoska, 2023-08-18:
@shariq-azim Can be this ticket closed?