espressif/arduino-esp32 #8302: GNU99

podaen, 2023-06-09:
### Related area

N/A

### Hardware specification

N/

### Is your feature request related to a problem?

In

[platform.txt](https://github.com/espressif/arduino-esp32/blob/master/platform.txt#LL40C431-L41C1) 

GNU++11 is in use for compiler.c.flags.esp32

While 

compiler.cpp.flags.esp32 is using a linux sdk from 1999. Many Linux developers don't use those anymore. I hear now over a year it will change to a new version. 

### Describe the solution you'd like

Why not using C11?? same as for c++

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

podaen, 2023-06-09:
Maybe this is related

https://github.com/espressif/arduino-esp32/issues/8225

podaen, 2023-06-10:
I reached the question once on the IDF core and no body could give me a profound answer on the std version in use. I think it's imbedded in tool chain "xtensa-esp32-elf.exe" so nobody actually knows. And as I sad it keeps hanging on that very old version 1999. I got some answers like... oh it's there isn't much of difference. I would say this: How longer we wait, how difficult it is to step over from one to an other, because of the they way around that has been programmed that can't been changed for backwards compatibility.

I found this

https://github.com/espressif/gcc

and that's the one that is probably in use, not actually sure. 
Here is the tool to config it

https://github.com/espressif/esp-xtensaconfig-lib

This is a part of the ESP Arduino core... Still don't know what plugin and what version IDF is using (and what about all those other components that has been added the last years. You see what I am telling? Seems that this issue is has put on the long way.

https://github.com/espressif/esp-idf/releases/tag/v4.4.4

Not one md file that tells me what c or c++ is used and everything runs on it.

Jason2866, 2023-06-11:
The used toolchains for core 2.0.9 and IDF v4.4.4 are based on gcc v8.4.0 and built with crosstool-NG https://github.com/espressif/crosstool-NG/tree/esp-2021r2-patch5

podaen, 2023-06-11:
Yeah, it supports C++17. We are on c++11, but I didn't had issues with that in the past. Std flag is and the 8.4.0 support vectors so it have to be >C11. Still there is C99 in the platform.txt

lbernstone, 2023-06-12:
> I would say this: How longer we wait, how difficult it is to step over from one to an other, because of the they way around that has been programmed that can't been changed for backwards compatibility.

Are you saying that if you change this in platform.txt, there is a significant difference in compile time?

podaen, 2023-06-17:
Yeah, speeding up the compiling time for sure, but it demands a decrease of the clock speed if you want to use vectors, regarding there will be not nothing to handle those complex values. Still this is my opinion what it could be!

podaen, 2023-06-17:
Yeah, speeding up the compiling time for sure, but it demands a decrease of the clock speed if you want to use vectors, regarding there will be nothing to handle those complex values. Still this is my opinion what it could be!

podaen, 2023-08-07:
I got some news,

I did some research and this link below you see we now using [IDF 4.4](https://github.com/espressif/arduino-esp32/releases/tag/2.0.11) When we look at the IDF SDK [4.4.5](https://github.com/espressif/esp-idf/blob/cc71308e2fdce1d6d27fc52d39943f5d0fb83f35/tools/cmake/build.cmake#L110 
) It is using by default std=gnu99 and std=gnu++11 You could use a new version, but it is not recommended by IDF or the Arduino core.

When we look at [IDF 5.0](https://github.com/espressif/esp-idf/blob/5eddb9e0160db851c5ff9954d73deff8bf24cd23/tools/cmake/build.cmake#L126), we see finally some movement for C too 17 and for C++ to 20... finally not really! They quickly step over ( [IDF 5.1](https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/tools/cmake/build.cmake#L134
) ) to C17 and C++23. And this look promising out of my experience with the standard language, working with the Arduino-ESP core and external lib's.

The Arduino core 2.0.3 was first released May 4, 2022 when the IDF 4.4.1 was released on Apr 19, 2022. This looks promising! Let the count down begin!!!

:-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-):-)

Shaquum, 2023-10-02:
dobrze jest mieć pewność.

closed 2025-01-16 by Parsaabasi