espressif/arduino-esp32 #8306: Ledc randomly invert PWM signal output

API-rep, 2023-06-10:
### Board

ESP32 Dev module (Wroom-32)

### Device Description

Wired on readboard

### Hardware Configuration

RC servo / motor driver.

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 10 

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm working on a small library to drive DC and servo motor via ledc PWM output. To go deeper in timer management, I write it with ESP32 IDF regular syntax and structure. I've noticed that randomly, PWM output is inverted, regardless of the port use. I supposed a hardware trouble, but this problem happened with other ESP32 dev kit module.

Trouble was fixed by adding a _**GPIO.func_out_sel_cfg[SERVO_PIN].inv_sel = 0**_ on my code.

During my research to fix this problem, I noticed that the Espressif ESP32 IDF have a  _**output_invert_ flag**_ on its _**ledc_channel_config_t**_ struct. I tried to implement it on my code, but Arduino ESP32 port seemed to miss this one and return me an error during compilation. I'm not a c expert programmer, but I suspect a non initialization of this variable by Arduino port who can cause this random problem ? Could you check that deeper in your code please.

### Sketch

```cpp
PWM setup code (cleaned for readability)

    // create timer config structure
_timers_config[_pwmTimer] = new ledc_timer_config_t;
			
    // seeding timer config parameters 
_timers_config[_pwmTimer]->speed_mode		= LEDC_HIGH_SPEED_MODE;		// set timer mode
_timers_config[_pwmTimer]->freq_hz		= pwmFreq;			// set frequency of PWM signal
_timers_config[_pwmTimer]->timer_num		= (ledc_timer_t)_pwmTimer;	// set timer index
_timers_config[_pwmTimer]->clk_cfg		= LEDC_AUTO_CLK;		// set LEDC source clock
_timers_config[_pwmTimer]->duty_resolution	= (ledc_timer_bit_t)resBit;	// set duty resolution

    // create PWM channel config structure
ledc_channel_config = new ledc_channel_config_t

    // seeding PWM channel config parameters 
_ledc_channel_config->channel		= (ledc_channel_t)_pwmChannel;
_ledc_channel_config->duty		= 0;
_ledc_channel_config->hpoint		= 0;
_ledc_channel_config->gpio_num		= pwmPin;
_ledc_channel_config->speed_mode	= LEDC_HIGH_SPEED_MODE;
_ledc_channel_config->timer_sel		= (ledc_timer_t)_pwmTimer;
_ledc_channel_config->intr_type		= LEDC_INTR_DISABLE;
// _ledc_channel_config->output_invert	= 0;	// don't work on Arduino ESP32 IDE

    // start PWM signal on pwmPin.
ledc_channel_config(_ledc_channel_config);	

    // set ledc fade service on for ledc_set_duty_and_update/ledc_set_fade_time_and_start functions
ledc_fade_func_install(0);

...
    // output duty to port
ledc_set_duty_and_update(LEDC_HIGH_SPEED_MODE, (ledc_channel_t)_pwmChannel, dutyToWrite, 0);
```

### Debug Message

```plain
Error returned by compilator while output_invert struct flag is provided

error: 'struct ledc_channel_config_t' has no member named 'output_invert'
  _ledc_channel_config->output_invert = 0;
                        ^~~~~~~~~~~~~
exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-06-10:
Since you are using IDF API, I suggest you to file an issue in IDF's github repo. You can state that you are running ESP-IDF v4.4.4 with Arduino

closed 2023-06-10 by me-no-dev

API-rep, 2023-06-10:
Thank for your support. I'll do this ... and hope that they don't forward my request to arduino-esp32 because I use arduino IDE.  ;)