espressif/arduino-esp32 #8307: esp32s3 does not work I2C

rel1ct, 2023-06-11:
### Board

ESP32S3 Dev and ESP32 Dev

### Device Description

no

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Huge address pool scanning time in esp32c3 compared to esp32.
How can this be fixed?

### Sketch

```cpp
#include "Wire.h"
unsigned long t_start,t_stop;

void setup(){
  Serial.begin(115200);
  Wire.begin();
  Serial.println("\nI2C Scanner");
  scan();
}

void loop(){
  //
}

void scan(){
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  t_start = millis();
  for(address = 1; address < 127; address++ ){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
      nDevices++;
    }else if (error==4){
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");
  t_stop = millis();
  Serial.print("Time scanning: "); Serial.print((t_stop - t_start) / 1000.00); Serial.println(" sec.");
}
```

### Debug Message

```plain
esp32
20:19:00.320 -> [    53][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
20:19:00.320 -> 
20:19:00.320 -> I2C Scanner
20:19:00.320 -> Scanning...
20:19:00.374 -> No I2C devices found
20:19:00.374 -> Time scanning: 0.02 sec.

esp32s3
20:22:28.984 -> [   108][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=8 scl=9 freq=100000
20:22:28.984 -> 
20:22:28.984 -> I2C Scanner
20:22:28.984 -> Scanning...
20:24:34.992 -> No I2C devices found
20:24:34.992 -> Time scanning: 126.00 sec.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-11:
I'm unable to reproduce.  Do you have something on the i2c bus that might be stretching or causing spurious signals?

rel1ct, 2023-06-11:
Thanks, problem found (I2C bridge).

closed 2023-06-11 by rel1ct

mailonghua, 2023-06-15:
I had the same problem, can you share your solution