espressif/arduino-esp32 #8309: TWAI Timing Problems (cant receive messages below 5ms period)

gateseee, 2023-06-13:
### Board

ESP32 Wroom 

### Device Description

ESP32 Wroom V4 Devkit

### Hardware Configuration

GPIO33, GPIO32

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

unknown

### PSRAM enabled

yes

### Upload speed

unknown

### Description

I am currently trying to read an existing canbus (Car) with my ESP32 Rev3. I got it to work fine with an external MCP2515 shield using a default library.

But when I connect an transceiver and trying to use the internal TWAI controller built into ESP32 I am starting to get troubles. When receiving messages with at least >5ms period, it works fine but as soon the period gets under 5ms, its rapidly filling up the RX buffer and once its full, its starting to drop packets. When I set the RX buffer to a higher value, this only extend the time until the first error appears. It seems like the transceiver works fine but the twai_receive() function can't catch up. 

If the internal limitation is >5ms per message period, the CAN controller is pretty useless as in most modern cars canbus you have periods as low as 0.4ms per message.

To verify, that I'm not missing anything I setup two ESP32 and sending from A to B.

### Sketch

```cpp
#include "driver/twai.h"
twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(GPIO_NUM_33, GPIO_NUM_32, TWAI_MODE_LISTEN_ONLY);
    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS(); 
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

    //Install TWAI driver
    if (twai_driver_install(&g_config, &t_config, &f_config) == ESP_OK) {
        Serial.println("Driver installed\n");
    } else {
        Serial.println("Failed to install driver\n");
        return;
    }

    //Start TWAI driver
    if (twai_start() == ESP_OK) {
       Serial.println("Driver started\n");
    } else {
        Serial.println("Failed to start driver\n");
        return;
    }

twai_message_t message;

void loop() {
    while (twai_receive(&message, 0) == ESP_OK) {
        // process message (which will only arrive every 5ms
        Serial.printf(msgString, "STD CAN-ID: 0x%.3lX       DLC: %1d  Data:", message.identifier, message.data_length_code);
            for(byte i = 0; i<message.data_length_code; i++){
                        Serial.printf(" 0x%.2X", message.data[i]);
            }
        Serial.println(); 
    }

return; 
}
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-06-13:
@gateseee  - try this istead:    `while (twai_receive(&message, portMAX_DELAY) == ESP_OK) {`
Printing the messsage to the console takes time... maybe some messages are lost while it is printing... not sure.

gateseee, 2023-06-14:
@SuGlider unfortunatly this doesnt work. I bought some new sn65hvd230 transceivers (definitly genuine now) and tested it again, i got down to 2ms now but as soon i start to send lower intervals (1ms)  its struggling again. 

After doing some math - with 128bit packages and a speed of 500kbit, in THEORY under bust circumstances, a full load of the canbus would be ~ 3906 frames per second, which would be like 1 frame every 0.25milliseconds. 

So i am stumbling around somewhere at 100kbit now, nowhere near 500kbit. 

Thats sad.

SuGlider, 2023-06-27:
Last suggestion would be to create a separated task with High Priority just for reading the TWAI messages from the interface.
In this case it must be necessary to use `twai_receive(&message, portMAX_DELAY)` with a timeout in order to allow FreeRTOS to run other tasks, such as the Arduino `loop()`.

SuGlider, 2023-06-27:
For the Reading Task, this IDF Example may help:
https://github.com/espressif/esp-idf/blob/67552c31dac8cd94fb0d63192a538f4f984c5b6e/examples/peripherals/twai/twai_self_test/main/twai_self_test_example_main.c

SuGlider, 2023-06-27:
There is an IDF TWAI Listen Only example... you may want to try it and verify if it works better.
https://github.com/espressif/esp-idf/blob/67552c31dac8cd94fb0d63192a538f4f984c5b6e/examples/peripherals/twai/twai_network/twai_network_listen_only/main/twai_network_example_listen_only_main.c

AAJAY5, 2023-09-01:
Basic Calculation

**CAN**
- For one frame it takes 128 + 3 = 131bits
- So, 500kbit/sec = 500 * 1000 bits in sec
- So 1 frame time is = 131/500 ms ~= 0.26ms
- **At 500kbit/sec rate you receive 4 frames in 1ms.** 

**Serial**
- Consider 115200 baud rate 8N1
- 1 Char time  is  ~86us
```cpp
Serial.printf(msgString, "STD CAN-ID: 0x%.3lX       DLC: %1d  Data:", message.identifier, message.data_length_code);
            for(byte i = 0; i<message.data_length_code; i++){
                        Serial.printf(" 0x%.2X", message.data[i]);
            }
        Serial.println();
```
- As your log messages it ~takes 64 char to print log.
- **So, 1 Frame print time is 86*64 us = 5504us ~=5.5ms**

**Conclusion**
- 1 Frame receive time = 0.25ms
- 1 Frame print time is = 5ms
- You definitely miss lots of packets

**Refs**
- [link](https://electronics.stackexchange.com/questions/121329/whats-the-maximum-can-bus-frame-message-rate-at-125-kbit-s)

closed 2025-01-16 by Parsaabasi