espressif/arduino-esp32 #8311: HTTPClient sendRequest does not work if provided a stream object without size

that-one-arab, 2023-06-13:
### Board

ESP32 dev kit

### Device Description

ESP32 dev kit

### Hardware Configuration

SD card connected to default SPI pins
2 I2s devices connected, one of them to default I2S pins

### Version

latest master (checkout manually)

### IDE Name

Arduino ide

### Operating System

Ubuntu 20.04

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

In the HTTPClient library, we have an override `sendRequest` method.
This method takes 3 arguments, HTTP method, Stream object and stream size

Stream size is 0 by default.

If we call the method with all 3 arguments, the stream object is sent over HTTP without any issues

However if we call the method without assigning the 3rd parameter (size parameter), the HTTP request is returned early, without any binary being transmitted from the stream object.

I have examined the HTTPClient library code, I believe the culprit is somewhere in here:
```cpp
...
    int len = size;
    int bytesWritten = 0;

    if(len == 0) {
        len = -1;
    }

...
    if(buff) {
        // read all data from stream and send it to server
        while(connected() && (stream->available() > -1) && (len > 0 || len == -1)) {
...
```
variable `len` is equal to `size` (3rd function parameter) and `size` is 0 by default.
if `len` is 0 then we assign value `-1` to len

In the while loop, we do not exit out of the loop due to the 3rd loop condition `len > 0 || len == -1` causing the code to run infinitely

I do not have enough knowledge and experience to debug this any further...

### Sketch

```cpp
#include "SD.h"
#include <SPI.h>
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";
const char* url = "YOUR_URL";

File file;

void setup() {
  Serial.begin(115200);

  SD.begin(5);

  // Setup Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1500);
    Serial.println("Connecting to WiFi...");
  }

  file = SD.open("/input.wav", FILE_READ);

  // Send POST request
  HTTPClient http;

  http.begin(url);
  http.addHeader("Content-Type", "application/octet-stream");
  int httpCode = http.sendRequest("POST", &file);

  if (httpCode > 0) {
    Serial.println("success");
  } else {
    Serial.printf("Failed to send POST request, error: %s\n", http.errorToString(httpCode).c_str());
  }

  http.end();
}

void loop() {
}
```

### Debug Message

```plain
No debug messages
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-06-13:
@that-one-arab - I was looking into the source code and it says that when `size` is zero, it sends nothing, which is the case when not assigning the thrird parameter in `sendrequest()`. 
This method may be used with zero size, but it makes sense when sending a GET. Sending POST requires using the correct content size.

https://github.com/espressif/arduino-esp32/blob/bad5af9e1c7768cff8da98fda3625b6454035d6b/libraries/HTTPClient/src/HTTPClient.cpp#L578-L585

SuGlider, 2023-06-13:
You can use something like `int httpCode = http.sendRequest("POST", &file, file.size());` to make it work.

that-one-arab, 2023-06-13:
> You can use something like `int httpCode = http.sendRequest("POST", &file, file.size());` to make it work.

Thank you for your time. Yes indeed the method works as expected when supplied the 3rd size argument, but are you sure about content length being a requirement when sending an http request using post method? I am unsure myself, I'm trying to stream live i2s data over HTTP and need to send data over remotely without knowing the size of it beforehand.

SuGlider, 2023-06-13:
`HTTPClient` is meant to be used within RESTful API. 
It is not clear to me why would it be necessary to refactor the code.

SuGlider, 2023-06-13:
@that-one-arab - OK, I was looking into the wrong method... You want the `int HTTPClient::sendRequest(const char * type, Stream * stream, size_t size)` that works on top of a reading stream.

I'll test it.

SuGlider, 2023-06-13:
@that-one-arab - could you please set `Debug Core Level: "Debug"`, run your example and post there the console output with the DEBUG messages? Thanks.

VojtechBartoska, 2023-08-18:
Closing this ticket as expired due to no answer, if needed you can reopen.

closed 2023-08-18 by VojtechBartoska

mengxiyou, 2024-06-06:
That's because http request need content-length, otherwise, chunked is needed.
I fixed this problem as follow:
[image]
[image]
[image]
note. I just test my case, the code has not been thoroughly validated.