espressif/arduino-esp32 #8317: WiFi and USB problem with CH9102 chip

Majubs, 2023-06-14:
### Board

ESP32-WROOM-32

### Device Description

ESP32 DevKitV1 (can't find manufacturer) with CH9102x USB-Serial chip

### Hardware Configuration

No.

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Ubuntu 22.04

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I got a ESP32 development board with a different USB-Serial converter, CH9102x, and I'm having trouble making it work. More specifically, I can't make WiFi and serial output work together. I've tried a bunch of different things, so I'll just list what I got so far:

1. Running code that prints to Serial and does NOT use WiFi works normally, eg. serial output is correct
2. Running code that both prints to Serial and uses WiFi does not. The serial output sometimes becomes garbage (as if baudrate was wrong), and the board keeps resetting, here is an example of the output (the first 2 lines are my output)
```bash
================================ SETUP ================================
Calling WiFi.begin
Disconnected (device reports readiness to read but returned no data (device disconnected or multiple access on port?))
Reconnecting to /dev/ttyACM0 ...         Connected!
```
3. Tried to run example code from Arduino IDE and ESP-IDF for a simple WiFiScan script, but results were the same as above
4. I've found similar issues [espressif/esp-idf#8046](https://github.com/espressif/esp-idf/issues/8046) and #6264, but the solution did not work. I've confirmed that the version I'm has the solution to the issue in [espressif/arduino-esp32#6287](https://github.com/espressif/arduino-esp32/pull/6287), but it seems it's only for ESP32-C3 board, and I've tried to manually add the workaround from [this comment](https://github.com/espressif/arduino-esp32/issues/6264#issuecomment-1040147331), but I get the following error:
```
main.cpp:24: undefined reference to `phy_bbpll_en_usb'
```
5. Finally, I tested on Windows 11, and to my surprise, it works normally, without even installing any drivers

My best guess so far is that it has something to do with the CH9102x chip and the Linux driver, but I'm not sure how to debug or work around it.

### Sketch

```cpp
void setup()
{
    Serial.begin(115200);

    // Any code that has Serial and WiFi
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}
```

### Debug Message

```plain
# with build_type = debug and monitor_filters = esp32_exception_decoder
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-14:
You tested this on Win11 on the same machine?  This sounds very much like it could be a power issue.

Majubs, 2023-06-16:
The only visible information on the back is "ESP32 DEVKITV1". For reference, what the back of the board looks like: [here](https://ecadio.com/image/cache/catalog/nodemcu/esp32-devkit/jual-esp32-devkit-800x800.jpg). 

In the issue you mentioned, some said they tried to contact DOIT to ask about this board, but got no response, so I believe it's DOIT, but don't know for sure.

SuGlider, 2023-06-27:
There is no known issue related to ESP32, UART and WiFi. 
It shall be, as already written here, a Linux issue related to the USB CH9102 driver.

VojtechBartoska, 2023-08-18:
closing as no action needed.

closed 2023-08-18 by VojtechBartoska