espressif/arduino-esp32 #8318: Possible bug WebServer::handleClient method

supersjellie, 2023-06-15:
### Board

ESP32C3

### Device Description

Different custom boards communication by digital pins and/or I2C

### Hardware Configuration

Switches/sensors

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows`10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

About 5 to 10% of the calls to the (default) webserver fails. The issue also occurs on the ESP8266 Arduino (same code for both devices). So this is a copy of https://github.com/esp8266/Arduino/issues/8941 . Making sure the same object is used solves it (not even a single timeout anymore in the (calling) openHAB log. Details check above link.

Soon a PR will be added.

### Sketch

```cpp
Details check above link.
```

### Debug Message

```plain
Details check above link.
```

### Other Steps to Reproduce

Details check above link.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

Jason2866, 2025-03-12:
Closing since the related PR https://github.com/espressif/arduino-esp32/pull/8319 is merged

closed 2025-03-12 by Jason2866