espressif/arduino-esp32 #8320: ESP32-S default upload speed in Arduino IDE is too high, resulting in errors

fdb, 2023-06-15:
### Board

NodeMCU ESP32-S

### Device Description

NodeMCU ESP32-S

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 13.4 Ventura

### Flash frequency

??

### PSRAM enabled

yes

### Upload speed

921600

### Description

When trying to upload to a NodeMCU ESP32-S on Mac M1, the default upload speed of 921,600 is too high resulting in errors like this:

```
A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).
```
Or this:
```
A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 7 bytes)).
```

This can confuse new users (like me) who don't know which default settings they have to tweak in order to upload.

I tested this with two NodeMCU ESP32-S.

Setting the upload rate to a lower value (e.g. 460800) fixes the error.

### To reproduce

- Open the Basics>Blink Example
- Select the NodeMCU32-S board and Serial Port (USB)
- Choose Upload

### Sketch

```cpp
This happens with any script, e.g. the default Blink script.
```

### Debug Message

```plain
Here is the full output:

Sketch uses 235937 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22056 bytes (6%) of dynamic memory, leaving 305624 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-4140
Connecting......
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:e5:7c:db:1c:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 8 bytes)).
Failed uploading: uploading error: exit status 2
````
```

### Other Steps to Reproduce

If I lower the baud rate to 460800, uploading works normally:

```
Sketch uses 235937 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22056 bytes (6%) of dynamic memory, leaving 305624 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-4140
Connecting.....
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:e5:7c:db:1c:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00049fff...
Compressed 17520 bytes to 12170...
Writing at 0x00001000... (100 %)
Wrote 17520 bytes (12170 compressed) at 0x00001000 in 0.7 seconds (effective 213.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 267.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 400.8 kbit/s)...
Hash of data verified.
Compressed 236304 bytes to 129987...
Writing at 0x00010000... (12 %)
Writing at 0x0001e222... (25 %)
Writing at 0x0002421b... (37 %)
Writing at 0x000293fa... (50 %)
Writing at 0x0002ea8c... (62 %)
Writing at 0x00036fdb... (75 %)
Writing at 0x0003f0ee... (87 %)
Writing at 0x000446ea... (100 %)
Wrote 236304 bytes (129987 compressed) at 0x00010000 in 3.6 seconds (effective 527.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-06-15:
Can another user please verify this? If so, we can adjust the default upload speed

mrengineer7777, 2023-06-16:
@me-no-dev
I had the same problem when using esptool.py directly.  I use "460800" when flashing on an older computer, which works most of the time.

For development I use PlatformIO.  Whatever baud rate they set always works.  Of course that's also a faster computer.

I've seen other users report issues with programming.  I've recommended 460800 baud to them, but haven't heard back if that solved their issue.

I feel like the default baud rate for programming may have changed somewhere around 2.0.6-2.0.7?

SuGlider, 2023-06-20:
@me-no-dev -I don't see 460800 upload speed option in the Arduino IDE Menu.

I maybe add a PR to add the 460800 upload speed option and set it as default... Let me know.

SuGlider, 2023-06-20:
I see that 460800 is available in MacOS and Linux only. Not for Windows.

HabibMCS, 2023-06-21:
I am also stuck in same issue while uploading sketch to ESP32 Lilygo display-S3, tried everything did not work, also checked every upload speed. There is no upload speed 460800 in windows i think

me-no-dev, 2023-06-21:
Funny enough I can confirm this issue. Probably some layout is causing problems, even though baudrate in USB context does not mean anything. We could probably add 460800 to the menu for Windows. It's not there, because Windows did not support it.

HabibMCS, 2023-06-21:
> > I am also stuck in same issue while uploading sketch to ESP32 Lilygo display-S3, tried everything did not work, also checked every upload speed. There is no upload speed 460800 in windows i think
> 
> This may be a different issue. Please try setting the ESP32 Lilygo display-S3 into `Download Mode` by holding the BOOT button pressed and then pulsing the RESET button. Do it right before uploading a new sketch.
> 
> In Arduino IDE please consider using the `USB Mode` in `Hardware CDC and JTAG` mode and also enable the `USB CDC on Boot` option. Using the `Hardware CDC and JTAG` USB mode will make it upload a lot easier.
> 
> In the `USB-OTG (TinyUSB)` mode, it may require the board to enter in the `Download Mode` by using the BOOT and RESET buttons whenever uploading doesn't work.
> 
> Some reference: https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html https://github.com/Xinyuan-LilyGO/T-Display-S3

I will try these solutions hope it works, Thanks for your quick response.

mmannes, 2023-06-22:
I had the same problem on macOS and lowering the upload speed to 460800 solved the problem.

VojtechBartoska, 2023-07-19:
> Funny enough I can confirm this issue. Probably some layout is causing problems, even though baudrate in USB context does not mean anything. We could probably add 460800 to the menu for Windows. It's not there, because Windows did not support it.

Ok, assigning to you @me-no-dev

satyamedh, 2023-10-25:
Hello! Any updates about this?

Waniusza, 2023-11-18:
I do confirm issue.

Spend few hiurs looking for reason of:
`A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).`

It was enough to low down UploadSpeed!

richardloxley, 2023-12-01:
Same issue with me, MacOS Sonoma 14.1.1, Apple Silicon M1, uploading to two different ESP32 boards at 921600 fails, but 460800 works.  Stumped me for a long time!

CloudyDory, 2024-01-18:
> Same issue with me, MacOS Sonoma 14.1.1, Apple Silicon M1, uploading to two different ESP32 boards at 921600 fails, but 460800 works. Stumped me for a long time!

@richardloxley Hi, do you know how to change the upload speed in Arduino IDE for Mac? I am facing a similar issue, but I am not sure how to change the upload speed from IDE. It seems that the menu bar on Windows is missing on Mac.

CloudyDory, 2024-01-18:
> @richardloxley Hi, do you know how to change the upload speed in Arduino IDE for Mac? I am facing a similar issue, but I am not sure how to change the upload speed from IDE. It seems that the menu [...]

Got it. The menu bar is on top of the screen 😂.

VojtechBartoska, 2024-02-20:
Is it doable to fix this for 3.0.0 @me-no-dev?

Jason2866, 2024-02-28:
@VojtechBartoska The fix is done with #9277 or do you mean an other issue?

P-R-O-C-H-Y, 2024-02-28:
This was supposed to be closed by merging the fix #9277. So I am closing it now.

closed 2024-02-28 by P-R-O-C-H-Y

aminzakizebarjad, 2024-05-25:
I had the same problem. as below:
`A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).`
My board was not the original ESP32-s3-devkitc-1 or  ESP32-s3-devkitm-1, it was a third party Chinese board. The name at the back of the board was yd-esp32-23 and it had a github page all in Chinese . At the top of the repository it had a driver for the usb-serial module embedded on the board which was ch343. I installed the .EXE file provided and problem was solved.
Amazingly before installing the driver I could get the default pre-programmed firmware serial on my terminal, but Uploading was not successful beforehand.

Jason2866, 2024-12-28:
@robertlipe Was a Windows user from the begining and switched to Mac when the M1 appeared. Maybe Mac Users are used to "it just works". Since I was used to nothing works without installing a driver, this was not a big thing that for this Serial chip a driver needs to be installed to get it working with all features.

Jason2866, 2024-12-28:
@robertlipe There is still something fishy in espressif code. I am flashing every esp32x with 1.5M without any issues (with all USB serial converter chips out there).
BUT I never managed to **read** any esp32x faster than 115200. Weird since reading is way more "work" for the flash chip than writing.

Jason2866, 2024-12-28:
`/dev/cu.usbmodem1101` is using the espressif inbuilt "USB modem" Here reading works fast without using or "respecting" speed settings. It uses whatever speed.
I am speaking from external USB Serial converter chips like the wch340. Try to ready a bigger flash area with higher speed than 115200. It will end in an error.

Jason2866, 2024-12-28:
Agree littering this issue. We can chat in pioarduino Discord https://discord.gg/Nutz9crnZr