espressif/arduino-esp32 #8322: ESP32-S3-DevKitC-1-N16R8V does not restart after unplugging

ReyWamba, 2023-06-16:
### Board

ESP32-S3-DevKitC-1-N16R8V

### Device Description

Nothing attached.

### Hardware Configuration

Nothing attached.

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Windows 10, Macos Ventura 13.4

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I can upload and monitor sketches on my device in both Plarformio and Arduino IDEs, by opening the corresponding port. However, when I disconnect the device, and reconnect it, either to the same port, or to an external source, or to another PC, the app does not start. The reset button has no effect.

If I reopen the port in an IDE to monitor it, the app starts instantly.

This is tested with different ESP32-S3-DevKitC-1-N16R8V (I bought 3), both on Windows and Mac, and with my own sketches or with the examples (blink or hello).

I have worked with other ESP32 devkits before and never had this problem.

What can I do? Is there any software or config solution?

(apologies for my English).

### Sketch

```cpp
Checked with blink and hello examples.
```

### Debug Message

```plain
Nothing.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Xylopyrographer, 2023-06-16:
Dollars to donuts this is the Serial bug seen when using USB CDC. Search the issues for Serial.flush and a few items will pop up from others who've hit this problem. 
#7779
#8238
#7721

ReyWamba, 2023-06-18:
Thank you @Xylopyrographer. 

Indeed it seems to be a problem of Serial communication. 

```
 //Serial.begin(115200);
 //while (!Serial) 
```
fix the problem... but obviously it is not the solution.

I have tried the following, with no results:

- Adding the files HWCDC.h and HWCDC.cpp the ones proposed in #7779.
- I have tried both the solutions posed by @sblantipodi and @imwhocodes in #7754 and neither.

However, I owe you half a dozen donuts for guiding me to the solution, even though I haven't found it yet. Thanks again.

Xylopyrographer, 2023-06-18:
For me, two things were needed to make it work with an external supply or battery pack:

1. Remove all Serial.flush() statements.
2. Add a Serial.setTxTimeoutMs( 0 ); statement after Serial.begin();

Didn’t modify the HWCDC.h or .cpp files.

Hope that helps too.

me-no-dev, 2023-06-19:
if you have `while(!Serial);` in the beginning, the code will wait for the USB Serial to be attached, since you are not actually opening the port when powering externally... the sketch hangs

ReyWamba, 2023-06-19:
Hello again, @Xylopyrographer:

I have no Serial.flush() in my code, and I had already tried the solution of  Serial.setTxTimeoutMs( 0 ) (it was mentioned in some of the links you mentioned to me in your first reply), even trying to upload the program in all USB modes (ESP-PROG, USB-JTAG and ESP USB Brigde), and both with -D ARDUINO_USB_MODE=1 enabled and disabled in platformio.ini, and nothing. 

However, thanks to your and @me-no-dev 's suggestions, I have solved the problem (I explain how in the following answer).

So thanks again, and I owe you a dozen donuts!!!

closed 2023-06-19 by ReyWamba

Xylopyrographer, 2023-06-19:
Glad you got it sorted. 😄

We’re all learning a something new with the USB-CDC ESP32 versions.

me-no-dev, 2023-06-20:
> However, in Arduino IDE and with other ESP32 boards, this statement never gave me problems. 

That is because they have a separate USB-Serial chip that we have no idea when is connected. On newer chips we have integrated USB, which we can detect when connected and wait for it. Same goes for Arduino Leonardo for example. Arduino UNO will boot regardless, but Leonardo will wait.

ReyWamba, 2023-06-20:
I am a simple amateur, and these things are beyond my knowledge. The truth is that they waste a lot of time.

Thank you very much for your help again.