espressif/arduino-esp32 #8323: memory: No such file or directory

SSHUSSAIN-86, 2023-06-16:
### Board

esp32-c3-devkitc-02

### Device Description

I am using PlatformIO to build my project and I am using "SPIFFS.h" that has dependency on "memory". Please help resolve!
/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:24:10: fatal error: memory: No such file or directory

### Hardware Configuration

No just build errors and dependency errors!

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

60 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Build errors

### Sketch

```cpp
#include <freertos/task.h>
#include <sdkconfig.h>
#include <ArduinoJson.h>
#include <esp_system.h>
#include <Arduino.h>
#include <SPIFFS.h>
#include <FS.h>
```

### Debug Message

```plain
cannot open source file "memory" (dependency of "SPIFFS.h"). Please run the 'Select IntelliSense Configuration...' command to locate your system headers.C/C++(1696)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-17:
You do not need to put SPIFFS.h in your platformio.ini.  It is part of the core.

SSHUSSAIN-86, 2023-06-17:
I have removed SPIFFS.h from the platform.ini. Please see the attached file. If SPIFFS is part of the core then why it is not being recognized?
[image]

lbernstone, 2023-06-17:
Arduino runs in c++.  Name your root file main.cpp

SSHUSSAIN-86, 2023-06-17:
I like to run in espidf framework with "FS.h" and  "SPIFFS.h". I have changed to main.cpp with app_main() as my entry point

framework = espidf

[image]

Jason2866, 2023-06-17:
espidf is a different framework than Arduino. So this will not work.

SSHUSSAIN-86, 2023-06-17:
I know we can use esp-idf framework and use arduino as a component. This works. I am now interested in setting up PlatformIO to do that. Please see the following official link to the documentation.

https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Jason2866, 2023-06-17:
@SSHUSSAIN-86 Here is an [example (LittleFS)](https://github.com/Jason2866/Arduino_IDF_LittleFS) when Arduino is used as component with IDF.

SSHUSSAIN-86, 2023-06-17:
@Jason2866 Thanks. This really helps 🥇

lbernstone, 2023-06-17:
As shown in all these [examples](https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-blink/platformio.ini#L12), the framework should be `framework = arduino. espidf`

VojtechBartoska, 2023-06-20:
I'm closing this as covered, if you need more help, don't hesitate to reopen this.

closed 2023-06-20 by VojtechBartoska