espressif/arduino-esp32 #8324: Use pin I2C pin 43 as I/O for DS18B20 temp sensor

teastainGit, 2023-06-18:
### Board

ESP32 S3R8

### Device Description

LilyGO T-Display S3 

### Hardware Configuration

GPIO43 is suggested by LilyGO as the I2C pins, they are brought out to the JST connector.
I am trying to connect a DS18B20 "oneWire" temp sensor that will work on other GPIO but mot pin 43 or 44.
I2C devices work on these pins as expected.

### Version

v2.0.6

### IDE Name

Arduino IDE v2.1.0

### Operating System

MacOS Ventura 13.4

### Flash frequency

QIO 80

### PSRAM enabled

yes

### Upload speed

460800

### Description

I am trying to connect a Dallas Semiconductor DS18B20 temperature sensor to either the GPIO43 or the JST pin 43, which are probably the same circuit.
There is no response from the sensor at all, the oscilloscope shows 3.3v steady.
The sensor works on GPIO 1 (or GPIO 18, e.g.).
The pin 43 can be programmed as an input or an output successfully, but not the serial serial I/O of the DS18B20.
Further another user says that the JST pin 43 works in ESPHome, but we cannot get it to work on Arduino.  I cannot understand why it works in only one software.
My question therefore is:
"Is there a configuration setup for the LilyGO T-Display S3 that configures GPIO43 for I2C only? How can we free it up?"

### Sketch

```cpp
#include <OneWire.h>
#include <DallasTemperature.h>
const int oneWireBus = 43;     
OneWire oneWire(oneWireBus);
DallasTemperature sensors(&oneWire);

void setup() {
  // Start the Serial Monitor
  Serial.begin(115200);
  delay(200);
  Serial.println(" setup  ");
  delay(200);
   sensors.begin();
}

void loop() {
  sensors.requestTemperatures(); 
  float temperatureC = sensors.getTempCByIndex(0);
  Serial.print(temperatureC);
  Serial.println("ºC");
  Serial.println("   ");
  delay(2000);
```
```

### Debug Message

```plain
Do not think this is the issue!
```

### Other Steps to Reproduce

It works on ESPHome, but not Arduino IDE.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-06-18:
Try latest Arduino core 2.0.9

teastainGit, 2023-06-18:
> Try latest Arduino core 2.0.9
 @[Jason2866](https://github.com/Jason2866) Jason2866
Arduino IDE 2.1.0 would only offer v2.0.6 at first so I  deleted it and installed v2.0.9, no change!
I was hopeful!

lbernstone, 2023-06-18:
pin 43 is used by the UART serial.  If you use USB CDC for your serial connection, you should be able to use this pin.

teastainGit, 2023-06-19:
So, still no joy!
Here are my settings, cribbed from the LilyGO LilyGO T-Display S3 GitHub repository !
[image]

An observation:
When I upload my sketch by USB I see the little DS18B20 indicator flicker as if showing the upload data!?
It does not flicker when using GPIO 1, e.g.
Some advice from [lbernstone](https://github.com/lbernstone) that 43 is U0TXD makes sense if I'm seeing download data on the GPIO43, maybe? 
What command in the sketch setup (like PINMODE, which works perfectly with digitalRead and digitalWrite) but what setting can free-up pin 43 as 1wire serial?
Thanks for listening!
-Terry

SuGlider, 2023-06-20:
@teastainGit - Let me know it executing `Serial0.end()` solves this issue.

teastainGit, 2023-06-20:
[SuGlider](https://github.com/SuGlider) Here is a screen shot showing all the advice I followed!
(I use CDC on boot (Enabled) because my output is Serial.print!)
[image]

Thanks to all for your help, I think we are getting closer!
-Terry

SuGlider, 2023-06-21:
@teastainGit - Is it working correctly?

teastainGit, 2023-06-21:
> @teastainGit - Is it working correctly?

Sorry to not be clear, it was late when I stopped testing!
I think we are on the right track, but I have done all the suggestions and still no success.
I do not have a project waiting for an answer, just curious as to why it works on GPIO 1 but not 43
-Terry

teastainGit, 2023-08-18:
[VojtechBartoska](https://github.com/VojtechBartoska)
Thanks for asking!
After I read your question I set the experiment up again, with ESP32 2.0.11 and all other libraries updated...
Nope.
I tied it on my new LilyGO T-Display S3 AMOLED and...
Nope.
No help from  LilyGO . I also asked at the ESP32 Arduino GitHub Issues and no solution there.
-Terry

SuGlider, 2023-09-15:
@teastainGit - Please try Arduino Core 2.0.13. It shall sove this issue. Please let me know.

teastainGit, 2023-09-15:
What is it about ver 2.0.13 that shall make it worK?
I set the CDC on Boot "Enabled" and the suggested sketch changes for Serial0:

```
#include <OneWire.h>
#include <DallasTemperature.h>
OneWire oneWire(43);
DallasTemperature sensors(&oneWire);

void setup() {
  Serial0.begin(115200);
  Serial0.end();
  Serial.begin();
  delay(200);
  Serial.println(" setup ");
  delay(200);
  sensors.begin();
}

void loop() {
  sensors.requestTemperatures();
  float temperatureC = sensors.getTempCByIndex(0);
  Serial.print(temperatureC);
  Serial.println("ºC");
  Serial.println("   ");
  delay(2000);
}
```

I tried again with all updated libraries and this version of ESP32 as well, but no luck. Again, the sensor LED flickered  dimly during download, but the sensor would not work, giving a reading of -127.
But the sensor works on GPIO 1 just fine.
I also wrote this sketch to test 43 and 44 as output and input, and it worked perfectly!
```
const int buttonPin = 44;  // the number of the pushbutton pin
const int ledPin = 43;     // the number of the LED pin
int buttonState = 0;

void setup() {
  Serial.begin(115200);
  pinMode(buttonPin, INPUT);
  pinMode(ledPin, OUTPUT);
}

void loop() {
  buttonState = digitalRead(buttonPin);
  Serial.println(buttonState);
  if (buttonState == HIGH) {
    digitalWrite(ledPin, HIGH);
  } else {
    digitalWrite(ledPin, LOW);
  }
  delay(100);
}
```
I am not stuck on a project, just curious what is happening.
BTW it is a LilyGO T-Display S3 .
-Terry

Jason2866, 2023-09-18:
The used lib can be the reason for non working too. Have you tried a other sensor, not OneWire?

teastainGit, 2023-09-19:
I cannot find another DS18B20 library and I am using the one suggested by Random Nerd Tutorials “https://randomnerdtutorials.com/esp32-ds18b20-temperature-arduino-ide[/](https://randomnerdtutorials.com/esp32-ds18b20-temperature-arduino-ide/)“ using an ESP32 DevKit.
I have two LilyGO T-Display S3 and one LilyGO T-Display S3 AMOLED and the DS18B20 does not work on any of them.
I connect to their intended I2C JST connector at the end of the board.
These are connected to GPIO43 and 44 and work fine as:
I2C temp sensor SHT30
Discrete inputs
Discrete output
The DS18B20 works fine on these boards using GPIO 1, 2 and so on.
The DS18B20 works fine on other ESP32 boards including the M5Stack Core Basic, which is -DOWD series.

But...on the M5Stack "StampS3" model (ESP32S3), It does not work on pins 43 or 44, but does on other pins, GPIO1 e.g. (!)
So it is not limited to the LilyGO T-Display S3, but it does fail on other ESP32 S3s as well.
So, there's THAT.
-Terry

Jason2866, 2023-09-19:
The onewire lib does a lot of low level settings. There is a high chance that not all gpios are correctly handled in the lib for the S3. Since other sensors do work well I think it is the lib

Jason2866, 2023-09-19:
@teastainGit  @SuGlider
It IS the onewire lib. This line prevents the use https://github.com/PaulStoffregen/OneWire/blob/v2.3.7/util/OneWire_direct_gpio.h#L200

SuGlider, 2023-09-19:
Thanks @Jason2866 - Nice catch!
@teastainGit - You may need to change the code line that @Jason2866 has found as the source of the issue.
ESP32S3 has more GPIOs than the ESP32 and that `if ( digitalPinIsValid(pin) && pin <= 33 ) // pins above 33 can be only inputs` shall be changed to `if ( digitalPinIsValid(pin) && pin <= NUM_DIGITAL_PINS)` 
NUM_DIGITAL_PINS comes from `pins_arduino.h` linked to the Variant board.

closed 2023-09-19 by Jason2866

369Martin369, 2024-01-20:
Am confused...error is still present in version 2.3.7. needed to increase to 44 for T-Display

matsekberg, 2024-04-06:
I have exactly the same issue on a ESP32S3 but on pin 38 so its not a USART issue. I will try the util library fix....

matsekberg, 2024-04-06:
And the change in the util library worked !!
I will also test this library: https://github.com/junkfix/esp32-ds18b20