espressif/arduino-esp32 #8329: ESP-WROOM-32 doesn't send notifications when the device is about to restart.

meakashrao, 2023-06-19:
### Board

ESP-WROOM-32

### Device Description

ESP-Wroom-32 Devkit

### Hardware Configuration

A Rotary Encoder is connected to the ESP which also RGB LED Embedded into it.

int redPin = 12;
int greenPin = 2;
int bluePin = 4;
const int encoderPinA = 27;
const int encoderPinB = 14;
int sw = 13;

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am constantly receiving a notification on a particular characteristics every 100ms but if the device is about to restart according to a particular logic implemented then the device won't send the notification and just restart. 

### Sketch

```cpp
This function is called in void Loop() and if button press (otaButtonCounter) is 10, i.e the butoon being pressed 10 times, then the device should have sent this and restarted. What we have observed is that the device just restarts.

void OTA_Handler::setToOTA()
{
  if(otaTaskCounter)
  {
    ota_timer_handler.start(otaTimerPeriod);
    otaTaskCounter = false;
    OTA_Count_Start_Flag = true;
  }
  if(ota_timer_handler.isRunning())
  {
    if(otaButtonCounter == 10)
    {
      ota_timer_handler.stop();
      deviceInOTA = 1;
      updateEEPROMValue(1);
      char dialArray[8];  
      dialArray[0] = 'o';
      dialArray[1] = 't';   
      dialArray[2] = 'a';
      dialArray[3] = 'm';
      dialArray[4] = 'o';
      dialArray[5] = 'd';
      dialArray[6] = 'e';
      dialArray[7] = '\0';
      BLECharacteristic *notifyCharacteristic7 = myBLE.getNotifyCharacteristic("DIAL");
      if (notifyCharacteristic7 != nullptr)
      {
        myBLE.sendNotification(notifyCharacteristic7, dialArray);
      }
      OTA_Count_Start_Flag = false;
      ESP.restart();
    }
    if(ota_timer_handler.isFinished())
    {
      if(otaButtonCounter != 10)
      {
        OTA_Count_Start_Flag = false;
        otaButtonCounter = 0;
        deviceInOTA = 0;
      }
    }
  }
}

But, 
void updateDial()
{
  char dialArray[5];  
  int rotary_rot = read_rotary_encoder_rotation();
  dialArray[0] = read_rotary_encoder_button()%10 + 48;
  dialArray[1] = (((int)rotary_rot/1)%10) + 48;   
  dialArray[2] = ((otaButtonCounter/10)%10)+48;
  dialArray[3] = ((otaButtonCounter/1)%10)+48;
  dialArray[4] = '\0';
  BLECharacteristic *notifyCharacteristic7 = myBLE.getNotifyCharacteristic("DIAL");
  if (notifyCharacteristic7 != nullptr)
  {
    myBLE.sendNotification(notifyCharacteristic7, dialArray);
  }
  encoderRotation = 5;
}

This function works correctly and sends data on the same characteristics. So, what I have observed is that I get 0501 to 0509 and then the ESP restarts.

void BLE_Wrapper::sendNotification(BLECharacteristic *characteristic, const char *value)
{
  characteristic->setValue(value);
  characteristic->notify();
  // Serial.println("Sent notification");
}
```

### Debug Message

```plain
There is no Debug Message, everything works fine, I even see characters getting notified on either NRF Connect or the custom office APP
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-06-19:
just delay a bit before restarting to allow the BLE code to execute

meakashrao, 2023-06-19:
adding a delay(50) didn't work for me. I'll try with to do a software timer for ESP.restart()

mrengineer7777, 2023-06-19:
I use a custom reboot routine to shutdown services, save settings, etc.  delay() works great to allow other threads to finish executing and replies to be send.  It's likely something in your architecture.

meakashrao, 2023-06-20:
> I use a custom reboot routine to shutdown services, save settings, etc. delay() works great to allow other threads to finish executing and replies to be send. It's likely something in your architecture.

Thank you for the guidance

closed 2023-06-20 by meakashrao