espressif/arduino-esp32 #8330: ESP Variables not getting updated

meakash, 2023-06-19:
### Board

ESP32 Dev Module (Wroom)

### Device Description

ESP 32 Dev Kit

### Hardware Configuration

Nothing. 

### Version

v2.0.9

### IDE Name

Platfrom IO

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The problem I face with my code is that sometimes, my variables at certain places don't get updated as they should.

For Ex: pause_flag in the below Snippet. If there some ESP logging Statements, it does get updated otherwise it doesn't.

Am I missing something with respect to SRAM or something

My program is huge, as it uses BLE and WIFI both

Linking .pio\build\esp32doit-devkit-v1\firmware.elf
Retrieving maximum program size .pio\build\esp32doit-devkit-v1\firmware.elf
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.0% (used 68956 bytes from 327680 bytes)
Flash: [========  ]  83.3% (used 1637725 bytes from 1966080 bytes)
Building .pio\build\esp32doit-devkit-v1\firmware.bin
esptool.py v4.5
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Configuring upload protocol...

### Sketch

```cpp
volatile int pause_flag = 0;

void pause_function()
{
  if (pause_flag == 0)
  {
    ESP_LOGI(TAG, "Inside Pause ");
    //ESP_LOGI(TAG, "PF in Pause FN %d", pause_flag);
    pause_datum = millis();
    pause_flag++;
    original_datum = datum_time;
    statusBeforePause = read_currentStatus();
    pause_state = command_received;
    pause_data = 'P';
    command_received = 'P';
    int bld = set_blade(0, 0);
    pb_write_stir(0);
    int tmp = set_temp(0, 0);
    if (readStirModeFlag() == 1)
    {
      updateStirMode(2);
    }
    pause_flag++;
  }
  current_pause_duration = (millis() - pause_datum);
  pause_duration = (current_pause_duration + int(total_pause_duration));
  update_currentStatus(8);
  updateStatusSentence();
}
```

### Debug Message

```plain
[ 12749][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:26:29.167 > [ 12811][I][main.cpp:1917] loop(): [Main] SBP: 0
17:26:29.219 > [ 12923][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:26:29.282 > [ 12929][I][main.cpp:1917] loop(): [Main] SBP: 0
17:26:29.334 > [ 12977][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:26:29.396 > [ 13040][I][main.cpp:1917] loop(): [Main] SBP: 0
17:26:29.448 > [ 13092][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0

17:26:29.511 > [ 13153][I][main.cpp:1600] pause_function(): [Main] Inside Pause 
17:26:29.581 > [ 13285][I][main.cpp:1917] loop(): [Main] SBP: 6
17:26:29.633 > [ 13347][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0

17:26:29.695 > [ 13348][I][main.cpp:1600] pause_function(): [Main] Inside Pause 
17:26:29.765 > [ 13468][I][main.cpp:1917] loop(): [Main] SBP: 8
17:26:29.817 > [ 13468][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2
17:26:29.880 > [ 13522][I][main.cpp:1917] loop(): [Main] SBP: 8
17:26:29.931 > [ 13634][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2
17:26:29.994 > [ 13637][I][main.cpp:1917] loop(): [Main] SBP: 8
17:26:30.046 > [ 13689][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2

So, as you see the device should have entered inside Pause only once and there is no resume being passed

but

 [ 12094][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:32:31.456 > [ 12156][I][main.cpp:1917] loop(): [Main] SBP: 0
17:32:31.508 > [ 12268][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:32:31.571 > [ 12271][I][main.cpp:1917] loop(): [Main] SBP: 0
17:32:31.623 > [ 12323][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 0
17:32:31.685 > [ 12384][I][main.cpp:1842] pbReadinessCheck(): [Main] Sending to Pause 
17:32:31.761 > [ 12460][I][main.cpp:1600] pause_function(): [Main] Inside Pause 
17:32:31.830 > [ 12530][I][main.cpp:1601] pause_function(): [Main] PF in Pause FN 0
17:32:31.904 > [ 12663][I][main.cpp:1917] loop(): [Main] SBP: 6
17:32:31.955 > [ 12724][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2
17:32:32.018 > [ 12724][I][main.cpp:1842] pbReadinessCheck(): [Main] Sending to Pause 
17:32:32.095 > [ 12794][I][main.cpp:1917] loop(): [Main] SBP: 6
17:32:32.147 > [ 12846][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2
17:32:32.208 > [ 12908][I][main.cpp:1842] pbReadinessCheck(): [Main] Sending to Pause 
17:32:32.285 > [ 12986][I][main.cpp:1917] loop(): [Main] SBP: 6
17:32:32.337 > [ 13100][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2
17:32:32.400 > [ 13100][I][main.cpp:1842] pbReadinessCheck(): [Main] Sending to Pause 
17:32:32.476 > [ 13177][I][main.cpp:1917] loop(): [Main] SBP: 6
17:32:32.528 > [ 13229][I][main.cpp:1834] pbReadinessCheck(): [Main] PF 2

as you see here, it didn't enter the pause logic twice.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-06-19:
Your sketch is incomplete. Setup a separate project where you can demo the issue.

mrengineer7777, 2023-06-19:
Also it looks like you are trying to setup a Semaphore.  Check out the Semaphore and Mutex examples.

meakashrao, 2023-06-21:
Thank you for the help.

closed 2023-06-21 by mrengineer7777