espressif/arduino-esp32 #8334: JTAG programming ESP32-C3 Not Working

mjs513, 2023-06-19:
### Board

ESP32-C3 Dev Board

### Device Description

[DIYmall NiceMCU-C3F V1.0 ESP32-C3](https://www.amazon.com/dp/B0BN1CHKWJ?psc=1&ref=ppx_yo2ov_dt_b_product_details)

### Hardware Configuration

RGB LED attached to GPIO8, GPIO5 and GPIO4.

GPIO18/19 attached to usb cable:
GPIO18 D- to WTH on the cable
GPIO19 D+ to GRN on the cable

### Version

v2.0.9

### IDE Name

Arduino IDE 2.1.1 nightly 20230529

### Operating System

Windows 10 x64

### Flash frequency

80Mz

### PSRAM enabled

yes

### Upload speed

921600

### Description

For last week or so we have been trying to get Debug working using the Arduino IDE 2.x debugger with no luck. On setup is based on attacing a USB cable to pins IO18/19 with gnd to gnd and powering from the USB connector.  This is based on the following:
[Configure ESP32-C3 built-in JTAG Interface](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html)

We also downloaded ESP-tools so we have the correct drivers install as we saw this:  [Arduino IDE 2.X Debug](https://esp32.com/viewtopic.php?f=19&t=32585&hilit=jtag)
however even with the driver installed I am see this for the USB connected to D-/D_
[image]

When I went to manually install the driver it did not update so went in and tried a different way and gave me an error that said I needed the x64 driver???

GDB console shows the following error:
```
Debug: 223 62 semihosting_common.c:107 semihosting_common_init():  
Error: 224 91 esp_usb_jtag.c:642 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 225 91 command.c:545 run_command(): Command 'init' failed with error code -4
User : 226 92 command.c:608 command_run_line(): 
Error: 227 92 riscv.c:1755 riscv_get_gdb_arch(): Unsupported xlen: -1
Error: 228 92 esp_semihosting.c:67 target_to_esp_semihost_data(): Unknown target arch!
Debug: 229 92 riscv.c:490 riscv_deinit_target(): riscv_deinit_target()
Debug: 230 92 target.c:2225 target_free_all_working_areas_restore(): freeing all working areas
[2023-06-19T17:27:37.548Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

### Sketch

```cpp
/*
  Blink
[... 457 lines cut ...]
SPI_PAD_CONFIG_HD (BLOCK1)                         SPI HD (D3) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI WP (D2) pad                                    = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI DQS pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI D4 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI D5 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)

Usb Config fuses:
DIS_USB_JTAG (BLOCK0)                              Disables USB JTAG. JTAG access via pads is control = False R/W (0b0)
                                                   led separately
DIS_USB_DEVICE (BLOCK0)                            Disables USB DEVICE                                = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = False R/W (0b0)

PS D:\Espressif\frameworks\esp-idf-v5.0.2>
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

KurtE, 2023-06-19:
@mjs513 - I have the same issue with a different C3 module: https://www.amazon.com/dp/B0BWN14X65

I have also tried hooking it up to an RPI Pico board. running as a CMIS-DAP JTAG using https://github.com/ataradov/free-dap
I have another Pico running the same software that talks to an Arduino MKRZero using swd mode. I believe the ESP32-C3 does not support SWD mode so I hooked it up with 4 signal wires:
RPI pin   ESP32_pin
11(SWCLK/TCK)  -> GPIO6
12(SWDIO/TMS) -> GPIO4
13 TDI -> GPIO5
14 TDO -> GPIO7

So far they are not talking.  It is unclear to me how to configure the IDE to support this?
[image]

me-no-dev, 2023-06-20:
@mjs513 your configuration is correct. Did you make sure that before starting debug you actually built and uploaded the sketch that you want to debug? This must be done any time you restart the IDE/reopen the sketch. Necessary because some files need to be copied to the sketch folder and ELF to be picked up by GDB. Use only the top 3 buttons of the IDE to compile and start debug.

me-no-dev, 2023-06-20:
@KurtE C3 has integrated JTAG. No external hardware or connections are necessary other than the USB port, as @mjs513 explained

mjs513, 2023-06-20:
@me-no-dev thanks for getting back to us so fast.  The answer to your question is yes I uploaded the sketch first and then hit the debug button on the top of the IDE.  As @KurtE said he tried it with a different board than I have as well as trying to use the JTAG connectors.

At this point do not know where to go from here.  As you said it should work.

mjs513, 2023-06-20:
@me-no-dev and @KurtE 
Tried using VSCode just to see if would recognize the port and upload and if failed with the same error.  Its not seeing the usb cable attached to pins 18/19 as as seperate device.  Only seeing one device on com8.  From what I understand I should be seeing 2 com ports but I am not.  If I try to install with have disk I am seeing:

[image]

SuGlider, 2023-06-21:
> Wondering when the board boots up and I open up the Serial Monitor window, I see a bunch of text messages: That appear like the keep repeating, with data like:
> 
> ```
> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
> Saved PC:0x403cf890
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd5810,len:0x438
> load:0x403cc710,len:0x91c
> load:0x403ce710,len:0x25b0
> entry 0x403cc710
> ```
> 
> If this is related with JTAG stuff or potentially some internal debug stuff that is outputting by default?

@KurtE 
This is the ESP32-C3 bootloader messages. 
Not related to the debugging  process.

mjs513, 2023-06-22:
Not sure if this pertains to the current issue or not.  After some more googling I found this
https://www.esp32.com/viewtopic.php?t=31380

basically it adds the following command arg to opencd:
`-c "gdb_memory_map disable"`
I would try to add it but not sure where to even give it a try.

mjs513, 2023-06-22:
I just found this related to wifi: https://github.com/espressif/esp-at/issues/615#issuecomment-1065018934

SuGlider, 2023-06-22:
> I just found this related to wifi: [espressif/esp-at#615 (comment)](https://github.com/espressif/esp-at/issues/615#issuecomment-1065018934)

Arduino 2.0.9 uses IDF 4.4, therefore this fix is already included.

mjs513, 2023-06-22:
> > I just found this related to wifi: [espressif/esp-at#615 (comment)](https://github.com/espressif/esp-at/issues/615#issuecomment-1065018934)
> 
> Arduino 2.0.9 uses IDF 4.4, therefore this fix is already included.

Thanks - need to make sure.  But I just received a ESP32-C3 WROOM DevKitC-02 and connected it up and it works out of the box - the second serial port shows as USB Serial that pins 18/19 are connected with the USB cable so it has to be something with the board.

mjs513, 2023-06-22:
Just to close the loop I finally tracked down the issue with JTAG not working on this paticular board.  Turns out its a hardware issue - pins18/19 are shorted together even though there does not seem to be a solder bridge on my soldering on the header pins.  Pulled out my multimeter and tested.  Tried it a different board (came in a package of three) and it worked so going to close out this issue.

KurtE, 2023-06-23:
@me-no-dev @mjs513 - I can start up the debugger on a couple of the different boards, but wondering does any of it actually work? 

Once it starts up can you get it to set a breakpoint and have it be hit.  And if it is hit, can you continue from this location?
Can you use the step in and step over? 

If I try to get the code to continue, the gdb-server panel starts producing lots of debug output. I am running the c3 devkit m...
And nothing beyond the startup appears to work for me.  
I am probably missing something obvious.  but wondering if any of the stuff works for you?

Kurt

mjs513, 2023-06-23:
@KurtE 
Just hooked the C3 up again (Espressif WROOM esp32-c3 devkit) and can confirm its doing the same thing for me as well.  Yesterday I did not notice it.

I am running the BlinkRGB example with a extra Serial.print right before the endif.  I hit continue button and I get
```
gdb_log_outgoing_packet(): [esp32c3] sending packet: $a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a [... cut ...]
Hardware assisted breakpoint 1 at 0x42000020: file D:\Users\Merli\Documents\Arduino\BlinkRGB_ESP32C3/BlinkRGB_ESP32C3.ino, line 20.
Note: automatically using hardware breakpoints for read-only addresses.
Program
 received signal SIGINT, Interrupt.
0x4004c0d6 in ?? ()
[New Thread 1070463660]
[New Thread 1070154348]
[New Thread 1070145936]
[New Thread 1070151672]
[Switching to Thread 1070463660]
Thread 2 "loopTask" hit Temporary breakpoint 1, setup () at D:\Users\Merli\Documents\Arduino\BlinkRGB_ESP32C3/BlinkRGB_ESP32C3.ino:20
20	  Serial.begin(115200);
```

VojtechBartoska, 2023-08-18:
@me-no-dev Can you please help with triage and possibly close this issue? Thanks

me-no-dev, 2023-08-22:
@VojtechBartoska we need to investigate all outcomes and provide a good info/tutorial on using JTAG with Arduino. The last message clearly states that the JTAG hit the `setup()` breakpoint, but I suspect that the interface in the IDE did not update. There are different issues between the plugin versions used in the different IDE versions and also some on our side (openocd/gdb) that we have poked the relevant teams about, but seem unfixed still.

I want to make clear that we are mere users of what Espressif offers as OpenOCD/GDB and what Arduino.cc offers as IDE2+cortex-debug. Our configurations were vetted by the IDF team and should work as best as possible on the current platform. It's the nuances between the chips/architecture/interface that are most important to figure out and document. When using external debugger (as required for ESP32 and ESP32-S2) there are far less issues and debugging works much more as expected. Having the USB-JTAG internal, means that the interface gets restarted with the chip and more.