espressif/arduino-esp32 #8341: ESP32-C3 - Wake up from deep sleep zeroes RTC memory and will NOT report GPIO as a wake up cause

bricomp, 2023-06-21:
### Board

WeActStudio ESP32C3

### Device Description

Dev board on plug in development

### Hardware Configuration

Serial on 20&21.
IO on 4,5,6&7 connected to Teensy (either 3.1 or LC)
Shared ground with Teensy.
Both MCUs connected to USB

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Window 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

RTC variables are reset to zero when waking up from deep sleep, also esp_sleep_get_wakeup_cause() always returns zero.
The Teensy (3.1 or LC) is providing the reset signal.

### Sketch

```cpp
#define LED 8  // or use LED_BUILTIN for on-board LED
#define INTERRUPT_PIN 4
RTC_DATA_ATTR int bootCount; // = 0;

void setup() {
    Serial.begin(115200);
    delay(1000);

    pinMode(LED, OUTPUT);
    digitalWrite(LED, LOW);

    ++bootCount;
    Serial.println("Boot number: " + String(bootCount));

    print_wakeup_reason();

    Serial.println("Going to blink the LED 10 times");
    blink(10);
    Serial.println("Going to sleep now");
    Serial.flush();  delay(50);
    esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN,
        ESP_GPIO_WAKEUP_GPIO_LOW);
    esp_deep_sleep_start();

    Serial.println("This will never be printed");
}

void loop() { }

void blink(int times) {
    for (int i = 0; i < times; i++) {
        digitalWrite(LED, HIGH);
        delay(1000);

        digitalWrite(LED, LOW);
        delay(1000);
        Serial.print("Blink ");
        Serial.println(i + 1);
    }
}

void print_wakeup_reason() {
    esp_sleep_wakeup_cause_t wakeup_reason;

    wakeup_reason = esp_sleep_get_wakeup_cause();

    switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:
        Serial.println("Wakeup caused by external signal using RTC_IO");
        break;
    case ESP_SLEEP_WAKEUP_EXT1:
        Serial.println("Wakeup caused by external signal using RTC_CNTL");
        break;
    case ESP_SLEEP_WAKEUP_TIMER:
        Serial.println("Wakeup caused by timer");
        break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
        Serial.println("Wakeup caused by touchpad");
        break;
    case ESP_SLEEP_WAKEUP_ULP:
        Serial.println("Wakeup caused by ULP program");
        break;
    default:
        Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
        break;
    }
}
```

### Debug Message

```plain
Port open
Boot number: 1
Wakeup was not caused by deep sleep: 0
Going to blink the LED 10 times
Blink 1
Blink 2
Blink 3
Blink 4
Blink 5
Blink 6
Blink 7
Blink 8
Blink 9
Blink 10
Going to sleep now
Port open
Boot number: 1
Wakeup was not caused by deep sleep: 0
Going to blink the LED 10 times
Blink 1
Blink 2
Blink 3
Blink 4
Blink 5
Blink 6
Blink 7
Blink 8
Blink 9
Blink 10
Going to sleep now
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-06-23:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html

bricomp, 2023-06-23:
@mrengineer7777 that document does NOT refer to the ESP32-C3!

lbernstone, 2023-06-24:
> @mrengineer7777 that document does NOT refer to the ESP32-C3!

So go to the top of the page, and switch to the esp32-c3 and IDF 4.4.4.
Deep sleep will shut down any power domains that aren't used.  If you aren't using a timer, the RTC domain can be shut down.  That page will show you how to ensure it stays up.  The wakeup cause should work though.  Does it work for you if you use the example without modification (timer based)?

bricomp, 2023-06-25:
IDF 4.4.4 was NOT available on the site so I looked at 4.4.5. The paragraph below is an extract from 4.4.5.

> In_ ESP32-C3 there is only RTC fast memory, so if some variables in the program are marked by 
> RTC_DATA_ATTR, RTC_SLOW_ATTR or RTC_FAST_ATTR attributes, all of them go to RTC fast memory. It will be 
> kept powered on by default. This can be overridden using esp_sleep_pd_config() function, if desired.

You should note from the code above that I am using **RTC_DATA_ATTR int bootCount;** which should according to the 
documentation keep the RTC alive.

bricomp, 2023-06-25:
> Does it work for you if you use the example without modification (timer based)?
NO!

SuGlider, 2023-06-26:
@bricomp - What is the Arduino Core version used?

SuGlider, 2023-06-26:
I have tested your sketch with Arduino Core 2.0.9 and it works fine.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
Boot number: 1
Wakeup was not caused by deep sleep: 0
Going to blink the LED 10 times
Blink 1
Blink 2
Blink 3
Blink 4
Blink 5
Blink 6
Blink 7
Blink 8
Blink 9
Blink 10
Going to sleep now
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
Boot number: 2
Wakeup was not caused by deep sleep: 7
Going to blink the LED 10 times
Blink 1
Blink 2
Blink 3
Blink 4
Blink 5
Blink 6
Blink 7
Blink 8
Blink 9
Blink 10
Going to sleep now
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
Boot number: 3
Wakeup was not caused by deep sleep: 7
Going to blink the LED 10 times
Blink 1
Blink 2
Blink 3
Blink 4
Blink 5
Blink 6
Blink 7
Blink 8
Blink 9
Blink 10
Going to sleep now
```

ferzone, 2023-07-22:
I have the same  problem here. One importante thing to reproduce it is the time. If you wake-up immediately after go deep sleep, everything is ok. But if you wait ‘human time’ like a second or more the wake up have this issue.
I am using ESP32-C3-Mini module, it have a version 0.4 of the silicon, the rest is exactly the same. I test it a lot, hardware and software, used version 2.0.8/2.0.9 and 2.0.10 of Arduino.

ferzone, 2023-07-24:
A few more thinks after a lot of test. If you use a button close to the module, everything works fine. The problem come when you try to use just a 'logical' '0' from another logical circuit (AK other uC) perhaps the slew rate? Espressif recommend a strong pull-up for use in this case, but I think a Schmitt trigger is a more reliable solution, but I'm unable to test it now.

SuGlider, 2023-07-24:
ESP32C3 GPIO Wakeup only works with Voltage Level (3.3v or 0v), not with edges.
Therefore, maybe, the issue is, as you already said, that the other logical circuit is not producing a recognizable Low level. 
Given that it works with a regular push button that takes the GPIO to 0v when pressed, I'd say that there is no issue with the C3. 
It seems to work as expected.

VojtechBartoska, 2023-08-21:
can this be closed @bricomp?

bricomp, 2023-08-21:
Yes, solution found. The problem occurs when the monitor is also open. A fix is in the working.

closed 2023-08-22 by VojtechBartoska