espressif/arduino-esp32 #8344: 使用蓝牙时出现库中的未定义引用

camerakang, 2023-06-24:
### Board

ESP32 S3R8N32

### Device Description

无连接

### Hardware Configuration

无连接

### Version

latest master (checkout manually)

### IDE Name

platformIO

### Operating System

macOS12.1

### Flash frequency

40M

### PSRAM enabled

yes

### Upload speed

115200

### Description

当我使用esp32自带的蓝牙时，调用了
```
BluetoothSerial SerialBT;
SerialBT.begin("ESP32");
SerialBT.print("hellow world\n");
```
出现了以下报错，它貌似是在库文件中的调用，我不是很理解为什么会出现这样的问题，要如何解决

   ^~~~~~~
Compiling .pio/build/esp32-s3-devkitc-1/liba0d/BluetoothSerial/my_bluetooth.cpp.o
Archiving .pio/build/esp32-s3-devkitc-1/liba0d/libBluetoothSerial.a
Indexing .pio/build/esp32-s3-devkitc-1/liba0d/libBluetoothSerial.a
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL8_stop_btv+0x18): undefined reference to `esp_spp_disconnect'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL8_stop_btv+0x1c): undefined reference to `esp_spp_deinit'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL16_spp_send_bufferv+0x24): undefined reference to `esp_spp_write'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL8btSetPinv+0xc): undefined reference to `esp_bt_gap_set_pin'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZN15BluetoothSerial5beginE6Stringb+0x70): undefined reference to `esp_bt_gap_register_callback'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZN15BluetoothSerial5beginE6Stringb+0x74): undefined reference to `esp_spp_register_callback'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZN15BluetoothSerial5beginE6Stringb+0x78): undefined reference to `esp_spp_init'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZN15BluetoothSerial5beginE6Stringb+0x7c): undefined reference to `esp_bt_gap_set_security_param'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZN15BluetoothSerial5beginE6Stringb+0x80): undefined reference to `esp_bt_gap_set_cod'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x44): undefined reference to `esp_bt_gap_set_scan_mode'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x48): undefined reference to `esp_spp_start_srv'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x50): undefined reference to `esp_spp_connect'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x2c): undefined reference to `esp_bt_gap_resolve_eir_data'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x34): undefined reference to `esp_bt_gap_cancel_discovery'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x38): undefined reference to `esp_spp_start_discovery'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x50): undefined reference to `esp_bt_gap_pin_reply'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL13esp_bt_gap_cb21esp_bt_gap_cb_event_tP21esp_bt_gap_cb_param_t+0x54): undefined reference to `esp_bt_gap_ssp_confirm_reply'
/Users/kzc/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/lib943/libBluetoothSerial.a(BluetoothSerial.cpp.o): in function `_stop_bt()':
/Users/kzc/.platformio/packages/framework-arduinoespressif32/libraries/BluetoothSerial/src/BluetoothSerial.cpp:714: undefined reference to `esp_spp_disconnect'
[... 46 lines cut ...]
}

#include "my_bluetooth.cpp"
void setup()
{
    Serial.begin(115200);
    my_bluetooth_init();
}

void loop()
{
    my_bluetooth_function();
    delay(5);
}
```

### Debug Message

```plain
无
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-24:
ESP32-S3 does not include bluetooth classic, only BLE.

VojtechBartoska, 2023-06-27:
Closing as asnwered, if needed you can reopen it.

closed 2023-06-27 by VojtechBartoska