espressif/arduino-esp32 #8346: Is there a library for handling SD cards on the ESP32-S3?

hzhh110, 2023-06-25:
### Related area

ESP32-S3-EYE

### Hardware specification

ESP32-S3-EYE

### Is your feature request related to a problem?

I am using the ESP32-S3-EYE(esp32s3camlcd) development board. How do you operate an SD card with arduino? Is there a library for that?

### Describe the solution you'd like

I am using the ESP32-S3-EYE(esp32s3camlcd) development board. How do you operate an SD card with arduino? Is there a library for that?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2023-06-26:
@P-R-O-C-H-Y can you please help with triage?

P-R-O-C-H-Y, 2023-06-26:
Hi @hzhh110, the SD / SDMMC libraries are already included in the arduino-esp32. 
You can go to examples -> SD -> SD_Test.ino or SDMMC -> SDMMC_Test.ino and give it a try.
You may need to specify pins for the ESP32-S3-EYE board in order to have it working properly.

If you want to test SDMMC example on S3-EYE, add this to the top to specify pins:
```
#define SDMMC_CLK  39
#define SDMMC_CMD  38
#define SDMMC_D0   40  
```
And in `setup()` function add call `SD_MMC.setPins()` and change default values of `SD_MMC.begin()` to following to run in 1bit mode at 20Mhz:
```
Serial.begin(115200);
SD_MMC.setPins(SDMMC_CLK, SDMMC_CMD, SDMMC_D0);
if(!SD_MMC.begin("/sdcard", true, false, 20000, 5)){
     Serial.println("Card Mount Failed");
     return;
}
```

keskinonur, 2023-07-19:
@P-R-O-C-H-Y do you have any sample project links for esp32-s3-eye for arduino or platformio?

P-R-O-C-H-Y, 2023-07-20:
Hi @keskinonur, I just have simple sketch to test writing, reading, renaming functions for SD library. It is one of the examples I mentioned in comment above. Just apply the changes from the comment and you should be ready to go :)

closed 2023-08-07 by P-R-O-C-H-Y

Shaquum, 2023-10-02:
we dont recive aby notices for the solving issue, FYI.